cncjs/cncjs #424: Wanting to build an LED Ring light

ItsRobbAllen, 2019-03-12:
Good evening, I have a rather odd request.

I have a NeoPixel ring that I am planning on attaching to my spindle. For the most part, I want it lit white so I can see better. It will be run by a separate Uno, but since it's a NeoPixel ring, I'd like have fun with it. 

My idea would be to set certain light patterns for certain events (not tied directly to CNC.js, but rather generic).

For Example -
When running, all lights would be bright white while 1 pixel is yellow, indicating the XY travel direction
During a tool change, it would animate in a chasing pattern in green.
If paused - it would blink yellow.
If in error - it would flash red

What I'm trying to figure out is where to start in CNC.js with a widget that 
a) could connect to a separate serial port
b) could send messages
c) how to read these states from CNC.js 

I'm a seasoned .Net dev with Angular & JavaScript experience. I can figure out most of the code necessary for the UI pieces, but I'm curious to know if there's any shortcuts on how to figure out how to read that stream? I'm going to look into the display section & see if I can peel off some of the info, but if anyone has any thoughts or ideas where to start, I'd appreciate it!

cheton, 2019-03-12:
@ItsRobbAllen

Using the pendant code, you can listen to the `Grbl:state` event emitted from CNCjs server:
https://github.com/cncjs/cncjs-pendant-boilerplate/blob/master/bin/cncjs-pendant-boilerplate#L63-L66

This sample code illustrates how to change led color with state change:
```js
// [Grbl Controller]
// state = {
//      status: {
//          activeState: '',
//          mpos: {
//              x: '0.000',
//              y: '0.000',
//              z: '0.000'
//          },
//          wpos: {
//              x: '0.000',
//              y: '0.000',
//              z: '0.000'
//          },
//          ov: []
//      },
//      parserstate: {
//          modal: {
//              motion: 'G0', // G0, G1, G2, G3, G38.2, G38.3, G38.4, G38.5, G80
//              wcs: 'G54', // G54, G55, G56, G57, G58, G59
//              plane: 'G17', // G17: xy-plane, G18: xz-plane, G19: yz-plane
//              units: 'G21', // G20: Inches, G21: Millimeters
//              distance: 'G90', // G90: Absolute, G91: Relative
//              feedrate: 'G94', // G93: Inverse time mode, G94: Units per minute
//              program: 'M0', // M0, M1, M2, M30
//              spindle: 'M5', // M3: Spindle (cw), M4: Spindle (ccw), M5: Spindle off
//              coolant: 'M9' // M7: Mist coolant, M8: Flood coolant, M9: Coolant off, [M7,M8]: Both on
//          },
//          tool: '',
//          feedrate: '',
//          spindle: ''
//      }
//  }

let prevState = null;

socket.on('Grbl:state', function(state) {
    if (!prevState) {
        prevState = { ...state };
        return;
    }

    const { status: { mpos: { x: prevX, y: prevY, z: prevZ } } } = prevState;
    const { status: { mpos: { x: currX, y: currY, z: currZ } } } = state;

    if (prevX !== currX && prevY !== currY) { // XY travel
        led.light('yellow');
    }

    prevState = { ...state };
});
```

ItsRobbAllen, 2019-03-12:
Super helpful, that should get me started. Thanks for the quick info.

karoria, 2019-03-12:
@ItsRobbAllen 
I am also interested to use it if you dont mind sharing your work once done :)
Very nice idea..

closed 2019-03-20 by cheton