cncjs/cncjs #425: G2 speed and other settings not persistent

SketchThis, 2019-03-12:
#### Description

Hey all! I'm new to CNCjs but not new to building/retrofitting CNC machines with new controllers.

I've recently installed CNCjs on a Pi, and connected it to a G2 controller (specifically a ShopBot G2 setup) http://gofabmo.org/

I'm essentially replacing the "FabMo Engine" (it's in Intel Edison with ShopBot's UI on it with a raspberry Pi running CNCjs. 

I connected the Pi to the G2 controller and it works. I'm able to jog around, run jobs, etc. What I'm running into is settings are not persistent. For example, if I type in $xjm=1000 (set the jog speed faster) the setting works, but if I reload CNCjs (disconnect and reconnect) that setting is not persistent. 

The same seems to be happening for other settings, like soft limits, etc. Is this a CNCjs "issue" or my lack of understanding about how settings work? Just curious where to look to try and solve this. Thanks for any help!

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

ril3y, 2019-03-12:
G2 does not persist currently.  Are you resetting the board ?  Or just reloading?  If you are reloading then it looks like cncjs, upon connection is using some default settings or something.

SketchThis, 2019-03-12:
This happens when I just reload CNC js, and I'm definitely seeing some settings sent to it when it does. This is the console when I first reconnect (see below)

I haven't made sense of all of this yet, but at a high level is CNCjs sending some default settings every time it connects? And if so, can I change those somehow?

`CNCjs 1.9.16 [TinyG]
Connected to /dev/ttyACM0 with a baud rate of 115200
{"r":{"fv":0.99,"fb":100.26,"fbs":"100.26-313-geac2c","fbc":"settings_shopbot_sbv300.h","hp":"sbv300","hv":"k","id":"0084-d639-08c6-108","msg":"SYSTEM READY"},"f":[1,0,1]}
feeder> {ej:1}
{"r":{"ej":1},"f":[1,0,8]}
feeder> {jv:4}
feeder> {qv:1}
{"r":{"jv":4},"f":[1,0,8]}
feeder> {sv:1}
feeder> {si:100}
{"r":{"qv":1},"f":[1,0,8]}
{"r":{"sv":1},"f":[1,0,8]}
{"r":{"si":null,"msg":"Input less than minimum value 200"},"f":[1,108,10]}
{"err":{"code":108,"msg":"Input less than minimum value"}}
feeder> {spe:n}
{"r":{"spe":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spd:n}
{"r":{"spd":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spc:n}
{"r":{"spc":0},"f":[1,0,9]}
feeder> {sps:n}
{"r":{"sps":0},"f":[1,0,9]}
feeder> {com:n}
{"r":{"com":0},"f":[1,0,9]}
feeder> {cof:n}
{"r":{"cof":0},"f":[1,0,9]}
feeder> {sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spc:t,sps:t,com:t,cof:t}}
{"r":{"sr":{"line":0,"vel":0,"feed":0,"stat":1,"cycs":0,"mots":0,"hold":0,"momo":4,"coor":2,"plan":0,"unit":0,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,213]}
feeder> {sys:n}
{"r":{"sys":{"fb":100.26,"fv":0.99,"fbs":"100.26-313-geac2c","fbc":"settings_shopbot_sbv300.h","hp":"sbv300","hv":"k","id":"0084-d639-08c6-108","jt":0.8,"ct":0,"zl":0.5,"sl":1,"lim":1,"saf":1,"m48":1,"froe":0,"fro":1,"troe":0,"tro":1,"mt":2,"tv":1,"ej":1,"jv":4,"qv":1,"sv":1,"si":250,"gpl":0,"gun":0,"gco":2,"gpa":2,"gdi":0}},"f":[1,0,9]}
feeder> {mt:n}
{"r":{"mt":2},"f":[1,0,8]}
feeder> {pwr:n}
{"r":{"pwr":{"1":0,"2":0,"3":0,"4":0,"5":0}},"f":[1,0,9]}
feeder> {qr:n}
{"r":{"qr":48},"f":[1,0,8]}
feeder> {sr:n}
{"r":{"sr":{"line":0,"vel":0,"feed":0,"stat":1,"cycs":0,"mots":0,"hold":0,"momo":4,"coor":2,"plan":0,"unit":0,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,8]}
{"sr":{"stat":1,"hold":0,"momo":4,"coor":2,"plan":0,"dist":0,"frmo":1,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0}}
>`

karoria, 2019-03-12:
You need to compile and flash g2core as per your machine settings which will remain persistant then. Because due uses flash memory instead of eeprom to save settings. There is no issue of CNCjs here.

MitchBradley, 2019-03-12:
Adding to what @karoria said: The settings for g2core are in the g2core source tree in a file named settings/settings-MACHINENAME.h .  The g2core source contains predefined settings files for different machines .  To change settings you must either edit the file that most closely matches your machine, or else make a new one.  Then you recompile the G2 source with a command line that refers to your settings file.  It's more cumbersome that GRBL, which has persistent settings, but also much much more adaptable to different CPUs and machines.
I looked at the Fabmo area on github.  It has a G2 fork "FabMo-G2-Core" which is 6 years old, almost 1800 commits behind the latest G2.  It was not clear which machine settings file they use.  I suppose it is possible that fork is stale and the FabMo project uses a more recent G2 core, but I wasn't able to determine from a quick look.
Maybe someone in the FabMo community could point you in the right direction.  As @karoria said, this is really not a cncjs issue.  It might be possible to work around the problem imperfectly by creating a cncjs macro that you run manually every time you reset your g2 firmware, but the best solution is to recompile g2.

MitchBradley, 2019-03-12:
Oh, I just noticed from the startup message that the settings file is settings_shopbot_sbv300.h , and the firmware version is pretty recent.  The fbs string 100.26-313-geac2c tells us the the build was based on version 100.26 - a fairly good version from last April - to which 313 commits have been added.  The final commit eac2c is not present in the synthetos source, nor can I find it in any of the FabMo branches, so I don't know where to find the exact source from which your binary was built.
If you can find the source tree, it should be possible to edit the file g2core/settings/settings_shopbot_sbv300.h and change lines like

`#define X_VELOCITY_MAX              (360 * 25.4)    // xvm  G0 max velocity in mm/min
`

The recompile with

`make CONFIG=sbv300`

SketchThis, 2019-03-13:
Ah, this is making a lot of sense now, I really appreciate the explanation. I wasn't entirely sure where the "fault" was.

So the default speed, acceleration and other settings are actually on the G2 board, so what am I changing when I change those in CNCjs? Am I altering them on the G2 board temporarily, or is CNCjs sending different speed commands until I reload it?

So it looks like I'm going to have to learn to recompile and flash the G2 board if I want to permanently change the settings. I wonder how tricky that'll be given that this board is a ShopBot board (although it's got the same Atmega chip as the Due...)

karoria, 2019-03-13:
When you send command for parameter change to g2, it alters it temporarily until the power off. I hope now you will be more clear. Follow the process mentioned by @MitchBradley and believe me, you will not regret using g2core+CNCjs. Once you are familier with full CNCjs interface, I recommend using CNCjs-shopfloor-tablet developed by @MitchBradley.
One more thing, I am from Mechanical background and still I have managed to compile and flash g2core as per my needs. So don't worry. It is easier for you. Just go ahead.

MitchBradley, 2019-03-13:
Here are some steps/issues:

- Get the right source code.  As I mentioned, the version you are running has 313 commits on top of the version from the synthetos repository. Since we don't know what they are for, it would be prudent to get the right source, in case there are important customizations related to your hardware.  In particular, there is an additional machine customization file (in addition to the settings file we talked about) that defines the correspondence between pins on the Due chip and machine control signals (step/direction/enable for various axes, spindle controls, limit switches, etc).  That file is likely to be special for the Shopbot hardware, even if it uses a "standard" gShield board.  So you will need to ask someone in Shopbot land where to find their customized source.
- Choose a compilation machine.  MacOS and Linux on a PC are easy.  On Windows, you either need a Linux VM or Atmel Studio.  Linux VMs can be a pain to install because you have to work through a lot of details related to managing the VM software.  Atmel Studio is enormous, and while its installation is probably streamlined, I always worry that anything that big will end up taking me down a rathole. Personally, I compile in a Linux VM on Windows because I already have to use that for other purposes. I tried compiling on a Raspberry Pi but ran into a lot of problems and eventually decided it wasn't worth the effort to fix them.
- Download to the controller.  The details depend on how the Shopbot board is set up.  A stock Due has two different USB ports.  One is download-only, and the second is for serial interaction with the app (like cncjs) - but the other one can also be used for download under some circumstances.  To put the board into download mode, you sometimes need to do a little dance with reset switches.  It depends on how your board is set up.  And getting the download utility program to work just right in your setup can also be a head-scratcher.

MitchBradley, 2019-03-13:
I tried once again to compile on Raspberry Pi, and it worked.  I had to make a small patch to the g2core source.  It was slow but it worked.  I will tell the g2core people about it and see what they think.
It might be easier for  you to make a gcode file that has your special settings, and just run that file whenever you restart the machine.  I have a setup file that I run to prepare for my main production job anyway.  It does homing and probe and a few other prep things; it could easily override some feedrates too.  Maybe a workaround like that would get you to where you want to be.

MitchBradley, 2019-03-15:
Some more ideas:

- You can compile g2core in the cloud using Cloud9 (c9.io).  Once you have an account there, you can create a C++ workspace.  Starting that workspace takes a few seconds.  Then you use "git clone" inside the workspace to get the appropriate g2core source.  You can edit the settings file using the cloud 9 editor, then run make.  It is super fast - a make from scratch, which included the step of installing the cross compiler, took 1 minute 10 seconds.  Subsequent makes, with the compiler already installed, took 34 seconds.  You can download the binary file onto your home system to transfer to the Due.
- Alternatively (and cheton will probably hate me for this) - you can go into the cncjs subdirectory dist/cnc/app and edit the file "index.js" - skipping all of the node compilation steps.  Look for the string "send('{sr:n}');" then add additional send() lines to issue any special setup steps that you want to do when initializing the g2 controller.  Save those additions somewhere because you will have to redo the edit if you update to a new version of cncjs.  Hack!  Hack!

SketchThis, 2019-03-15:
This is fantastic info @MitchBradley. I really appreciate it a lot.

So I've been giving this a little more thought based on what I've learned, and here's what I've landed on:

The ShopBot G2 board is setup and configured for a ShopBot, and it works just fine the way it is. I only have one of their boards, and going along with the old adage "if it ain't broke don't fix it" I've decided that it's probably better for me to hack/learn on another board.

My ultimate goal is to do a conversion of an old Oliver CNC machine I have that currently has a terrible controller on it. As I mentioned earlier I'm comfortable with conversions, just not all this G2 stuff! :)

So, here's the plan:

I have an setup already that's got 4 drivers and a power supply. So all I need to do is deliver step and direction to those drivers and I'm good to go for motion. I have an Arduino Uno sitting unused. Now I know the Uno isn't a good choice for the machine permanently, but perhaps it's a good (cheap) choice to use to lean to flash G2 on to it and just use it to spin some steppers (remember I already have drivers, so all I need is the step and direction signals) so if I'm right I can get all that from an UNO?

This would mean that I would need to compile my own G2 file from their repository?

My other choice would be to just skip the UNO and buy a TinyG V8 board (which I'd likely be getting anyway for this build) and learn to modify the settings on that and flash it.

Starting with the UNO also has a side benefit in that I have another build that I want to make a simple 1 axis moving fence, and an UNO would likely be great for that...

karoria, 2019-03-15:
Oh... Here is the problem. You can't flash g2 on a Uno board. You need to get arduino due board for flashing g2. Due is super fast and has much more processing power which makes it more suitable for CNC.
Alternatively, you can flash grbl to existing Uno board and within 2 minutes you have your new cnc controller which is also supported by cncjs.

SketchThis, 2019-03-15:
That all makes sense now! I definitely want to experiment with G2 because of the super smooth motion. I think I'll skip the Due and go right for the Tiny G. Thanks to everyone here for the help.

MitchBradley, 2019-03-15:
Just to throw out another option - the moderator of the Taig Mill interest group on Facebook swears by UCCNC with an Ethernet based step/direction controller.  He makes some good points about the excellent support that is provided for the product and the quality of their user-focused UI.  I have a good friend who is rebuilding a large-ish mill and depending on me for the controller/UI work.  I was originally planning to steer him toward g2core + CNCjs (or my shopfloor-tablet UI), but I now think he would be better served by a commercial product like UCCNC.  Open Source stuff is great if you are a programmer with spare time, but if you are not, then you can find yourself with your machine down due to an upgrade that didn't go well.  The developers of commercial products have one huge advantage  - they have more control over the overall ecosystem (controller hardware, firmware, UI platform), so they can do far more end-to-end testing of a full range of system configurations.  And they have a financial incentive to do that testing as part of their full time job.

But back on the topic at hand  - if I were you I would go ahead and put GRBL on your Uno just as a baseline.  Even if you don't end up using it in the long run, you can still use it to test steppers.  The price is essentially zero and GRBL is easy to install and use, within its limitations.

SketchThis, 2019-03-15:
I'm totally with you on this. I actually use Centroid controllers on my large machine. Similar to UCCNC in that it's controlled by ethernet, and it's essentially a scaled down industrial controller so it's rock solid reliable and proven. The downside to Centroid (and it's not really a downside) is that you need a Windows computer to run it. For this little hobby conversion I'm doing it's overkill, and I'm actually looking for something to hack and tinker, and GRBL/G2 is perfect for this hobby machine and tinkering around! :)

MitchBradley, 2019-06-16:
Closing, since this discussion has wound down.  To make g2core settings persistent, you need to recompile, and the steps to do so are cited above.

closed 2019-06-16 by MitchBradley