cncjs/cncjs #428: g2core job kill

MitchBradley, 2019-03-17:
#### Description

With g2core, when you pause a job with feedhold, the spindle stops (correctly), but if you then decide to kill the job, the spindle restarts, leaving the spindle running when the machine is supposed to be idle.  Dangerous!
I think the root cause of this problem is a difference between the TinyG and g2core protocols.  g2core supports a "job kill" character - ^d - that TinyG does not have.  In src/app/controllers/TinyG/TinyGController.js, if the gcode:stop handler issues '!\x04' instead of '!%', the problem goes away.
But that probably does not work for TinyG.  I have opened an issue in the g2 area to ask for guidance on the right solution https://github.com/synthetos/g2/issues/404

cheton, 2019-03-18:
https://github.com/synthetos/g2/wiki/Feedhold,-Resume,-and-Other-Simple-Commands#jogging-using-feedhold-and-queue-flush

The documentation mentioned that these should each be on their own line. Will it help using separate line?
```js
this.writeln('!'); // feedhold
this.writeln('%'); // queue flush
```

cheton, 2019-03-18:
> I think the root cause of this problem is a difference between the TinyG and g2core protocols. g2core supports a "job kill" character - ^d - that TinyG does not have. In src/app/controllers/TinyG/TinyGController.js, if the gcode:stop handler issues '!\x04' instead of '!%', the problem goes away.

For g2core, I think it is good to issue `!` and `^d` separately for a program end. I will wait for the response of a right solution that cat support both TinyG and g2core.

---

I'm also wondering if I should change the <kbd>Reset</kbd> command from "^x" to "%d" for g2core?

https://github.com/cncjs/cncjs/blob/v1.9.17/src/app/controllers/TinyG/TinyGController.js#L1165-L1171
```js
'reset': () => {
    this.workflow.stop();
    this.feeder.reset();
    this.write('\x18'); // ^x
},
```

The reset command here is stand for a soft-reset, not hardware reset. I think I may have get it wrong for years.

MitchBradley, 2019-03-18:
I tried ! and % on separate lines.  That did not work.  The problem is that the semantics of % do not include stopping the spindle.

My current solution of issuing !^d in one writeln works just fine.  I'm not sure that the ! part is necessary; just doing ^d might work.  ! is good for feedhold that can be resumed later.

^x is correct for reset.  I use it when I want to make sure that the controller has completely restarted.  Please don't change it to ^d.  ^d is for canceling a GCode file run.  It is like a hard, immediate M30

I also have a hardware reset through a GPIO, but I rarely if ever need to use it.

MitchBradley, 2019-03-18:
After more testing, I think the best way to do gcode:stop for g2core is simply to issue ^d , no need for ! at all.  So the question of whether to do them on separate lines goes away, since you only have to issue the one character ^d
The delayed {clear:null} is also unnecessary.  The sole action of {clear:null} is to transition from either ALARM state to PROGRAM_STOP state, or from SHUTDOWN state to READY state.  It has no effect in any other state.  Since ^d causes an immediate transition to PROGRAM_END state, issuing {clear:null} after ^d (or even after ! or !%) is redundant.

cheton, 2019-04-14:
@MitchBradley

The release note stated that the Job Kill (^d) command was added in firmware build 101 `{fb:101.xx}`.
https://github.com/synthetos/g2/releases/tag/101.02

Should I make the change only for firmware build 101 `{fb:101.xx}` and later releases?

MitchBradley, 2019-04-14:
@cheton 
The g2core command parser has supported ^d since at least version 100.00, and probably even before, since the code appeared in development at version 079.  It is hard to tell when that job kill function actually started working correctly.
Your patch that restricts it to versions >=101 looks good, except that I don't think you need to issue M30 after ^d.

cheton, 2019-04-22:
Fixed in [1.9.19](https://github.com/cncjs/cncjs/releases/tag/v1.9.19)

closed 2019-04-22 by cheton