cncjs/cncjs #430: How to debug a macro

chuyskywalker, 2019-03-19:
*Description*:

Looking for information about how Macros are evaluated and how one could debug them when they are going wrong.

|--|--|
| Version | cncjs desktop osx 1.19.17 |
| CNC Controller | Grbl |
| Hardware | Laptop |
| OS | OSX |

*Details*

I have a macro I'm trying to suss out that should be able to find the center for WCS setting by probing three points within a circle cutout. However, when I run this, I eventually get something like:

```
G0 XNaN YNaN
```

which, as I have some passing familiarity with Javascript, is JSpeak for "you did a bad thing with numbers and I'm confused".

However, when I plug all these values into a JS console in my browser, it all maths out just fine.

So, while I'd love someone to say "Oy, line 54 has error X on it" to fix my macro, what I think might be more helpful is: How does one debug macros? How are the evaluated?

_(Based on an answer in #250 it looks like the data in between brackets `[]` is just, like, `eval()`ed? I couldn't quite figure out where the codebase does this to understand it...)_

Here's the macro in question:

```
;Start with probe in hole, BELOW Z surface

; Wait until the planner queue is empty
%wait

; Set user-defined variables
%PROBE_DISTANCE = 20
%PROBE_FEEDRATE_A = 150
%PROBE_FEEDRATE_B = 50

; for restoration at the end
%UNITS=modal.units
%DISTANCE=modal.distance

G91 ; Relative positioning
G21 ; Use millimeters

G10 L20 X0 Y0 ; set current position as xy zero for now

; Probe rear, slight right
G38.2 X1 Y[PROBE_DISTANCE] F[PROBE_FEEDRATE_A]
G1 Y-2 F[PROBE_FEEDRATE_B] ; back off a bit
G38.2 Y3 F[PROBE_FEEDRATE_B] ; probe again, slowly
%p1x = [posx] ; set the x/y coords for later
%p1y = [posy]

; set back to absolute mode and return to original spot
G90 
G0 X0 Y0
G91

; Probe 45deg
G38.2 X[PROBE_DISTANCE] Y[PROBE_DISTANCE] F[PROBE_FEEDRATE_A]
G1 X-2 Y-2 F[PROBE_FEEDRATE_B] ; back off a bit
G38.2 X3 Y3 F[PROBE_FEEDRATE_B] ; probe again, slowly
%p2x = [posx] ; set the x/y coords for later
%p2y = [posy]

; set back to absolute mode and return to original spot
G90 
G0 X0 Y0
G91

; Probe slight rear, far right
G38.2 X[PROBE_DISTANCE] Y1 F[PROBE_FEEDRATE_A]
G1 X-2 F[PROBE_FEEDRATE_B] ; back off a bit
G38.2 X3 F[PROBE_FEEDRATE_B] ; probe again, slowly
%p3x = [posx] ; set the x/y coords for later
%p3y = [posy]

; Figure out proper center https://stackoverflow.com/a/30106470 and set it
%ma = [(p2y - p1y) / (p2x - p1x)]
%mb = [(p3y - p2y) / (p3x - p2x)]
%cx = [(ma * mb * (p1y - p3y) + mb * (p1x + p2x) - ma * (p2x + p3x)) / (2 * (mb - ma))]
%cy = [(-1 / ma) * (cx - (p1x + p2x) * 0.5) + (p1y + p2y) * 0.5]

; Return to origin, move to new center, set center
G90 
G0 X0 Y0
G91
G0 X[cx] Y[cy]
%wait
G10 L20 X0 Y0

[UNITS] [DISTANCE] ;restore unit and distance modal state
```

cheton, 2019-03-19:
Each line prefixed with `%` is a JavaScript statement, you don't need to wrap expression with square brackets `[ ]`, or the value will be treated as an array.

Try to remove all square brackets from your macro, as shown below:
```
%p1x = posx
%p1y = posy
%p2x = posx
%p2y = posy
%p3x = posx
%p3y = posy
%ma = (p2y - p1y) / (p2x - p1x)
%mb = (p3y - p2y) / (p3x - p2x)
%cx = (ma * mb * (p1y - p3y) + mb * (p1x + p2x) - ma * (p2x + p3x)) / (2 * (mb - ma))
%cy = (-1 / ma) * (cx - (p1x + p2x) * 0.5) + (p1y + p2y) * 0.5
```

cheton, 2019-03-19:
To avoid latency when getting work position, I suggest you add a dell time to make sure the planner queue is empty and the machine is in idle state:

```
; A dwell time of one second
G4 P1
%p1x = posx
%p1y = posy
```

```
; A dwell time of one second
G4 P1
%p2x = posx
%p2y = posy
```

```
; A dwell time of one second
G4 P1
%p3x = posx
%p3y = posy
```

chuyskywalker, 2019-03-19:
> Each line prefixed with % is a JavaScript statement

Ahh, ok, thanks. I had been cribbing from some of @neilferreri 's work on [hole centering](https://github.com/neilferreri/CNCjs-Macros/blob/master/Hole_Center) where there's a lot of:

```
G38.2 Y-5 F[PROBE_FEEDRATE_B] ;Slow Probe
%Y_BTM = [posy]
%Y_CHORD = [Y_TOP - Y_BTM]
G0 Y[Y_CHORD/2]
%Y_CENTER = [posy]
```

And thought that the `[]`'s indicated "do javascript within these".

> To avoid latency when getting work position, I suggest you add a dwell time to make sure the planner queue is empty and the machine is in idle state:

Is there a reason to use that as opposed to `%wait`?

neilferreri, 2019-03-19:
@chuyskywalker Ha!  I just started removing some brackets after I saw this thread.

The %wait is just a dwell of 0.5s

chuyskywalker, 2019-03-20:
I did finally find that in the code, and yup, it's just a convenient helper. Thanks, I'll give all this a shot tonight.

chuyskywalker, 2019-03-20:
[Worked like a charm](https://www.dropbox.com/s/7xse5awz6q38n28/2019-03-19%2019.00.16.mov?dl=0), thanks!

closed 2019-03-20 by chuyskywalker

cheton, 2019-03-20:
Hi @neilferreri 

May I know your interest if you are willing to be the organization member of CNCjs? You can transfer your repo https://github.com/neilferreri/CNCjs-Macros to the CNCjs organization to increase visibility. The old link will still exist after the transfer.

Thank you for the awesome macros.

neilferreri, 2019-03-20:
Hi @cheton,
I'm not sure what that would entail.  I'm not experienced with git.  I'm happy to share though.  how would I move the repo?
Thanks

cheton, 2019-03-20:
Hi @neilferreri,

You don't need to be an experienced Git user, you can update content using the web-based interface provided by GitHub. Furthermore, it is very important to provide comprehensive user documentation including turorials, user guides. That's why I need your support to make up for my shortage.

An invitation letter was sent, you can check your mailbox and follow the link to join the CNCjs organization.

cheton, 2019-03-20:
Click the **Settings** tab on your repo https://github.com/neilferreri/CNCjs-Macros, you will find **Transfer ownership** at the bottom. You can enter new organization name to complete the transfer, as shown below:

[image]

cheton, 2019-03-20:
Hi @chuyskywalker 

There is a technique that can use inline comment to troubleshoot, it allows you print the value of a variable in the console:

```
%X0=posx
(X0=[X0]) ; Print the value in the inline comment
G4 P0 (X0=[X0]) ; Print the value in the inline comment right after a G4 dwell command
```

See more examples at https://github.com/cncjs/cncjs/wiki/User-Guide#examples

chuyskywalker, 2019-03-21:
Hey, that example code looks familiar! Jokes aside, very cool and should help a lot in further runs of things, thanks :)