cncjs/cncjs #431: Login from different Client

juup, 2019-03-19:
Hi,
my cncjs 1.9.16 is running on a Laptop and woks fine, Thanks for your work.
Now I want to check the status of process from another PC. I hoped that the option _--allow-remote-access_ is what I need. But when I log in with same account from another PC all fields are empty and I can't see the status of machine / process.
Is the option the right one? How should I configure it?
Regards
juup

cheton, 2019-03-20:
> But when I log in with same account from another PC all fields are empty and I can't see the status of machine / process.

I didn't quite get your point as you mentioned. Did you click **Open** to create connection on another PC? Providing a screenshot will be useful to address what was wrong on the UI.

As for the `--allow-remote-access` option, it is used to bypass remote connection from public IP addresses. By default, your can only connect to the server within a private network for security reason. See below for the default whitelist:
```js
// https://github.com/cncjs/cncjs/blob/master/src/app/access-control.js
const whitelist = [
    // IPv4 reserved space
    '127.0.0.0/8', // Used for loopback addresses to the local host
    '10.0.0.0/8', // Used for local communications within a private network
    '172.16.0.0/12', // Used for local communications within a private network
    '192.168.0.0/16', // Used for local communications within a private network
    '169.254.0.0/16', // Link-local address

    // IPv4 mapped IPv6 address
    '::ffff:10.0.0.0/8',
    '::ffff:127.0.0.0/8',
    '::ffff:172.16.0.0/12',
    '::ffff:192.168.0.0/16',

    // IPv6 reserved space
    '::1/128', // loopback address to the local host
    'fc00::/7', // Unique local address
    'fe80::/10' // Link-local address
];
```

juup, 2019-03-20:
Thanks cheton for your help and clear the option `--allow-remote-access.`
May I get you wrong, but **Open** is for a connection to the CNC (Arduino via USB in my case), right?
I want to have the Laptop be connected to the CNC and run the program. I just want to monitor it from another place / room.
[image]

cheton, 2019-03-20:
In another PC, you still have to choose the connected port and click Open to establish a WebSocket connection. The first established connection on your laptop will create a serial port connection to your Arduino, all subsequent connections from other computers will use the same serial port.

While sending a G-code program, close the browser on your laptop won't stop the movement, you can safely close the browser on your laptop, and monitor the progress from another browser.

juup, 2019-03-20:
Great, that's it.
I've never tried to select a port, because I thought it's already done for the Laptop.
Thank you

closed 2019-03-20 by juup

cheton, 2019-03-20:
That was exactly a lack of documentation that needs to be improved, including tutorials and user guides.