cncjs/cncjs #432: how to use cncjs with modified grbl?

erikparr, 2019-03-19:
Hi is it possible to run cncjs with a modified version of grbl? I want to run it with a [servo enabled version of grbl ](https://github.com/bdring/Grbl_Pen_Servo) that basically a normal grbl but with a modified spindle_control.c file. But I don't know how that owuld work here as cncjs seems to have its own js version of grbl?  Much thanks for any help

saxa, 2019-03-19:
I think would be not a problem. But I amalso interested in this servo version of the GRBL. Since CNCjs is mainl a Gcode feeder and if the commands GRBL servo receives are the same as the ones standard GRBL have there should be no problems I think.

MitchBradley, 2019-03-19:
cncjs does not have its own version of grbl.  cncjs just sends gcode to an ordinary grbl installation that is running on an Arduino.  The Javascript code in cncjs for grbl - and similar modules for tinyg/g2core, smoothieware, and Marlin - knows how to handle grbl's serial line protocol and some other idiosyncrasies that differ between those various controllers, but does not replace the function of grbl itself.
Therefore, cncjs should work just fine with that modified grbl.  You will just have to send the appropriate GCodes to do what you want.  cncjs just passes through gcodes to grbl.

erikparr, 2019-03-19:
thanks that makes sense