cncjs/cncjs #434: Trying to update Pendant for Marlin

mattr254, 2019-03-23:
#### Description

I'm trying to update Pendant to work with Marlin, I was able to get the XYZ jog advance to update by updating the code from G90, G91 G0 (move) G90 to G91 G0 (Move)G90. So I was able to get that to work.
I have a few issues that I'm trying to figure out. 
1 How does Pendant tell CNCjs the controller type, I noticed an error for a service:cncengine when I selected null, is that it? Does it recognize the word 'Marlin' 
2 This would be a real pipe dream, how would I get the position updates from the server?

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller
- [X] MARLIN
- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-03-23:
The controller type is determined by <pendant>/src/app.js starting at line 39.  Here is an excerpt:

```
    cnc.controllerType = controllerType;
    cnc.port = port;
    cnc.baudrate = baudrate;

    $('[data-route="workspace"] [data-name="port"]').val(port);

    Cookies.set('cnc.controllerType', controllerType);
    Cookies.set('cnc.port', port);
    Cookies.set('cnc.baudrate', baudrate);

    if (controllerType === 'Grbl') {

```

In src/controller.js there is a list of callbacks for "System Events".  That list includes events like 'Grbl:state', 'Smoothie:state', etc.  Just below that there are lines like:

```
            if (eventName === 'Grbl:state') {
                this.type = GRBL;
                this.state = args[0];
            }

```

You will need to add corresponding entries for Marlin.

 src/app.js contains the code that handles those events.  For example:

```
controller.on('Grbl:state', function(data) {
    // If we do not yet know the reporting units from the $13 setting, we copy
    // the data for later processing when we do know.
    if (typeof grblReportingUnits === 'undefined') {
        savedGrblState = JSON.parse(JSON.stringify(data));
    } else {
        renderGrblState(data);
    }
});
```

To determine the machine position, you can look at the data that is the argument to the Marlin:state event.  Based on my reading of the cncjs code, that data will have a "pos" property that contains subproperties like x, y, z, and e.  So data.pos.x should be the x position.

One easy way to determine exactly what is in that data object is to use Developer Tools in the Chrome browser.  Browse to the pendant page, fire up Developer Tools, then set a breakpoint on your function that handles the Marlin:state event.  Then you can hover over the identifier 'data' and developer tools will show you what is inside it.

As a matter of good coding practice, it is best not to assume that all properties will always be present.  You might get some Marlin:state reports that contain only a subset of possible properties, so it is good to write the code so it does the right thing with only the properties that are currently present.

mattr254, 2019-03-24:
I can't wrap my head around it, Ive updated the controller.js to include new constraints 
var MARLIN = 'Marlin';
added new call backs
'Marlin:settings': [],
'Marlin:settings': [],
and even added events
 if (eventName === 'Marlin:state') {
                this.type = MARLIN;
                this.state = args[0];
            }
            if (eventName === 'Marlin:settings') {
                this.type = MARLIN;
                this.settings = args[0];
The pendant still sees the device as "Not Connected"
getting past that 
mpos.x even trying to plug it into a text box I get null data
I don't really understand how apps.js is calling the position from the server I look into the Marlin controller on CNCjs and see that it identifies each axis by posx , posy etc
I see how it determines the location by

if (this.query.type === QUERY_TYPE_POSITION) {
                this.connection.write('M114\n', {
source: WRITE_SOURCE_SERVER
I just need to understand how to pull data from CNCjs,

MitchBradley, 2019-03-24:
You don't "pull" data from CNCjs.  CNCjs sends events with the data.

The first notification you get is from "controller.on('serialport:open', function(options) {" with cnc.controllerType == 'Marlin'

Later you get notifications from "controller.on('Marlin:state'), function(data) {"

mattr254, 2019-03-27:
So I've learned a lot and I've even made some progress on the issue. My first discovery is the console of the inspector tool. This has helped loads. Right off the bat I had 2 status instead of 1 status and one settings. 
Moving forward I got to see the server and the terminal interact with each other. 
So food for thought controller.writeln('M114') sends Gcode M114 to the server which sends me a reply of coordinates. 
I'm using controller.on('serialport:read', function(data) {  to read the next line and start to break down the string into pieces that I can graft to machine Positions. I'm still trying to work on splitting the string and not save the whole thing.
 I wanted to thank you for reaching out and helping me with the basics once I figure this out I would like to post to pendant for some of the Marlin entrenched people such as my self. 
controller.on('serialport:read', function(data) {
	console.log(data);
	if (data.length > 3){
		var myXpos = data.split(' ');
		console.log(myXpos[0]);
		
	}
It's a work in progress but it's the first major break I've had !

MitchBradley, 2019-03-27:
Using serialport:read is the wrong approach.  The cncjs server already parses the serialport result for you and, based on a timer, issues a Marlin:state event containing the position data.

As I mentioned earlier, I think you need to respond to the "Marlin:state" event in addition to "Marlin:settings".  There is no "Marlin:status" event  - the ":status" events are  "feeder:status" and "sender:status", which report the progress of GCode transmission.

"settings" are things that rarely change, like the firmware version and the machine type.  "status" is the progress of sending a batch of GCode.  "state" is the timely machine state, including the position and modes.

mattr254, 2019-03-27:
Yes apologies, Marlin:state not status I'm just now running into that headache, apparently console does not display the Gcode movements during operation.  
I'll keep plugging at it! Thanks Again Mitch!

MitchBradley, 2019-03-27:
The console suppresses routine status reports from the controller because they would cause endless scrolling which would obscure the other stuff in the console.

MitchBradley, 2019-03-27:
If you start the server with enough -v's, it will log position events to standard output on the computer that runs the server.  From <cncjs>/src/app/controllers/MarlinController.js:

```
       this.runner.on('pos', (res) => {
            log.silly(`controller.on('pos'): source=${this.history.writeSource}, line=${JSON.stringify(this.history.w\
riteLine)}, res=${JSON.stringify(res)}`);

```

mattr254, 2019-03-27:
I was able to get it to work after playing with console.log I was able to discover data.pos.x and so on,  plugging those in directly I was able to get the position to update! I've uploaded a Patch to Pendant with the modified changes.

MitchBradley, 2019-03-27:
Where is your patch?  I would like to look at it.

mattr254, 2019-03-27:
I added it to https://github.com/mattr254/cncjs-pendant-tinyweb/tree/patch-1/src

johnboiles, 2019-04-14:
I've also got a branch started at https://github.com/johnboiles/cncjs-pendant-tinyweb/tree/marlin-support

johnboiles, 2019-04-22:
Try out the latest `master` branch of the tiny web pendant. I’ve committed a few marlin fixes including removing or rewiring a few of the buttons that were unsupported in Marlin