cncjs/cncjs #435: Laser connected but not "toggling" when running .nc file

JalenTk, 2019-03-27:
#### Description

I'm currently running CNC.js on my laser engraver connected to arduino uno with cnc shield. The problem is the laser is continuosly on when running, meaning there isn't any "toggling" and "pauses". 

I previously run my laser engraver on Benbox with their given hex file, and it works fine. 
However the issue occured when I turned to CNC.js, running on grbl 1.1f hex. With different version of hex file, the laser is connected to "SPINDLE ENABLE" on the Benbox hex file, while "Z+" on the grbl 1.1f.

I can actually control the "Laser Test" and "Laser Off" on CNC.js, but it just won't do the toggling when running the .nc file. What can I do to fix this issue? 

Thank you in advanced!  

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-03-27:
On my laser setup, the laser is connected to spindle enable.

JalenTk, 2019-03-27:
I tried to use the SPINDLE ENABLE pin, but it doesn't work for me

MitchBradley, 2019-03-27:
I can't help you debug without more information.  "doesn't work" is too vague.

JalenTk, 2019-03-29:
[image]
[image]

I’m running 2 stepper motors and a laser module on an arduino uno with CNC shield.

Regarding to the “toggling and pauses”, I’m actually refering to the event when there’s gap between 2 points during engraving, the laser would normally turned off during the gap and toggle on once it reaches the other point.

In my case, the laser would just stay on/off entirely, which it engraved along the path moving with the stepper motors, without “blinking”

MitchBradley, 2019-03-29:
This is not exactly a cncjs issue because the same thing will happen no matter what gcode sender you use.

Here are some things to check:

- If you issue the GCode line "M3 S255" the laser should go on
- "M5" should make it go off
- GRBL laser mode should be on, i.e. $32=1
- With laser mode on, if you use M4 instead of M3, the laser will turn off during G0 rapid moves and on during G1/G2/G3 feedrate moves.  That is probably what you want - it is what I use.  To test this, you can issue M4 and the laser will stay off, but when you then do a move like "G1 X10 F400", the laser should turn no during the move, and then back off when the motion stops.
- Your gcode program needs to be written accordingly, i.e. it should enable the laser with M4 instead of M3, and leave it enabled during a sequence of G0 rapid (non-burning) moves and G1/G2/G3 feedrate (burning) moves. This is the best way because, firstly, the laser will automatically turn off when motion stops so you don't accidentally burn something, and secondly, GRBL is good at automatically managing the laser power so corners do not burn too much.
- Alternatively, your gcode program can issue explicit M3 and M5 commands to turn the laser on and off at the appropriate times, perhaps with delays to let the laser come on.  I used to do it that way, but I get better results with M4.

Here is a thread that tells how the hardware and GRBL must be configured:

https://github.com/arkypita/LaserGRBL/issues/368

cheton, 2019-04-20:
Hi @JalenTk 

@MitchBradley's explanation is very clear. Grbl laser mode is also expalined in the [Grbl wiki page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode).

I will close this issue because it's not quite related to CNCjs. You can still continue the discussion here.

closed 2019-04-20 by cheton