cncjs/cncjs #441: Support for fully closed loop industrial servo motors

karoria, 2019-04-02:
#### Description

I want to make a VMC machine with fully closed loop system using industrial servo. At present I believe CNCjs + g2Core doesn't give a fully closed loop experience. However, I feel with some workaround in CNCjs, we may achieve this.

I have studied siemens and panasonic servo drive manuals and they both support pulse + direction inputs, so no worry about driving them. The only concern is taking its position's realtime feedback to CNCjs or to controller and work accordingly. I mean the MPos and WPos of all the axis to be displayed as per the encoder's reading (here comes interface) and not just by sent pulses. This will make any machine more safe and more reliable to operate. I strongly believe, if implemented this feature, one can make professional/commercial machines instead of hobby machines. My idea is to use more common interface which is available in almost all the industrial servos to make its compatibility more accepted.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

scottlaird, 2019-04-02:
One thing you can do that is easy and doesn't require any software support is to wire your servo drive's error output into your CNC controller's halt/error/estop input.  So, if any axis encounters enough of a problem that it can't catch up and falls out of whatever the defined error range is, then it'll end up aborting the job.  Generally by that point things are pretty much a lost cause anyway.

If you haven't aborted, then your actual location is within some controllable error margin of where your controller told it to be. 

I have closed-loop steppers (Leadshine CS-M22331-L) on my Grbl CNC (a Beaver HDZero), and I've tripped axis faults *many* times while building it.

My motors have relative encoders, so I still need limit switches and a homing cycle.  With absolute encoders things would be somewhat different; I'm not sure if anything cheap has support for that.

saxa, 2019-04-02:
Hi I have 2 servos from DMM Tech with absolute encoders, those I want to use on my cnc gantry retrofit.
I will probably have to go with a controller other than grbl and cncjs as @MitchBradley suggested. But I am still tempted to try it with servo GRBL version and CNCjs. Any suggestions on how to connect the Dyn4 servo drives ?

karoria, 2019-04-03:
Thanks @scottlaird. Your suggestion is pretty straightforward but will make a semi close loop system. However, i want to try it. I want to know where to give halt/error/estop input to g2Core.
Further, can anybody guide me for taking a servo position feedback by profinet or RTU Modbus interface to cncjs or g2core for the machine coordinates? I know it is difficult and may require additional coding in cncjs or g2core.

MitchBradley, 2019-04-03:
For the alarm function on g2core, you need to select a digital input that is not already used for something else and, in the settings file, configure it for INPUT_ACTION_ALARM.
For servo feedback, this g2core issue https://github.com/synthetos/g2/issues/361 reports work that someone has done.  It was the first hit in a search for "g2core servo".  Use the search engine, Luke.

karoria, 2019-04-03:
Thanks Mitch. I see several inputs in settings file are unused. But how to know that DI represents which pin on due board? I also gone through g2core thread you referenced but it seems the work is still immature and some superbrain is required good efforts to get results. I will go by semi close loop by just alarm feedback as of now.
Pls describe me the process of digital input and tracking its physical pin on the board.