cncjs/cncjs #447: Is there a full set of hotkeys for Windows desktop version?

funbobby2001, 2019-04-10:
#### Description

I want to use a wireless keyboard to jog and shuttle the machine... I see the keypad jogging hotkeys in the widget, but I'd like access to more functionality like all the hotkeys listed for Mac in the documentation. Do these exist?

#### Versions

- CNCjs: 1.9.18
- Node.js: 
- NPM: 

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Marlin
- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-04-21:
@funbobby2001 

You can find all hot key definitions through the following places:
* https://github.com/cncjs/cncjs/wiki/User-Guide#keyboard-shortcuts
* https://github.com/cncjs/cncjs/blob/v1.9.18/src/web/lib/combokeys.js

closed 2019-04-21 by cheton

funbobby2001, 2019-04-21:
> @funbobby2001
> 
> You can find all hot key definitions through the following places:
> 
> * https://github.com/cncjs/cncjs/wiki/User-Guide#keyboard-shortcuts
> * https://github.com/cncjs/cncjs/blob/v1.9.18/src/web/lib/combokeys.js

@cheton 

Unless I'm missing something, those are for Mac and none of the combos that use the "command" key can be used in Windows to my knowledge.

cheton, 2019-04-21:
https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/metaKey

On Macintosh keyboards, this is the Command key <kbd>⌘</kbd>.
On Windows keyboards, this is the Windows key <kbd>⊞</kbd>.

funbobby2001, 2019-04-22:
Thanks, that works. I tried it before but I must have been doing something else wrong.