cncjs/cncjs #448: how can I compile and generate .exe files?

hupf0211, 2019-04-12:
#### Description

I am a little white, how can I compile and generate .exe files?

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-04-18:
I'm not an experienced Windows developer, the following procedures may help:

* Install [Git For Windows](https://gitforwindows.org/)

* Install [Python 2.7.16](https://www.python.org/ftp/python/2.7.16/python-2.7.16.amd64.msi) - Python 3 is not compatible with the build process
  Be sure to check **Add python.exe to Path**
[image]

* Install [Node.js v10](https://nodejs.org/en/download/)

* Install Windows Build Tools under **PowerShell**. You need to right-click on **PowerShell** and run "as Administrator".

    ```sh
    npm install --global --production windows-build-tools
    ```

    [image]

    [image]

* Launch **Git Bash** here (do not use **Windows Command Prompt** or **Windows PowerShell**). Under **Git Bash**, clone the repo and build the code
    ```sh
    $ git clone https://github.com/cncjs/cncjs.git
    $ cd cncjs
    $ npm i
    ```
* Make a Windows x64 installer
    ```sh
    $ npm run prepare
    $ npm run build:win-x64
    ```

    [image]

```
$ npm run build:win-x64

> cncjs@1.9.18 build:win-x64 D:\GitHub\cncjs
> bash -c 'scripts/electron-builder.sh --win --x64'

/d/GitHub/cncjs/dist/cnc /d/GitHub/cncjs
Cleaning up "/d/GitHub/cncjs/dist/cnc/node_modules"
Installing packages...

> serialport@6.2.2 install D:\GitHub\cncjs\dist\cnc\node_modules\serialport
> prebuild-install || node-gyp rebuild

added 529 packages from 292 contributors and audited 1395 packages in 40.654s
found 9 vulnerabilities (3 low, 5 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
audited 1395 packages in 2.994s
found 9 vulnerabilities (3 low, 5 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
/d/GitHub/cncjs
Rebuild native modules using electron
v4.1.4

> cncjs@1.9.18 electron-rebuild D:\GitHub\cncjs
> electron-rebuild "--version=" "v4.1.4" "--module-dir=dist/cnc" "--which-module=serialport"

- Searching dependency tree
√ Rebuild Complete

> cncjs@1.9.18 electron-builder D:\GitHub\cncjs
> build "--win" "--x64"

Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
  • electron-builder version=20.39.0
  • loaded configuration file=package.json ("build" field)
  • rebuilding native production dependencies platform=win32 arch=x64
  • packaging       platform=win32 arch=x64 electron=4.1.4 appOutDir=output\win-unpacked
  • asar using is disabled — it is strongly not recommended solution=enable asar and use asarUnpack to unpack files that must be externally available
  • downloading               parts=1 size=5.6 MB url=https://github.com/electron-userland/electron-builder-binaries/releases/download/winCodeSign-2.4.0/winCodeSign-2.4.0.7z
  • downloaded                duration=4.564s url=https://github.com/electron-userland/electron-builder-binaries/releases/download/winCodeSign-2.4.0/winCodeSign-2.4.0.7z
  • building        target=nsis file=output\CNCjs Setup 1.9.18.exe archs=x64 oneClick=true perMachine=false
  • downloading               parts=1 size=1.4 MB url=https://github.com/electron-userland/electron-builder-binaries/releases/download/nsis-3.0.3.2/nsis-3.0.3.2.7z
  • downloaded                duration=3.861s url=https://github.com/electron-userland/electron-builder-binaries/releases/download/nsis-3.0.3.2/nsis-3.0.3.2.7z
  • downloading               parts=1 size=1.0 MB url=https://github.com/electron-userland/electron-builder-binaries/releases/download/nsis-resources-3.3.0/nsis-resources-3.3.0.7z
  • downloaded                duration=4.651s url=https://github.com/electron-userland/electron-builder-binaries/releases/download/nsis-resources-3.3.0/nsis-resources-3.3.0.7z
  • building block map blockMapFile=output\CNCjs Setup 1.9.18.exe.blockmap

Cheton Wu@TW-CHETON-WIN10 MINGW64 /d/GitHub/cncjs (master)
$ ls -al output/
total 51853
drwxr-xr-x 1 Cheton Wu 197121        0 Apr 18 19:31  ./
drwxr-xr-x 1 Cheton Wu 197121        0 Apr 18 19:29  ../
-rwxr-xr-x 1 Cheton Wu 197121 53014176 Apr 18 19:31 'CNCjs Setup 1.9.18.exe'*
-rw-r--r-- 1 Cheton Wu 197121    56336 Apr 18 19:31 'CNCjs Setup 1.9.18.exe.blockmap'
-rw-r--r-- 1 Cheton Wu 197121      349 Apr 18 19:31  latest.yml
drwxr-xr-x 1 Cheton Wu 197121        0 Apr 18 19:29  win-unpacked/
```

* Open Windows Explorer to run the installer

    [image]

hupf0211, 2019-04-18:
Thank you very much, I will try to compile, if there is a problem, I will come back to ask you questions.
Thanks again.

cheton, 2019-04-18:
@hupf0211 The build process for Windows was not well-documented, I will update it to the wiki.

closed 2019-04-18 by cheton

MitchBradley, 2019-04-21:
I tried it.  It works.  I can also confirm that it does not work under PowerShell.  In that case, the electron build step fails with with a message saying "wine is required".  I tried drinking a glass of wine [image] which made me feel happier, but did not help the computer.

hupf0211, 2019-04-22:
Hi，I need help！When I enter the command ""npm i", the prompt is as follows：
[image]

cheton, 2019-04-22:
As stated in my comment, you have to use Git Bash rather than PowerShell:

> Launch Git Bash here (do not use Windows Command Prompt or Windows PowerShell). Under Git Bash, clone the repo and build the code

```
$ git clone https://github.com/cncjs/cncjs.git
$ cd cncjs
$ npm i
```

hupf0211, 2019-04-22:
Why does the program not move when it reaches this step?
[image]

cheton, 2019-04-22:
It took about 30 seconds for the step of "Searching dependency tree" from my side.

Try finding the line in package.json:

```
 "electron-rebuild": "electron-rebuild",
```

Then changed it to:

```
 "electron-rebuild": "cross-env DEBUG=electron-rebuild electron-rebuild",
```

This will output more information when rebuilding native modules.

hupf0211, 2019-04-23:
Successful! thank you very much.
[image]

venuasg, 2019-11-13:
Hi,
I get error when i try npm i 
[image]

MitchBradley, 2019-11-14:
That is the error you get when you have not installed the window build tools, including Visual Studio.  You need to run "npm install --global --production windows-build-tools" in an Administrator CMD or PowerShell

venuasg, 2019-11-15:
> That is the error you get when you have not installed the window build tools, including Visual Studio. You need to run "npm install --global --production windows-build-tools" in an Administrator CMD or PowerShell

Yes i have installed Windows Build tools and followed all the steps that were mentioned.

MitchBradley, 2019-11-15:
Have you tried using PowerShell or CMD to do the cncjs installation, instead of MINGW64? (as I suggested on the Facebook group?)
My best guess is that the VCINSTALLDIR environment variable, which tells where to find Visual Studio, did not propagate to the MINGW64 shell.  Another related possibility is that the path syntax isn't quite the same between "real" Windows shells and Unix-style shells.
There is no advantage to asking your question both here and on Facebook.  The same people monitor both.

venuasg, 2019-11-15:
> Have you tried using PowerShell or CMD to do the cncjs installation, instead of MINGW64? (as I suggested on the Facebook group?)
> My best guess is that the VCINSTALLDIR environment variable, which [...]

Now i tried with powershell, cmd and git bash, issue is same. i rechecked and installed windows build tools aswell.

MitchBradley, 2019-11-15:
https://stackoverflow.com/questions/58445330/node-pre-gyp-install-fallback-to-build has some suggestions.  Other than that, all I can think of is to uninstall everything - node, npm, build tools, etc - and try reinstalling from scratch, using only PowerShell.  I do most of my cncjs development under Linux so I don't understand the deep details of how the Windows build works.