cncjs/cncjs #453: Server has stopped after updating to 1.9.18

daniel-ex, 2019-04-14:
#### Description

Hi
I just upgraded from 1.9.14 to .18. And Server has stopped message keep popping up. Only for a second and then its working again. I havent run any jobs yet. Also, at some point I was jogging but the coordinates stayed 000. I havent had any problems with v .14
Thank you

#### Versions

- CNCjs: 1.9.18
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x ] Windows
- [ ] Mac
- [ x] Linux

johnboiles, 2019-04-15:
I also noticed the “Also, at some point I was jogging but the coordinates stayed 000” issue once or twice.

cheton, 2019-04-18:
@daniel-ex Can you take screenshots or a video to show the problem you encountered? Thanks.

bensuffolk, 2019-04-20:
I too have noticed this, also after it has happened it sometimes stops updating the position widget. You can still jog and the machine responds as expected, but the display never updates.

The only fix is to reload the web page so it starts working again.

bensuffolk, 2019-04-20:
Another thing related is when it’s in this state if you upload a new g code file it hangs with h spinning circle, never finishing the upload, but if you refresh the page it’s all working again and you can see the file has been uploaded and is in the preview window ready to go.

cheton, 2019-04-21:
It’s probably a client-side JavaScript error that stops updating UI. I will take much testing and see if it can be reproduced by my side.

bensuffolk, 2019-04-21:
Ok, I’ll open the JavaScript console and see if it’s showing any errors when it’s doing it.

cheton, 2019-04-21:
@bensuffolk 

Opening the Developer Tools while sending a G-code program (i.e. frequent updates for work position) may occasionally crash Chrome DevTools in recent CNCjs builds (1.9.1x). Even though it does have a rendering performance issue in the Axes widget, it should not cause the browser to crash, I'm not sure if it is a bug of Chrome DevTools. 

I haven't had time to track down more details, just to let you know the issue if you encountered a crash when opening the Developer Tools.

bensuffolk, 2019-04-21:
I actually use safari on Mac OS/X so hopefully won’t have any crashes. I shall keep you posted on anything I see.

bensuffolk, 2019-04-26:
Ok, so nothing in the javascript window during the time it's not updating. 

There are a couple of unrelated errors shown when you load the page, thats all.

THREE.WebGLRenderer - "94"
Three.MeshBasicMaterial: .shading has been removed. Use the boolean .flatShading instead.
Three.MeshBasicMaterial: .shading has been removed. Use the boolean .flatShading instead.

bensuffolk, 2019-04-26:
Issue 394 seems to be the same, and thats from version 1.9.15 so this has possibly been around for a few versions now. I had not used my CNC machine for 6 - 8 months or so, and did the upgrade. Unfortunately I can't remember what version I was on prior to the upgrade when it was all working, but I think as it was reported for 1.9.15 that might give us a clue to when it started.

daniel-ex, 2019-04-26:
One day the problem just stopped occurring. But then it happend a few times on an android phone. I'm starting to think that I have those issues whenever I use a wifi instead of a lan.

cheton, 2019-04-26:
@bensuffolk [CNCjs 1.9.20](https://github.com/cncjs/cncjs/releases/tag/v1.9.20) is the most recent release, it fixed a bug where the feeder queue may be blocked in certain condition. A detailed description can be found [here](https://github.com/cncjs/cncjs/pull/465).

If you found the UI stopped updating work position after jogging or running a macro, that may probably be caused by a bug in the feeder queue.

Please let me know if 1.9.20 can resolve your issue. Thank you.

bensuffolk, 2019-04-28:
Have upgraded to 1.9.20, will have a play and see if the issues are still there. Fingers crossed the feed queue blocking was the cause.

bensuffolk, 2019-05-04:
Looking good so far.