cncjs/cncjs #454: Slow GCode processing with Marlin

johnboiles, 2019-04-14:
#### Description

I've set up CNCJS running on a Pi2 connected to my MPCNC (Marlin). I'm noticing that with CNCJS (but not OctoPrint), the movements are much jerkier. It seems like there is a pause between processing lines of GCode sometimes. This is especially noticeable in long, smooth arcs.

I'm wondering if it's something about the timing of sending GCode to the controller. Perhaps OctoPrint pre-emptively sends the next few commands and the controller uses that to smooth motion between moves. Does anyone think this would improve with a faster Pi (Pi3)?

I'm happy to dig and attempt changes, but would appreciate some initial guidance.

I can post a side-by-side video later if that'd be helpful.

#### Versions

- CNCjs: 1.9.18 (f166744d420987c314274626dc751ac2392489ae)
- Node.js: v8.15.1
- NPM: 6.9.0

#### How Do You Install CNCjs?

- [x] NPM (built from source)
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [x] Raspberry Pi (Pi 2)
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2019-04-14:
I'm sort of speculating here because I don't have direct experience with Marlin and I have only studied  the code superficially.  That said, it appears that the base Marlin serial protocol is "send a line, wait for 'ok' ACK"  without a lot of buffering.  If that is the case, then the program that is sending to Marlin probably needs to have pretty good real-time response.  JavaScript isn't exactly optimized for real-time, and running it on a Pi only makes matters worse.
But I could be wrong...

johnboiles, 2019-04-14:
I’ll take a shot at comparing octoprint’s logic with the cncjs logic

jeffeb3, 2019-04-15:
Octoprint is written mostly in Python. I'm not sure if javascript would be any worse for real time. I run octoprint on a pi zero with no speed problems for gcode (the web ui is a little slower).

Unless there's something much worse about javascript or the way node.js run on arm, I doubt that is the issue.

johnboiles, 2019-04-15:
I made a comparison [video](https://www.youtube.com/watch?v=4-tBZmR1K_Q) of CNCJS (left) vs OctoPrint (right). OctoPrint is much smoother (though Repetier Host on my laptop is even smoother still!) This is running on my Raspberry Pi2. I have a Pi3B+ coming today, so I can run another comparison.

Looks like Marlin supports buffering a [number of moves](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L876) in advance (16 by default). I wonder if CNCJS takes advantage of this. Perhaps the `SP_TYPE_SEND_RESPONSE` used for Marlin in CNCJS only sends a single command at a time and `SP_TYPE_CHAR_COUNTING` used for Grbl streams ahead by a certain number of characters. @cheton does that seem plausible?

OctoPrint has [historically](https://github.com/foosel/OctoPrint/issues/450) had this issue as well and there wasn't a silver bullet for them. Seems like lots of digging and performance optimizations. I just want to make sure the basic stream-ahead is working for CNCJS.

I also noticed a vague [issue](https://github.com/cncjs/cncjs/issues/392) about TinyG planner buffer. TinyG also uses `SP_TYPE_SEND_RESPONSE`. Possible these are related.

cheton, 2019-04-15:
Hi @johnboiles 

What you mention is correct. Both Grbl and Smoothie controllers use the character counting protocol, which will give best performance with buffered motion commands. For TinyG/g2core controller, it should not have significant performance issue because it also [take advantage of the planner buffer](https://github.com/cncjs/cncjs/blob/v1.9.18/src/app/controllers/TinyG/TinyGController.js#L459-L472).

To be honest, I rarely had the chance to tune the performance with a connected machine running Marlin controller, if there is a chance I'd like to know how Octoprint implemented their buffering mechanism, that will definitely help for performance improvement.

cheton, 2019-04-15:
Before making the change to take the advantage of buffering solution, you can try commenting out `this.queryPosition()` and `this.queryTemperature()`, and check if it will affect the performance.

https://github.com/cncjs/cncjs/blob/v1.9.18/src/app/controllers/Marlin/MarlinController.js#L720-L724
```js
// M114: Get Current Position
this.queryPosition();

// M105: Report Temperatures
this.queryTemperature();
```

johnboiles, 2019-04-15:
I tried for a bit to understand how octoprint was doing it’s buffering/sending but couldnt figure out where the logic is to send commands in advance or control the rate of sending commands. Needs more study. @cheton can we simply use the character counting streaming protocol for marlin?

johnboiles, 2019-04-15:
@cheton I'll try removing `this.queryPosition()` and `this.queryTemperature()` and see if it makes a difference. OctoPrint never queries position, but I think it _must_ be querying temperature since it maintains temp graphs.

jeffeb3, 2019-04-15:
There is a checkmark in the settings in octoprint for "Support temperature autoreporting by firmware, if detected" and Marlin supports (if configured) M155, which will automatically broadcast the temperatures:

http://marlinfw.org/docs/gcode/M155.html

jeffeb3, 2019-04-15:
> Looks like Marlin supports buffering a number of moves in advance (16 by default). I wonder if CNCJS takes advantage of this.
>  That said, it appears that the base Marlin serial protocol is "send a line, wait for 'ok' ACK" without a lot of buffering

I don't remember where I saw this, but I thought Marlin responds with "ok" as soon as it's been stored in the buffer, not when it's done moving. So sending a command, waiting for "ok" and sending the next should be fine, as long as it's done faster than the machine can move. Needs confirmation though.

johnboiles, 2019-04-15:
@jeffeb3 thanks for your thoughts! Good to know about `M155` maybe that is more efficient than `M105` since it's whenever the controller feels like sending it.

I _think_ you're right about Marlin responding with "ok" as soon as it's been stored. I didn't find any special code in OctoPrint about pre-buffering lines or any mention of that in the discussion in foosel/OctoPrint#450, which suggests to me that there's not much special handling needed to take advantage of buffering moves. Maybe we can read the Marlin code to be certain.

MitchBradley, 2019-04-15:
Perhaps CNCjs is intermittently stalling for long enough to cause buffer underruns.  Garbage collection could cause that, as could other processes running on the Pi.  If you run "top" you can see if there is interference from other processes, and also how much memory is being used. In my setup, the Pi Zero is essentially running only cncjs - no window systems or any other intrusive processes.
One might ask "why doesn't it happen with OctoPrint".  It could be that OctoPrint is written in such a way that it puts less pressure on garbage collection, or uses less memory, or any number of other hard-to-pin-down factors.  CNCjs code, to me, looks like it was written with "big machine" assumptions, using every fancy JavaScript feature imaginable. I find it somewhat remarkable that it works as well as it does on my Pi Zero.

johnboiles, 2019-04-15:
@MitchBradley I’ll try also disabling the window system I’m using to run the tiny web pendant and see if that makes a difference. Node is single process right? I would think there should still be 3 other cores to run this other stuff though. I’ll also try the electron app on my Mac to make sure it goes away (and rule out a code bug).

johnboiles, 2019-04-15:
Mostly for my own sanity, here are the list of things I want to do/try:

1) Comment out `queryPosition` and `queryTemperature` (@cheton's [suggestion](https://github.com/cncjs/cncjs/issues/454#issuecomment-483341306))
1) Look at `top` while running a job in cncjs
1) Try running on a Pi3B+
1) Try running the Electron app from my 2.8GHz i7 quad-core Mac
1) Try running after disabling the window system + chromium + pendant on my Pi2B
1) Look at the Marlin code to verify for sure that it returns `ok` after storing the move in the buffer (not after the move completes) (@jeffeb3's [idea](https://github.com/cncjs/cncjs/issues/454#issuecomment-483374487))

MitchBradley, 2019-04-15:
If there is a betting pool for this, I want 5.

johnboiles, 2019-04-15:
Haha. We'll see! I could totally see the possibility that Node's v8 and Chromium's v8 have some sort of resource contention.

johnboiles, 2019-04-16:
Here are the results so far:

*5.* @MitchBradley you lose this time! Killing the window system (Xorg + openbox + chromium) didn't make any difference
*2.* Node barely uses half a core while running this job. The highest CPU usage I saw was in the 50%s.
[image]
*1.* We probably should have seen this coming, but @cheton was right. Removing those two lines makes CNCJS perform comparably to OctoPrint :tada: Temperature reporting is probably unnecessary for most CNC users. Doesn't look like there's an [auto-report for position](http://marlinfw.org/docs/gcode/M155.html) unfortunately in Marlin (thought maybe could be added). @cheton what's the next step here? Do we remove these lines or put them behind a setting or something?
*6.* I think the success of 1 confirms that Marlin is able to prebuffer commands without any special hand-holding.

I still want to test 3 & 4.

johnboiles, 2019-04-16:
I tested 3 (run on Pi3B+) and there was no obvious difference from the Pi2. I'll compare side-by-side videos to see if there was anything subtle.

I tested 4 (Electron app on 2.8GHz i7 quad-core Mac). I accidentally left the controller setting on Grbl and the Gcode ran just fine in Grbl mode. I switched the controller setting to Marlin and had the same Slow GCode processing issue. I haven't tried the Electron app with the above fix though. I'll be interested to compare the fixed Marlin controller with the Grbl controller to see if their performance is comparable.

cheton, 2019-04-16:
@johnboiles 
Per your testing, which factor will dramatically affect the performance? `queryPosition` or `queryTemperature`

@whimsycwd 
Did you have similiar performance issue when comparing with OctoPrint? It appears that removing temperature query and position query will give better result.

cheton, 2019-04-16:
Increase the timer interval of position query and temperature query to a reasonable value might help.

Default interval is 500ms
```js
// Get the current position of the active nozzle and stepper values.
queryPosition = (() => {
    return _.throttle(() => {
        //
    }, 500);
})();
```

Default interval is 1000ms 
```js
// Request a temperature report to be sent to the host at some point in the future.
queryTemperature = (() => {
    return _.throttle(() => {
        //
    }, 1000);
})();

jeffeb3, 2019-04-16:
Do the CNC.js query commands wait for a response? I wonder if Marlin stops to answer or if CNC.js stops while waiting for the answer.

Is it easy to still query position when jogging or idle, but not when playing gcode?

cheton, 2019-04-17:
@jeffeb3 @johnboiles 

There is a preemptive position/temperature query in CNCjs that may block the execution of a G-code program. If there is a delay of two seconds (or more) without returning position or temperature information, it will temporarily block G-code execution. This might be an area for improvement, but I'm not really sure whether it will help by removing only preemptive query.

https://github.com/cncjs/cncjs/blob/v1.9.18/src/app/controllers/Marlin/MarlinController.js#L608-L614
```js
// Perform preemptive query to prevent starvation
const now = new Date().getTime();
const timespan = Math.abs(now - this.query.lastQueryTime);
if (this.query.type && timespan > 2000) {
    this.query.issue();
    return;
}

// Continue executing G-code program
```

jeffeb3, 2019-04-17:
So putting those query lines back in, and removing this preemptive block would only query temp and position when idle?

johnboiles, 2019-04-18:
> Do the CNC.js query commands wait for a response? I wonder if Marlin stops to answer or if CNC.js stops while waiting for the answer.

This is a good question. Is it fundamentally not possible to get position from Marlin without slowing down GCode execution OR is there something we can do in CNC.js to be able to do both. Could we test this simply by just shoving some `M114`s into the middle of a gcode file and running it?

> There is a preemptive position/temperature query in CNCjs that may block the execution of a G-code program.

So should I try without this block? In my testing, I didn't remove this block, and CNCJS performed comparably to OctoPrint. What do you think might happen if I remove this block?

MitchBradley, 2019-04-18:
Why not just try it instead of asking what someone thinks might happen?

jeffeb3, 2019-04-18:
> So should I try without this block?

I think you should try it without that block, but put the query functions back in. That should not interrupt while printing, but still have valid temps and position when idle.

johnboiles, 2019-04-18:
Ok so things to try are:

1) Try inserting some M114s into gcode (and running with Octoprint) -- answer the question of whether Marlin is slow when being queried, or if the situation can be remedied in CNCjs.
1) Try without the 'preemptive query' block and re-add the query functions
1) Try adjusting the timer intervals based on [@this comment](https://github.com/cncjs/cncjs/issues/454#issuecomment-483528334)
1) Try removing the temperature query and the position query independently to see if one affects the speed more than the other
1) (Optional) Try the `SP_TYPE_CHAR_COUNTING` protocol with Marlin as per [this comment](https://github.com/cncjs/cncjs/issues/454#issuecomment-483523372

Anything else?

cheton, 2019-04-18:
Hi all,

Just to mention that I found a bug in the code.

If a query type is assigned in position & temperature query, it should update `this.query.lastQueryTime` to current timestamp (i.e. `now`). However, I misspelled the expression from `this.query.lastQueryTime = now` to `this.lastQueryTime = now`, therefore results in frequent preemptive queries.

You can find the expression `const timespan = Math.abs(now - this.query.lastQueryTime)` in preemptive query, the returned timespan will always be greater than 2000.

#### Position / Temperature Query

https://github.com/cncjs/cncjs/blob/v1.9.18/src/app/controllers/Marlin/MarlinController.js#L125-L148

```js
query = {
    // state
    type: null,
    lastQueryTime: 0,

    // action
    issue: () => {
        if (!this.query.type) {
            return;
        }

        const now = new Date().getTime();

        if (this.query.type === QUERY_TYPE_POSITION) {
            this.connection.write('M114\n', {
                source: WRITE_SOURCE_SERVER
            });
            this.lastQueryTime = now; // <-- a typo here
        } else if (this.query.type === QUERY_TYPE_TEMPERATURE) {
            this.connection.write('M105\n', {
                source: WRITE_SOURCE_SERVER
            });
            this.lastQueryTime = now; // <-- a typo here
        } else {
            log.error('Unsupported query type:', this.query.type);
        }

        this.query.type = null;
    }
}
```

#### Preemptive Query

```js
const now = new Date().getTime();
const timespan = Math.abs(now - this.query.lastQueryTime);
if (this.query.type && timespan > 2000) {
    this.query.issue();
    return;
}]
```

cheton, 2019-04-18:
@johnboiles 

You can add another test case using the bug fix below that tries to reduce the amount of preemptive queries:

```js
if (this.query.type === QUERY_TYPE_POSITION) {
    this.connection.write('M114\n', {
        source: WRITE_SOURCE_SERVER
    });
    this.query.lastQueryTime = now;
} else if (this.query.type === QUERY_TYPE_TEMPERATURE) {
    this.connection.write('M105\n', {
        source: WRITE_SOURCE_SERVER
    });
    this.query.lastQueryTime = now;
}
```

johnboiles, 2019-04-18:
Great find! @cheton to be clear I should revert the other changes and try only this fix to see if it solves the issue? Ill do this one first.

Hoping to get some time with the machine in the next few days

johnboiles, 2019-04-20:
Ok I got a chance to try your fix @cheton and it works beautifully. In my side-by-side comparison with OctoPrint, CNCjs performs very comparably. Given the same GCode file, CNCjs actually finished faster than OctoPrint this time, though it seems to have ever-so-slightly jerkier motion. I don't understand why CNCjs would be faster but also jerkier, since it seems jerky motion would cause it to fall behind. But I can only see the difference watching the videos side-by-side and synced up. It wouldn't be noticeable otherwise. And either way I'm not going to be running 7200mm/min cuts in real life.

I went ahead and made https://github.com/cncjs/cncjs/pull/462 in case it's easier for you to pull in. But feel free and discard that if you'd rather have the fix attribute to you!

cheton, 2019-04-20:
@johnboiles 

That's great news. I'd also like to know if there is a frequency of jercky motion. I guessed disabling preemptive query may help.

May you upload your comparison video? Thank you.

johnboiles, 2019-04-20:
Here's the [uploaded comparison video](https://www.youtube.com/watch?v=wFs6p9hSVGo). Hard to tell the frequency of the jerky motion since it's very subtle. Possible it's 500ms because of the position query timer. I wonder if there'd be a better test pattern that would the timing easier to see. Perhaps a spiral or something? If there's something else you'd want me to test, I might be able to do it tonight.

cheton, 2019-04-20:
From 43s to 48s in your video, I noticed a slight jercky motion when doing arcs. It is about 2 seconds that's similiar to the frequency of preemptive query:

```js
if (this.query.type && timespan > 2000) {
    this.query.issue();
    return;
}
```

Repeating a lot of same arcs may be a straightforward way for making a proof.

johnboiles, 2019-04-20:
If we do determine that the period is definitely every two seconds, would we remove the preventive query or just make it longer? Would position updates still work during the cut if we removed it?

At this point, it’s working great for my use, so I’m wondering how much more we could expect to accomplish. Unless you feel like there could be a definite win here I might move on to a different task (maybe https://github.com/cncjs/cncjs-pendant-tinyweb/issues/9)

MitchBradley, 2019-04-20:
In principle, one could infer position by counting lines and dry-running the GCode to determine the approximate position.  It would, of course, not be synchronized exactly, but it never really is anyway while the machine is moving.  To make it even more complicated, the position inference code could model the time taken by individual gcode moves and issue position requests during long ones.   That would rarely come into play in a 3D printing situation, but it would help for traverses on a milling machine.  My milling programs often include long straight moves or long arcs that can takes several seconds to complete.
We are suffering from the fallout of the evolutionary process that gave us GRBL and Marlin and friends.  The fact that those projects started out on 8-bit AVR chips led to some protocol choices that continue to challenge us.  It's a bit unfortunate that the Arduino world developed around AVRs, thus training a generation of hobbyists to cut corners.  ARM chips in the $1 price range have been available for a long time.
Imagine how much more robust it would be if controllers spoke TCP instead of using ad-hoc flow control. TCP can run just fine over a serial connection, but it does require more RAM than is available on some AVRs.  There are TCPs, though, that fit in a few K of FLASH and a few hundred bytes of RAM.
I am toying with the idea of doing a CNCjs server on a (dual core) ESP-32, with one core running the real-time motion control and the other managing the network.  No need for serial; communicate via shared memory.

jeffeb3, 2019-04-20:
Is there a serial log? It sure seems like it's querying position, but less frequently. At any rate, it's definitely good enough, but if there were motivation to push it further, removing the queries by settings altogether would be a smart choice. Those pauses would certainly add warts to a 3D print. 

@Mitch, the biggest unknown on the location prediction is the amount of code in the buffer. There's not a great way to know that, and on something like cutting the perimeter on a cabinet side, 16 moves could be a large error. That said, it is more accurate than no position updates.

I still wonder why it pauses at all. I am still thinking there's probably a problem in Marlin, and it should be able to return occasional position updates without stopping motion. I bet it's adding a dwell or something and then waiting during transmission before starting again.

cheton, 2019-04-22:
Close this issue because the bug is fixed in [1.9.19](https://github.com/cncjs/cncjs/releases/tag/v1.9.19), although it is still not good enough for 3D printing.

We can continue discussion here to find out a better solution. I will make further improvements in a future release.

PR #462 (2019-04-20): Fix for the lastQueryTime Marlin bug
This is the bug @cheton discussed in https://github.com/cncjs/cncjs/issues/454#issuecomment-484427057

Closes #454
closed 2019-04-22 by cheton