cncjs/cncjs #456: Machine hangs while running a program

karoria, 2019-04-16:
#### Description

Machine hanged 2 times at different locations (job kill by reset only). The feed stops while keeping spindle running. This was my first lengthy program. I also tried by disabling 3d visualization but no help.
Is it a cncjs related issue or g2core? Will I get any benefit from altering status report interval settings of g2core?

#### Versions

- CNCjs: 1.9.18
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

karoria, 2019-04-16:
Sorry forgot to mention that planner buffer was showing 5 at that time.

AustinSaintAubin, 2019-04-17:
Do you have anything else plugged into the Raspberry Pi.
I had a similar issues, a faulty USB camera kept causing a reset on the USB interface.
I replaced the camera and no more issues. 

If you are using [pm2 to run cncjs](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*#install-production-process-manager-pm2), then you can run the command `pm2 monitor cncjs` to [monitor](http://pm2.keymetrics.io/docs/usage/monitoring/) the running cncjs process in realtime.

karoria, 2019-04-17:
Thanks for the reply. I have nothing attached to raspi except g2core due board. I use a 7 inch original pi display connected to pi.
One more input. When my interface hangs, i can send cycle start through full interface (though it doesn't update xyz values) and that makes the machine move again. I kept this doing for several times till the program ended. Any clues?

cheton, 2019-04-17:
@karoria 

There are something you can try yourself:
* What's your g2core firmware build? Do you have the same issue with other G-code sender like UGS?
* Attach your g-code file for further investigation.
* Run `cnc -vvv` to enable verbose output and check if there are errors occurred when it hanged. Also attach your archived log file if possible.

cheton, 2019-04-17:
> I also tried by disabling 3d visualization but no help

It's not necessary to keep your browser running after starting a G-code program. You can close/open browser window anytime during G-code program execution, it won't affect the server, unless you click the "Close" button to close serial connection from the connection widget.

karoria, 2019-04-17:
Thanks a lot for the tips @cheton. Will surely check it and update here.

LongLiveCHIEF, 2021-01-31:
I'm having this same issue. I have no other peripherals, but I am running https://github.com/kreso-t/cncjs-kt-ext.

Specs:
```
Rpi 4 - 4gb RAM
Ubuntu Server 20.02
gnea/grbl v1.1h.20190825 on a ATMega 328p
Node v12.18.4
cncjs v1.9.22
```

It's stopped in the fashion described above on 3 attempts of the same job, but in different places in the job each time. This last time it stopped, I was tailing the verbose output, and there were no hard errors... one second it's logging coordinates and moves, and then all of a sudden it stops, and the log just shows:

```
Continue status report query: timespan=5012ms
Continue status report query: timespan=5007ms
Continue status report query: timespan=5010ms
Continue status report query: timespan=5008ms
```

No errors were present in the autolevel plugin either.  

After this happens, the spindle just keeps spinning, and the machine won't respond to any moves from the client.

LongLiveCHIEF, 2021-01-31:
It sorta acts like a serial communication interruption, except that the interface doesn't reset.  I'm going to watch for udev events during my next job to see if there is indeed a disconect.

Also, comPort is `/dev/ttyUSB0`

LongLiveCHIEF, 2021-01-31:
Ran `udevadm monitor` while running the job, same disconnect, no udev events happened.  I did notice that the sent/received seemed to be stalled though.  It only sent about half the gcode, and the it wasn't sending the other half.

karoria, 2021-02-01:
I suspect this is due to noise. Try to check the program again with spindle vfd off. If it runs ok, then you should consider shielded and shorter usb cable. Check also for floating or not connected input pins and its wiring (like limit switches, probes etc.).

LongLiveCHIEF, 2021-02-01:
Yeah, that could def be it.   

I spent yesterday switching out to one of my spare [SKR 1.4's and set things up with Marlin](https://github.com/MarlinFirmware/Marlin/compare/2.0.x...HackerHappyHour:cnc-skr-1.4-lpc1769) instead, and that's running pretty well right now. That uses different control techniques so EMF's are different, plus that particular cable is shielded.

I'll see if I can shield a shorter cable of the right type at some next weekend and see what happens. Should be a fun experiment.

LongLiveCHIEF, 2021-02-11:
Yep, fixed. Was definitely noise.

closed 2021-02-11 by MitchBradley