cncjs/cncjs #457: CNCjs Unexpected Error was detected: Error : 22

JJZ71, 2019-04-18:
#### Description

Running the same gcode  >x6 and its failing 70% of the time with:

Unexpected Error was detected: Error : 22
The job hangs at different points/lines within the same job

#### Versions

- CNCjs: 1.9.18

#### How Do You Install CNCjs?

- [] NPM
- [ X] Download the CNCjs Desktop Application (Windows 10 64bit Notebook)

#### CNC Controller

- [x] Grbl (
[image]

Arduino)
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Arduino via Laptop
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [X ] Windows 10
- [ ] Mac
- [ ] Linux

cheton, 2019-04-18:
Check out https://github.com/gnea/grbl/issues/255 and see if feedrate is undefined in your gcode file.

closed 2019-04-18 by cheton

JJZ71, 2019-04-18:
Added $N1=F180 to the first line of my GCODE and there is no improvement
[image]

MitchBradley, 2019-04-18:
Add the line

G1 F180

To your GCode program before any other G1 lines.  This really is not a cncjs issue. Cncjs just sends the GCode that you tell it to send. If that GCode is not suitable for your controller you need to fix the GCode.

For most controllers, F180 by itself is not meaningful. It needs to be preceded by G1

JJZ71, 2019-04-18:
Ok Thanks, Second line of the GCODE is G1 F100.0

MitchBradley, 2019-04-18:
There must be an F word on the first line that is either G1, G2, or G3

JJZ71, 2019-04-18:
Amended the Gcode, no change :(

Cant see how it could be the GCODE is if works fine sometimes and not others.

cheton, 2019-04-19:
@JJZ71 

Do carefully read the Grbl wiki page especially the startup blocks:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#n---view-startup-blocks

You can set a feed rate in the startup block, it will take effect upon Grbl reset, but you should not add `$N1=F180` to the top of your G-code file, doing so will eventually wear out Arduino's EEPROM everytime you run the G-code program.

A correct usage is to set a feed rate to the first appearance of motion commands (G1/G2/G3):

```gcode
G1 X1 Y1 F180
G1 Z-1
```

You can also set a desired feed rate prior to motion commands:

```gcode
F180
G1 X1 Y1
G1 Z-1
```

cheton, 2019-04-19:
You can attach your G-code file if you need further help.

JJZ71, 2019-04-19:
[dremel gbrl postprocessor.txt](https://github.com/cncjs/cncjs/files/3096865/dremel.gbrl.postprocessor.txt)

Attached is the gcode as a ,txt file, i re-ran it and it still crashed

MitchBradley, 2019-04-19:
Please attach your gcode file

MitchBradley, 2019-04-19:
Oh, sorry, I see it above

cheton, 2019-04-19:
Moved the discussion to https://github.com/gnea/grbl/issues/638

MitchBradley, 2019-04-19:
Try putting the G17, G21, and G90 lines at the top.  Those commands set the coordinate system, so before they are executed, the parameters of the G1 command are ambiguous.  For example, if the system happens to be in G20 mode initially, the F180 will be interpreted as 180 inches/min, not 180 mm/min.