cncjs/cncjs #458: Please re-open CNCjs Unexpected Error was detected: Error : 22 #457

JJZ71, 2019-04-18:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-04-22:
It's closed because it is not exactly an issue in CNCjs. However, you can continue the discussion on https://github.com/gnea/grbl/issues/638 to get more help.

MitchBradley, 2019-06-16:
Closing, since there is nothing that CNCjs can do to resolve this issue (it's a problem with GRBL or the hardware setup).

closed 2019-06-16 by MitchBradley