cncjs/cncjs #459: mount is only valid in commandline, not cncrc file

prune998, 2019-04-19:
#### Description

Why is the `mount` option only valid on the commandline ? 
in `src/app/index.js` line 105 : 
```
ensureArray(options.mountPoints).forEach(mount => {
```

Why not using the same behaviour as other options like : 
```
const mountPoints = options.mountPoints || config.get('mountPoints');
ensureArray(mountPoints).forEach(mount => {
...
```

With `.cncrc` like : 
```
"mountPoints": [
      {
        "route": "/tinyweb",
        "target": "~/cncjs-pendant-tinyweb-1.2.4"
      }
    ]
```
 
Any reason ? 

#### Versions

- CNCjs: 1.9.18
- Node.js: 10.15.3
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-04-19:
I made a change in https://github.com/cncjs/cncjs/pull/461 that add support for specifying mount points in the `.cncrc` file.

cheton, 2019-04-22:
It's now available in [1.9.19](https://github.com/cncjs/cncjs/releases/tag/v1.9.19)

closed 2019-04-22 by cheton