cncjs/cncjs #467: Feature Request, custom text in M0 / M1 dialogues

bensuffolk, 2019-04-26:
#### Description

Feature Request, custom text in M0 / M1 dialogues.

I am using a tool change macro that has an M0 and a few M1 commands in to pause the program whilst I do the usual things like changing the tool, attaching the problem, removing the probe etc.

It would be great if I could have the dialogue box tell me what to do at each stage. At the moment it says "M0 Program Pause", "click the Continue button to resume execution". I'd like to be able to add an additional block of text after that, e.g "Connect the probe wire and position the probe"

I think the best place to put this text in the macro would be as a comment on the line before the M0 / M1. e.g.

```
(Connect the probe wire and position the probe)
M0

```

This could also then expand to the M6 manual tool change command to tell you which tool to fit. At the moment it just says "**M6 Tool Change**", "Run a tool change macro to change the tool and adjust the Z-axis offset. Afterwards click the resume button to resume program execution."

I tend to use Carbide Create for generating the majority of my gcode, and it looks like this:

```
(TOOL/MILL,0.2000,0,1.0000,15.0)
M6 T18

```

So from that it gives a bit of info about the tool as well as the actual tool number.

So the prompt for that could include the tool number in the title, e.g. "**M6 Tool Change: Tool T18**", and then the additional text after the other text in a new paragraph, e.g. "TOOL/MILL,0.2000,0,1.0000,15.0", which although cryptic does remind me what tool 18 is.

I guess I could actually number all the tools the same as Carbide Create does, but displaying the Tool number would still be really good for that.

#### Versions

- CNCjs: 1.9.18
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

EricKPhoto, 2019-12-24:
In Marlin M0 commands the custom text comes after like: 

`M0 Tool Change - 1/8 endmill
M0 Attach probe
M0 Turn on Router`

bensuffolk, 2020-01-02:
I’d be just as happy with it following as in Marlin.

Billiam, 2020-04-07:
This would be great with a postprocessor to translate tool RPM to a speed setting for manual speed control routers also.

excessnet, 2020-11-04:
any update or walk around for this? :)

PR #708 (2022-04-11): Propagate line comments in GRBL M0/M1/M6 commands to UI dialog
Capture any line comment (everything after a semicolon) on an M0, M1
or M6 command and display is as part of the pause dialog in the UI.

Fixes #467.
closed 2022-04-11 by cheton