cncjs/cncjs #474: Touch screen operation issue

Foxbiker, 2019-05-08:
#### Description

Hallo, I found issue in touch sreen operations. Buttons like settings, minimalize and keyboard on wingets is not possible press by touchscreen only with mouse. I have this problem on Win8.1 but I had the same issue on Raspberry Noobs web browser so it looks general. Maybe some wrong settings in object properties.

Regards 

Radek

#### Versions

- CNCjs: 1.9.20
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [x] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-05-08:
The cncjs-shopfloor-tablet UI was developed specifically to deal with this problem.  Many of the controls in the full CNCjs UI are too small or too close together for reliable touchscreen use.  cncjs-shopfloor-tablet uses large buttons that stay in place on the screen so you can touch them predictably.  It does not have all the features of the full UI, but it does have what you need for most job-running situations.

Foxbiker, 2019-05-08:
For small displays I uderstand it but with 15" display buttons are large anought for pressing. For example buttons for open/close wingets panels working normally and have the same size. It's pitty that most operations is possible make by touch but from these reason you still need a mouse in workshop.

MitchBradley, 2019-05-08:
If you are using the full CNCjs UI in a workshop, I hope that your machine is low power so that, when it crashes, nothing really bad happens.  The full UI is problematic for critical use because controls can easily scroll off-screen so you cannot get to them quickly if, for example, you need to stop in a hurry.  And the controls are small and hard to hit when something is going wrong and you are not calm.

Foxbiker, 2019-05-08:
It´s low cost and power home made hobby plotter not profesional production machine.
[image]

MitchBradley, 2019-05-09:
Nice looking setup

cheton, 2019-05-09:
I never had a chance to test the UI with a touch screen monitor, but I can reproduce this issue by emulating Chrome on a mobile device, as shown below:

[image]

Buttons on widget header don't work, those buttons are basically HTML anchor tags. I will take a look why it won't fire the `onClick` event on mobile devices.