cncjs/cncjs #475: Not smooth movement in Marlin

Foxbiker, 2019-05-08:
#### Description

Hallo, I spend many days with finding the reason why spindle move in X and Y are not smooth. Aprox every 4 seconds move stop for a very short time and continue. This make vibrations and marks on workpiece. I try change every possible parameters and buffers in Marlin FW but without any change. Today I found the reason. I found that this short movement delay happen everytime between gcode commands and when actual values change in Axes Widget. I try Pronterface SW with the same gcode for compare and it run absolutely shooth without any stops. For testing I made wavy circuit object withou sharp conners. Gcode contained only G2 and G3 commands.
It looks that is some period in CNCJS to ask Marlin for actual positions and this make this issue.

Regards 

Radek  

#### Versions

- CNCjs: 1.9.20
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-05-09:
Looking at the user interfaces for both Pronterface and Octoprint, it appears that neither display real-time position updates.  The closest thing I can see is an overlay on a display of the toolpath.  That overlay could be generated by counting lines that have been sent - but it is probably not real-time accurate. If the GCode consists of many short segments, such a display would typically be pretty close, and accurate enough for monitoring a 3D printer job that could take hours to run.

The Marlin support in cncjs is relatively new and we are facing some challenges in making Marlin work well within the CNCjs paradigm.  In particular, Marlin makes it very difficult to get real-time position updates.  Getting them during the execution of a long move is essentially impossible, and getting them between moves, as you have seen, sometimes cause stuttering. Another thing that is tricky in Marlin is determining when the machine has paused.

For a machine that is not a 3D printer, for example a milling machine, plotter, router, laser engraver, etc, it could be that another firmware is more suitable than Marlin.  GRBL, Smoothie, and TinyG/g2core all have good position reporting and reasonable machine-state reporting.  I personally like GRBL for laser engraving due to its excellent "laser mode" which modulates the laser power based on motion and g2core for milling due to its strong handling of machine state and spindle operations.

For a 3D printer, Marlin is an outstanding choice because it is optimized for that use case, with many relevant M-codes for the special needs of 3D printing - temperatures and extrusion and stuff like that.

Is there a strong reason for the use of Marlin on your machine?  And if so, why are you looking at CNCjs instead of the other UIs that are so well integrated with Marlin and 3D printing?  (@johnboiles might like to weigh in here, as a prominent Marlin user in the cncjs community).

Foxbiker, 2019-05-09:
I start build my plotter on Marlin base because I have 3d backgraund and many people say that is it posible. I already fixed most problems in Marlin FW and found perfect CAM software for this aplication. Client SW is my last challenge. Because I wery like your SW I want to try fix it for Marlin aplications. This issue is from program view wery easy to fix with keeping coordinates reading. Ask for position only after G0 better G0 Z..... command. In this moment all milling movements are done and mill is up before move on another penetration pisition and some little delay is not problem. I think it is easy Marlin solution and fix this issue for everytime. In standard gcode is manny G0 lines so coordinates refresh will be OK.

MitchBradley, 2019-05-09:
That approach solves your problem but not mine.  I need real-time coordinate display when cutting.

Foxbiker, 2019-05-09:
If it is my problem, ok. If you need real-time coordinate display I can´t more help with this issue. Realy realy-time coordinates are currently unavailable only some refresh which causes this issue. My opinion is that more important for cutting is have smooth movemets then absolutely real-time coordinate display. As you wrote, Marlin is problematic and you are right with Pronterface, Octoprint and Repetier-Host not show real-time values. Probably this is the reason. My solution proposal is compromise solution for this issue, movement will be smooth and you will have some coordinates for show progress in 3d preview. But your project your rules.

cheton, 2019-05-09:
> Ask for position only after G0 better G0 Z..... command.

For this purpose, you can implement a G-code processor to insert `M114`s after `G0`s to query position. It's not hard to do using the [gcode-parser](https://github.com/cncjs/gcode-parser). If you need a sample code I can write it for you.

Other than this approach, one thing we can do is to provide configurations that allows you to decide whether to enable position / temperature reports especially when streaming a G-code program.

Here is an example for the settings:

```js
{
  state: {
    controller: {
      marlin: {
        query: {
          position: { // Use M114 to get current position
            interval: 1, // Interval in seconds between position updates. 0 to disable.
            streamingInterval: 0 // [optional] Interval in seconds between position updates. 0 to disable.
          },
          temperature: { // M155 will be used over M115 if temperature auto-report is supported
            interval: 1, // Interval in seconds between auto-reports. 0 to disable.
            streamingInterval: 0 // [optional] Interval in seconds between reports. 0 to disable.
          }
        }
      }
    }
  }
}
```

johnboiles, 2019-05-09:
> @johnboiles might like to weigh in here

I use Marlin because it's the default controller for the MPCNC (Mostly Printed CNC) machine I've built. There are Marlin modifications specifically for the MPCNC (which has 4 endstops) that don't exist for GRBL. I think the MPCNC designer chose Marlin is because he assumed that 'Mostly Printed' CNC builders would already be familiar with it. I would rather use GRBL, but would need to add support to GRBL for the 'Dual Endstops' system on the MPCNC.

> Other than this approach, one thing we can do is to provide configurations that allows you to decide whether to enable position / temperature reports

I think more configuration options can't hurt!

> I need real-time coordinate display when cutting.

I think probably the best way to support this would be add something like [M155](http://marlinfw.org/docs/gcode/M155.html) (Temperature Auto Report) for position updates in Marlin. I suspect that without this, there's not going to be a perfect way to add real-time position updates without causing hiccups in the gcode processing.

In the meantime however, I agree with @Foxbiker that I'd rather have smooth GCode processing than real-time position updates.

Foxbiker, 2019-05-09:
Cheton, this solution is nice and easy but unfortunately not fix this issue. It must say programmer. I´m presuming that CNCJS have in Marlin mode some timer which sending M114 requests to the Marlin during run and it do without some regard on ongoing movement proceses. I spend some time with reading many Marlin forums. Marlin have in buffer more gcode command and is assuming if next movement will be continuing or stop movement (sharp corner atc.). When Marlin find M114 next the move command stop the movement, send answer and continue. Hard to say why stop. Maybe this procedure with all Marlin functions is big load for 8bit procesor. I don´t know. We can discuss if it is issue in Marlin but most 3d print client SW are working properly. Problem is that most 3d printer clients are very bad for cutting use. For this I don´t know better client like CNCJS. And this is the main reason why I try change something here. If CNCJS say we are compatible with Marlin they must play Marlin game not say it is your problem we want real-time value like from Grbl or Smootie . Sorry it is my opinion.

cheton, 2019-05-09:
CNCjs is a software for general purpose use, that's why I propose adding more [configuration options](https://github.com/cncjs/cncjs/issues/475#issuecomment-490925868) for different needs. It's hard to determine the right timing for position update, appending `M114` after `G0 Z..` might be good in your case, but it could be worse for others.

For a 3D printing job, most of time you only care about the print time and the temperature, not the position updates. If you want CNCjs to have smooth movements as well as Octoprint, you might choose to turn off position query during printing. For the temperature report, we can possibly use M155 (temperature auto-report) to gain better performance than polling with M115.

For a CNC/Laser job, it is no doubt that the temperature report should be off. Also, you can choose to turn off position query for laser engraving to get smooth lines.

johnboiles, 2019-05-09:
@Foxbiker do you have any links to the discussion or code that confirms that `M114` stops the movement. I'm very curious if there was a strong reason for that to be the behavior in Marlin. Seems like a position query wouldn't be very expensive on the processor.

Your idea about putting `M114` after `G0 Z...` is clever, but at least for my use, since I'm only making a handful of `G0 Z..` moves per cut, I'd only get a few position updates -- and usually at the same spot in the cut where it plunges. I'm not sure how useful it'd be for me.

@cheton sounds like a config or UI switch to turn off the position query while running GCode is probably going to be the best bet for Marlin for now.

MitchBradley, 2019-05-09:
For M114 the Marlin code calls planner.synchronize(), which idles until the planner queue is empty.

https://github.com/MarlinFirmware/Marlin/blob/5d96a6d983c065403216ed52a8c92ed22dc268ba/Marlin/Marlin_main.cpp#L9151

https://github.com/MarlinFirmware/Marlin/blob/5d96a6d983c065403216ed52a8c92ed22dc268ba/Marlin/planner.cpp#L1539

MitchBradley, 2019-05-09:
@johnboiles - Just FYI, there is a way to use dual endstops with g2core.  It isn't nicely packaged, but it looks like it would work.  See https://github.com/synthetos/g2/issues/359  That probably is not much use to you if you are using a RAMPS style board though.

johnboiles, 2019-05-09:
Good find @MitchBradley with the MarlinFirmware links!

I'm using an Ultimaker Rambo board (I think it's an evolution of RAMPS). A quick Google search doesn't show anything about hooking that up with the g2core firmware.

MitchBradley, 2019-05-09:
g2core is primarily for 32-bit chips, the best-supported being the ARM flavor that is on Arduino Due.  My setup has a Due underneath a gShield that drives the Z and A steppers, with higher-power integrated closed-loop steppers on X and Y.  With that setup I can rapid a leadscrew Taig mill at the same speed as the ballscrew version of the Taig mill.
The ancestor of g2core that is for 8-bit chips is TinyG.  TinyG normally runs on an ATXmega192A3 which is faster and has more RAM than is usual for RAMPSy boards.
As you can probably tell, I really like the g2core firmware. Unfortunately, there are signs that its developers may be losing interest or at least too busy with profitable work to spend much time on it.

MitchBradley, 2019-05-09:
I just tried a quick test where I commented out the planner.synchronize() line in Marlin's M114().  Nothing bad happened; my quick-and-dirty Marlin test program ran as expected.  I don't know if anything good happened, because I don't have an actual machine hooked up to the Marlin board (which is a 32-bit Teensy 3.1 running the bugfix-2.0x branch).  The Marlin team's attempt to support a huge variety of processors, both 8 and 32 bit, on 2.0 is truly impressive.  I wish I could say the same thing about their serial line protocol.

bDuthieDev, 2020-03-19:
I'm using Marlin 2.0.x (2.0.5.1) for testing with a Creality 1.1.4 board. I also commented out planner.synchronize() to see actual positions (not projected positions), but noticed a new option to use "M114 R" to get the same result.