cncjs/cncjs #476: Possible Sender bug?

MitchBradley, 2019-05-11:
I have been trying to understand the Sender and Workflow logic in detail and I have found what I think might be a bug.

In Sender.js, we have this code:
```
                   // The newline character (\n) consumed the RX buffer space
                   if ((sp.line.length > 0) && ((sp.dataLength + sp.line.length + 1) >= sp.bufferSize)) { 
                        break;
                  }
```
If sp.line.length == 0, then the remainder of the test is skipped, and sending proceeds. That seems incorrect to me.  A blank-line-plus-newline consuming the RX buffer space seems the same to me as a non-blank line plus newline consuming it.  If that interpretation is correct, then the `(sp.line.length > 0) &&` part should be removed.

cheton, 2019-05-11:
Here is a detailed explanation to the [code](https://github.com/cncjs/cncjs/blob/master/src/server/lib/Sender.js):

#### Loading phase
All empty lines will be filtered out when loading a gcode, the `filter()` method returns a new array with non-empty lines:

```js
const lines = gcode.split('\n')
    .filter(line => (line.trim().length > 0));

this.state.lines = lines; // non-empty lines without "\n"
```

#### Processing phase
If `sp.line` is an empty string (`''`), it will return a string at the specified index (i.e. `lines[sent]`) and call the `trim()` method to remove whitespace from both ends of a string:
```js
sp.line = sp.line || this.state.lines[this.state.sent].trim();
```

The dataFilter is used to [translate expression](https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Grbl/GrblController.js#L265-L313) in a line, gcode comment will be removed before returning a new line. **_Note that the filtered result may be an empty string after transformation._**

```js
if (this.dataFilter) {
    sp.line = this.dataFilter(sp.line, this.state.context) || '';
}
```

Actually, the `(sp.line.length > 0) &&` part should not be removed from the expression below because it only care about non-empty string when computing the available buffer length:
```js
// The newline character (\n) consumed the RX buffer space
if ((sp.line.length > 0) && ((sp.dataLength + (sp.line.length + 1)) >= sp.bufferSize)) {
    break;
}
```

Empty lines will not emit a `data` event to write data to the controller, and these empty lines will be acked by the sender itself:
```js
this.state.sent++;
this.emit('change');

if (sp.line.length === 0) {
    this.ack(); // ack empty line
    continue;
}

const line = sp.line + '\n';
sp.line = '';
sp.dataLength += line.length;
sp.queue.push(line.length);
this.emit('data', line, this.state.context);
```

MitchBradley, 2019-05-11:
Understood, thanks.

closed 2019-05-11 by MitchBradley