cncjs/cncjs #479: Feature : add step-by-step execution of G-code

prune998, 2019-05-18:
#### Description

I was running a pretty simple job and got some errors. So I tried to run the file step-by-step, one command at a time to see what's going wrong... and I think this is a missing feature.
Would it be hard to add a "run next command" (or forward one step) with the play/pause/stop buttons ? 
Maybe it could be in a widget too if it's easier ? 

#### Versions

- CNCjs: 1.9.18
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2019-05-19:
This is not exactly what you asked for but it might solve your problem. I'm working on cncjs mods so it will stop on errors and show the problem line.

mmdoogie, 2020-06-30:
I've got a branch where I was working on doing something like this if anyone wants to try it out.  It adds a button beside the play button that does a single-step.  Also highlights the current line in the g-code listing view.

I only have a Grbl controller so and have only done superficial testing on that.  I plan to play around with it some more eventually.  The Smoothie and Marlin code are close enough that I think they probably work.  The TinyG probably needs some tweaks, especially near the end of a program.  Have an e-stop handy...

[mmdoogie/cncjs@single-step](/mmdoogie/cncjs/tree/single-step)