cncjs/cncjs #480: show gcode operation number and command in gcode widget

Raffa2s, 2019-05-19:
#### Description
I know that this will be embedded in the new version under 3D View but is possible to activate it under the settings menu? I just like to see what command will be executed and when like this one in the [picture](https://imgur.com/hgasSnd)
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

m33hew, 2019-07-04:
I would really like to see this also, when proving out a new programme knowing the next step is vital.

emcniece, 2022-03-27:
Related:
- https://github.com/cncjs/cncjs/issues/536
- https://github.com/cncjs/cncjs/pull/617