cncjs/cncjs #481: Accessing cncjs launched on rasp pi from a tablet, is port forwarding set up on rasp pi necessary?

JalenTk, 2019-05-20:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.20
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

I wanted to access CNC.js launched on rasp pi from a mobile tablet. Both the rasp pi and mobile tablet are connected to a mobile hotspot, what can I do to access the CNC.js server (port 8000) from the tablet? Do i have to set up port forwarding from the rasp pi?

Thanks in advanced!

JalenTk, 2019-05-20:
`wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.204.63  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::136c:904a:21f4:c4e5  prefixlen 64  scopeid 0x20<link>
`

I believe the IP address for the rasp pi will be 169.254.204.63 as shown above

MitchBradley, 2019-05-20:
Have you tried adding "-host 169.254.204.63" to the cncjs command line that starts the server?

JalenTk, 2019-05-20:
@MitchBradley , oh man, totally forgotten about that! Thanks for the help! Appreciate it!

closed 2019-05-20 by JalenTk

cheton, 2019-05-21:
169.254.x.y is an auto-configuration IP address, it tells that your network setup isn't working well and cannot react the DHCP server. However, this is fine for a workshop if you don't need Internet connectivity.