cncjs/cncjs #483: CNCJs Electron on AMD system

nickm324, 2019-05-21:
#### Description

Install 64 bit version of CNCJs Electron app, double click the icon and windows asks for admin permission, click yes, app looks like its going to run but never displays any UI. I can look in the Task manager and I see two instances of CNCJs.exe one using more memory than the other, but no UI ever displays. 
Running on an AMD FX-4100 based system.

#### Versions

- CNCjs: 1.9.20
- Node.js: 10.15.3
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [X ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-05-22:
@nickm324 

Try terminating CNCjs processes and run again, if it doesn't work, you can follow the troubleshooting guide to see what is wrong with the electron app.
https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows

Note that you have to download Electron 4.1.0 if you're using CNCjs 1.9.20.
* Electron 4.1.0
    https://github.com/electron/electron/releases/download/v4.1.0/electron-v4.1.0-win32-x64.zip

Run `C:\Temp\electron.exe main.js -vvv` under the CNCjs installation directory to see verbose output:

[image]

nickm324, 2019-05-22:
Terminating in task manager and re-running does not make a difference. 
I tried following the trouble shooting instructions but there are no files in the  C:\Users\{Username}\AppData\Local\Programs\cncjs\resources\app\ path, only thing in that folder is a node_modules folder which has a iconv-lite folder then an encoding folder which is empty.

nickm324, 2019-05-22:
Only thing I can find besides the empty CNCjs\resources\app folder is a cncjs-app folder that has the executable in it and other files. 
[image]

cheton, 2019-05-22:
Try running `Uninstall CNCjs.exe` on the above folder and install CNCjs 1.9.20 again (https://github.com/cncjs/cncjs/releases/download/v1.9.20/cncjs-app-1.9.20-win-x64.exe).

If the installation is successful, you should see those files and folders (e.g. `app`, `electron-app`, `web`, `cnc.js`, `main.js`) under `C:\Users\{Username}\AppData\Local\Programs\cncjs\resources\app\`

nickm324, 2019-05-22:
Did exactly what you said and the same thing, there is nothing in the CNCjs\resources\app folder. All I did was uninstall the existing install through Add/Remove Programs, then downloaded and ran the installer from the link you provided above.

[image]

nickm324, 2019-05-22:
Ok went back completely uninstalled the app again, manually deleted the CNCjs folder under Local\Programs. Ran the installer again and it never creates a CNCjs folder at all under Local\Programs. Only folder that is created is a cncjs-app folder with the executable and other files. Think there may be an issue with the installer ?

cheton, 2019-05-23:
I just forgot that the installation directory was changed from `CNCjs` to `cncjs-app` since 1.9.13, you should see `cncjs-app` when you make a fresh install. Did you see any other files under `cncjs-app\resources\app`?

nickm324, 2019-05-23:
These are the files in the C:\Users\rippe\AppData\Local\Programs\cncjs-app folder.

[image]

nickm324, 2019-05-23:
Ok I was finally able to run the troubleshooting command using electron and below is what I get. Seems that its looking for a C:\maker\.cncrc folder that does not exist on my system after the install is completed.

[image]

cheton, 2019-05-23:
There is a trailing comma after the "secret" key, that is invalid according to JSON standard. You can  paste the content to https://jsonlint.com/ to validate a JSON file.

Try fixing incorrect JSON format or removing the **C:\\Users\\maker\\.cncrc** file and start again should fix this issue.

nickm324, 2019-05-23:
That resolved the issue, removing the comma. Curious why it was there in the first place since I didnt even know that file existed and it was a fresh install.