cncjs/cncjs #485: How to remove widgets from my code

hupf0211, 2019-05-24:
#### Description

How do I remove widgets from my code?
For example, how do I modify the code to remove the "Macro" widget?

[image]

#### Versions

- CNCjs: 1.10.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-05-24:
You do not need to modify the code.  Just click on one of the Manage Widgets buttons at the top left or top right, and uncheck the widget that you do not want to see.

cheton, 2019-05-24:
@hupf0211 

If you are willing to modify the source code, you can filter out unwanted widgets as listed in below lines:

**PrimaryWidgets**
https://github.com/cncjs/cncjs/blob/master/src/app/containers/Workspace/PrimaryWidgets.jsx#L189-L206

**SecondaryWidgets**
https://github.com/cncjs/cncjs/blob/master/src/app/containers/Workspace/SecondaryWidgets.jsx#L189-L206

NitramLegov, 2019-05-31:
In addition, you can also check out issue #359 
We have created our own fork which modifies the default workspace settings in order not to show some of the widgets.

hupf0211, 2019-05-31:
Ok, I will try it, thank you!

NitramLegov, 2019-06-01:
Just as another hint, here´s what we did:
Since we only wanted to have the widgets supressed in the workplace, we took them out of the default state:
src/web/store/defaultState.js 
This leads to the widgets not to be shown when a new users connects. However, it still enables users to include the widgets onto their workspace in case they need them (e.g. some of our power users decided to include the grbl widget back to their workspace).

In addition, this enabled us to put in a default url for our webcam. So if any new device connects to cncjs, the user immediately sees the webcam stream.

hupf0211, 2019-06-05:
I need help. When I try to delete some controls, I get the following error.
[image]
Current interface：
[image]
Please help me，thank you very much.
cncjs-app-1.10.0.json:
[image]
[image]
[image]

se3d, 2019-10-27:
why the code when remove the widgets is not saving?