cncjs/cncjs #489: Docker images module version mismatch

yohcop, 2019-06-03:
#### Description

Docker images don't start, says a module was compiled against a different Nodejs version.

Same results with latest, and v1.9.20 (except the `NODE_MODULE_VERSION`s are different)

```
> sudo docker run cncjs/cncjs
[sudo] password for y: 
Unable to find image 'cncjs/cncjs:latest' locally
latest: Pulling from cncjs/cncjs
c5e155d5a1d1: Pull complete 
221d80d00ae9: Pull complete 
4250b3117dca: Pull complete 
3b7ca19181b2: Pull complete 
425d7b2a5bcc: Pull complete 
69df12c70287: Pull complete 
ad53476a61f2: Pull complete 
204bb8bac4a1: Pull complete 
f72f5ced5338: Pull complete 
6fe64b0754e2: Pull complete 
a6dfb1f83469: Pull complete 
b8a063378942: Pull complete 
Digest: sha256:aa682ea3616a6a2a5ef0f9573f31650a79361a2f1c2e29c958bde1d7fcffdcbc
Status: Downloaded newer image for cncjs/cncjs:latest
Error: Error: The module '/dist/cncjs/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 69. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:730:18)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at bindings (/dist/cncjs/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/dist/cncjs/node_modules/@serialport/bindings/lib/linux.js:1:98)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/dist/cncjs/node_modules/@serialport/bindings/lib/index.js:14:22)
```

And 1.9.20:

```
> docker run cncjs/cncjs:v1.9.20
[sudo] password for y: 
Unable to find image 'cncjs/cncjs:v1.9.20' locally
v1.9.20: Pulling from cncjs/cncjs
e79bb959ec00: Pull complete 
d4b7902036fe: Pull complete 
1b2a72d4e030: Pull complete 
d54db43011fd: Pull complete 
69d473365bb3: Pull complete 
6e2490ee2dc8: Pull complete 
24bcfd3d8296: Pull complete 
4c485b32137c: Pull complete 
10c7ea03b1e0: Pull complete 
7da3ff434f8a: Pull complete 
306d2f7957ad: Pull complete 
4ce325a7a1f4: Pull complete 
Digest: sha256:51a130be239e7a1686125eb484d921ab5e70dd5b9f227d4dc3b55b864142103f
Status: Downloaded newer image for cncjs/cncjs:v1.9.20
Error: Error: The module '/dist/cnc/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 69. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:682:18)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at bindings (/dist/cnc/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/dist/cnc/node_modules/serialport/lib/bindings/linux.js:2:36)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/dist/cnc/node_modules/serialport/lib/bindings/auto-detect.js:16:22)
```

#### Versions

- CNCjs: 1.9.20, latest

#### How Do You Install CNCjs?

Docker

#### Hardware

- [x] Desktop or Laptop

#### Operating System

- [x] Linux

voneiden, 2019-07-14:
On a Linux host you can get the image rolling with the following dockerfile

```
FROM cncjs/cncjs:v1.9.20
RUN apt-get update
RUN apt-get install -y udev          
RUN ./node_modules/.bin/electron-rebuild -f -v 2.0.18 --module-dir /dist/cnc/
```

Explanation: 1.9.20 uses nodejs v8 as the system default, but the build uses a electrum version that uses a newer and custom nodejs version that is incompatible with the system. When attempting to run the image, it uses again the system nodejs and therefore fails. The dockerfile above recompiles the serialport dependency against an older electrum version which is compatible with the system version (ABI 57).

Note: This "fix" will not work with the latest version, because latest raises the system nodejs to v10 (ABI 64) and there's no matching ABI from electron.

antwal, 2019-08-20:
MacOS Sierra 10.12.6 - Error:

```
$ docker run --privileged -p 8000:8000 --rm --name cncjs cncjs/cncjs:latest
Error: Error: The module '/dist/cncjs/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 69. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:805:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at bindings (/dist/cncjs/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/dist/cncjs/node_modules/@serialport/bindings/lib/linux.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/dist/cncjs/node_modules/@serialport/bindings/lib/index.js:14:22)
```

MitchBradley, 2019-08-21:
Have you tried building directly on Mac OS instead of using docker?

If you have node and npm installed on Mac, you could just do:

$ npm install cncjs@latest

Using docker and electron just adds layers of complexity - and things to go wrong - to an already-complex build process.

Actually, it is likely that the following fast-installl instructions https://github.com/cncjs/cncjs/issues/357#issuecomment-522692724 would work on Mac OS too.  The cncjs build process creates a rather large set of files that are platform independent with the single exception of a platform-dependent executable for handling low-level serial port access.  Doing a full build on Raspberry Pi is horribly slow, and often fails completely due to resource exhaustion, so I have been looking at ways to streamline the process.  I determined the minimal set of files needed to run cncjs (after building under Linux), then created a tarball with just those files. The tarball is 12 MB, which is tiny compared to the other release images. You can just expand that tarball on another machine, then rebuild the serialport module for your machine.  On a slow Pi, it reduces the cncjs update time from hours to a few minutes.

The tarball method does not get you the electron app.  You have to run the cncjs server then connect to it with a browser.  For those unfamiliar with electron, it is a way of bundling a nodejs server process and a browser inside a single executable.  When you run that executable, it starts both the server process and a browser process (separate from browsers that you might already have running) and connects them.  It appears to be an old-school app where you run it and the UI just pops up - albeit after a time delay because so many different (and large) programs need to start up.  There is a lot that can go wrong.

MitchBradley, 2019-08-21:
If you are dead set on using the docker method, I think I know of a way to fix it.  You will need to run the docker container interactively:

`docker run -i -t --privileged -p 8000:8000 --name cncjs cncjs/cncjs:latest /bin/bash`

(Note no --rm option.  With that option, it would not be possible to save the following changes.)

Then, inside the running container:

```
root@26501939a4ba:/# npm uninstall dist/cncjs/node_modules/@serialport/bindings
root@26501939a4ba:/# npm install dist/cncjs/node_modules/@serialport/bindings
root@26501939a4ba:/# bin/cncjs
```
Back in the docker shell, you can save the changes to a new image:

$ docker commit cncjs cncjs-fixed

And later run the new cncjs-fixed image.

$ docker run --privileged -p 8000:8000 --rm --name cncjs cncjs-fixed

There is probably a way to do it all from the docker command line instead of having to go interactive, but I am not motivated to figure it out.

antwal, 2019-08-21:
> Have you tried building directly on Mac OS instead of using docker?
> 
> If you have node and npm installed on Mac, you could just do:
> 
> $ npm install cncjs@latest
> 
> Using docker and electron just adds layers of complexity - and things to go wrong - to an already-complex build process.
> 
> Actually, it is likely that the following fast-installl instructions [#357 (comment)](https://github.com/cncjs/cncjs/issues/357#issuecomment-522692724) would work on Mac OS too. The cncjs build process creates a rather large set of files that are platform independent with the single exception of a platform-dependent executable for handling low-level serial port access. Doing a full build on Raspberry Pi is horribly slow, and often fails completely due to resource exhaustion, so I have been looking at ways to streamline the process. I determined the minimal set of files needed to run cncjs (after building under Linux), then created a tarball with just those files. The tarball is 12 MB, which is tiny compared to the other release images. You can just expand that tarball on another machine, then rebuild the serialport module for your machine. On a slow Pi, it reduces the cncjs update time from hours to a few minutes.
> 
> The tarball method does not get you the electron app. You have to run the cncjs server then connect to it with a browser. For those unfamiliar with electron, it is a way of bundling a nodejs server process and a browser inside a single executable. When you run that executable, it starts both the server process and a browser process (separate from browsers that you might already have running) and connects them. It appears to be an old-school app where you run it and the UI just pops up - albeit after a time delay because so many different (and large) programs need to start up. There is a lot that can go wrong.

I prefer to use docker, which is much easier to use for tests than creating dependency problems or downgrading packages on nodejs directly

> If you are dead set on using the docker method, I think I know of a way to fix it. You will need to run the docker container interactively:
>
> `docker run -i -t --privileged -p 8000:8000 --name cncjs [...]

now it works but generates some errors; now I can make some changes in the code...

DominikPalo, 2020-04-21:
Any updates? @voneiden's solution works great (with the version 1.9.20), but I would like to update my container to use the latest version (1.9.22)

rhass, 2020-08-06:
I have gone with a different approach of just installing the cncjs package from the npm repos instead of source for now. For development purposes it is not ideal. However, it installs the latest release into a container and functions properly.

Eg:

```dockerfile
FROM node:lts

RUN export debian_frontend=NONINTERACTIVE \
    && apt-get update && apt-get install -y udev \ 
    && apt-get autoclean -y && apt-get autoremove -y \
    && npm i npm@latest -g \
    && npm install -g cncjs@latest --unsafe-perm

EXPOSE 8000/tcp
CMD ["/usr/local/bin/cncjs"]
```

poldim, 2020-12-28:
Just a point of reference, I created the Dockerfile based on @voneiden's recommendation, and it appears to come up fine.  I was previously getting the same error.

```
[node2] (local) root@192.168.0.27 ~
$ mkdir cncjs
[node2] (local) root@192.168.0.27 ~
$ cd cncjs/
[node2] (local) root@192.168.0.27 ~/cncjs
$ vi Dockerfile
[node2] (local) root@192.168.0.27 ~/cncjs
$ docker build -t cncjs/cncjs:rebuild .
Sending build context to Docker daemon  2.048kB
Step 1/4 : FROM cncjs/cncjs:v1.9.20
v1.9.20: Pulling from cncjs/cncjs
e79bb959ec00: Pull complete 
d4b7902036fe: Pull complete 
1b2a72d4e030: Pull complete 
d54db43011fd: Pull complete 
69d473365bb3: Pull complete 
6e2490ee2dc8: Pull complete 
24bcfd3d8296: Pull complete 
4c485b32137c: Pull complete 
10c7ea03b1e0: Pull complete 
7da3ff434f8a: Pull complete 
306d2f7957ad: Pull complete 
4ce325a7a1f4: Pull complete 
Digest: sha256:51a130be239e7a1686125eb484d921ab5e70dd5b9f227d4dc3b55b864142103f
Status: Downloaded newer image for cncjs/cncjs:v1.9.20
 ---> c5e90b6854f0
Step 2/4 : RUN apt-get update
 ---> Running in 4acc64b91b17
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://security.debian.org/debian-security stretch/updates InRelease [53.0 kB]
Get:3 http://deb.debian.org/debian stretch-updates InRelease [93.6 kB]
Get:4 http://deb.debian.org/debian stretch Release [118 kB]
Get:5 http://deb.debian.org/debian stretch Release.gpg [2410 B]
Get:6 http://security.debian.org/debian-security stretch/updates/main amd64 Packages [637 kB]
Get:7 http://deb.debian.org/debian stretch-updates/main amd64 Packages [2596 B]
Get:8 http://deb.debian.org/debian stretch/main amd64 Packages [7080 kB]
Fetched 7986 kB in 2s (3715 kB/s)
Reading package lists...
Removing intermediate container 4acc64b91b17
 ---> a99988257f00
Step 3/4 : RUN apt-get install -y udev
 ---> Running in d499ed128a24
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libkmod2 libudev1
The following NEW packages will be installed:
  libkmod2 udev
The following packages will be upgraded:
  libudev1
1 upgraded, 2 newly installed, 0 to remove and 124 not upgraded.
Need to get 1287 kB of archives.
After this operation, 7353 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian stretch/main amd64 libkmod2 amd64 23-2 [48.1 kB]
Get:2 http://deb.debian.org/debian stretch/main amd64 libudev1 amd64 232-25+deb9u12 [126 kB]
Get:3 http://deb.debian.org/debian stretch/main amd64 udev amd64 232-25+deb9u12 [1113 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 1287 kB in 0s (25.7 MB/s)
Selecting previously unselected package libkmod2:amd64.
(Reading database ... 29978 files and directories currently installed.)
Preparing to unpack .../libkmod2_23-2_amd64.deb ...
Unpacking libkmod2:amd64 (23-2) ...
Preparing to unpack .../libudev1_232-25+deb9u12_amd64.deb ...
Unpacking libudev1:amd64 (232-25+deb9u12) over (232-25+deb9u9) ...
Setting up libudev1:amd64 (232-25+deb9u12) ...
Selecting previously unselected package udev.
(Reading database ... 29984 files and directories currently installed.)
Preparing to unpack .../udev_232-25+deb9u12_amd64.deb ...
Unpacking udev (232-25+deb9u12) ...
Setting up libkmod2:amd64 (23-2) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Setting up udev (232-25+deb9u12) ...
Adding group `input' (GID 102) ...
Done.
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Removing intermediate container d499ed128a24
 ---> da028344861e
Step 4/4 : RUN ./node_modules/.bin/electron-rebuild -f -v 2.0.18 --module-dir /dist/cnc/
 ---> Running in 7a35273dc1b9
- Searching dependency tree
✔ Rebuild Complete
Removing intermediate container 7a35273dc1b9
 ---> c4759795256e
Successfully built c4759795256e
Successfully tagged cncjs/cncjs:rebuild
[node2] (local) root@192.168.0.27 ~/cncjs
$ docker run --privileged -p 8000:8000 --name cncjs cncjs/cncjs:rebuild
2020-12-28T23:00:45.264Z - info init Loading configuration from "/root/.cncrc"
2020-12-28T23:00:45.429Z - info init Starting the server at http://172.17.0.2:8000
```

Dockerfile for ref:

```
FROM cncjs/cncjs:v1.9.20
RUN apt-get update
RUN apt-get install -y udev          
RUN ./node_modules/.bin/electron-rebuild -f -v 2.0.18 --module-dir /dist/cnc/
```

isaiah-v, 2021-01-07:
Based on the install instructions I was able to get the latest version working (1.9.22) on node 12

Dockerfile
```
FROM ubuntu:latest

RUN apt-get update
RUN apt-get -y install curl gnupg
RUN apt-get -y install git

RUN adduser --disabled-password --gecos "" node
USER node

ENV NVM_DIR=/home/node/.nvm

RUN git clone https://github.com/creationix/nvm.git $NVM_DIR;
RUN cd $NVM_DIR; git checkout `git describe --abbrev=0 --tags`;
RUN . "$NVM_DIR/nvm.sh"; \
    nvm install 12; \
    nvm use 12; \
    npm install -g cncjs@1.9.22;

CMD . $NVM_DIR/nvm.sh && cncjs;
```

proffalken, 2021-01-18:
It would be great to get this rolled in to the "official" Docker container at https://hub.docker.com/r/cncjs/cncjs - I'm not sure if `feature-next` is what I should be using, but `:latest` definitely has this issue.