cncjs/cncjs #490: Pendant API commands - ege start/stop/pause

bradanlane, 2019-06-03:
#### Description

I have a NECS controller attached to the Raspberry Pi running CNCjs. I use it to home/jog/set W0,0 etc.

I can not figure out how to send operation commands such as start/stop/pause.

I assume there is GRBL already loaded. I want to have be able to start/stop the job from the my NECS pendant app.

#### Versions

- CNCjs: 1.9.15
- Node.js: 10.15.3
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [] NPM
- [ ] Download the CNCjs Desktop Application
- [x] Raspberry Pi image

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-06-04:
@bradanlane

Sorry for that it's not well documented in the wiki, you can see all supported commands in the controller code:
https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Grbl/GrblController.js#L1029-L1330

Some examples here:
```js
socket.emit('command', options.port, 'gcode:load', name, gcode);
socket.emit('command', options.port, 'gcode:start');
socket.emit('command', options.port, 'gcode:pause');
socket.emit('command', options.port, 'gcode:resume');
socket.emit('command', options.port, 'gcode:stop');
socket.emit('command', options.port, 'cyclestart');
socket.emit('command', options.port, 'feedhold');
```

bradanlane, 2019-06-04:
Thank you for the reply and answer. CNCjs is a valuable project for me.

closed 2019-06-04 by bradanlane