cncjs/cncjs #491: No communication option other than COM1

grumpytwo, 2019-06-11:
No communication option other than COM1

I have just loaded the latest version of cncjs onto my desktop along with the latest version of Node.js. In the comms widget there is only COM1 available and no USB option, how do I get a USB option?

Versions

- CNCjs: 1.9.x20
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ x] Windows 10
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-06-11:
On Windows, when you plug in a USB-to-serial device, it appears as a COM port with a name like COM1. You can see more information about them in the Windows Device Manager.
[image]
There is no indication in CNCjs as to whether a given COM port is implemented via USB - CNCjs just shows the base Windows COMn name - but since modern PCs rarely include old-style motherboard-resident COM ports, it is a good bet that any COM port on a modern PC is a USB-to-serial adapter.

grumpytwo, 2019-06-14:
Thanks for that Mitch it is most appreciated. We run a small charity helping people with a disability back into work by teaching them workshop skills, how to handle the paperwork, use generic machinery etc. We bought a little 3018 and a bigger Shapeoko to try and teach some computer and CNC skills. We are trying to add a Raspberry Pi to the 3018 and this was the first problem we encountered. Unfortunately none of us are IT savvy and are just muddling along but we will get there, so thank you again for your help.

MitchBradley, 2019-06-14:
I applaud what you do.  I spent 7 years on the team that made the One Laptop Per Child computer for kids in developing countries.
That said, there is a certain irony in the non-IT-savvy teaching computer skills. But I suppose there is value in people stumbling along together in a journey of discovery. Experts often make poor teachers as the gulf between their knowledge and the student's is too wide to bridge.
I am a little concerned about the use of small machines to teach workshop skills. Hobbyist machines are relatively safe - they can hurt you but due to their low power they probably won't hurt too bad.  In contrast, I recently repaired a CNC bending brake in a friend's sheet metal shop.  It was about 15 feet long and 8 feet high, and powerful enough to chop right through your arm without slowing down. I saw a Facebook video the other day where a CNC fly cutter spun too fast and came apart, destroying a heavy vise.  A piece then shot through a cinder block wall, ricocheted off a metal chair (bending it), then bounced off something else and disappeared inside another heavy wall.  The point being that the key workshop skill is how not to kill yourself or someone else. Small machines can lull you into the wrong mindset.
Good luck with the Raspberry Pi.  I was a computer designer for 45 years, so my computer skill level is close to "off the charts", yet I find Raspberry Pi's frustrating to administer.  But they are better than a lot of the alternatives, so you do what you have to do.

closed 2019-06-16 by MitchBradley