cncjs/cncjs #492: Node.js Installation Issues on Pi Zero

calummmm, 2019-06-16:
#### Description

Hey there,

_*disclaimer - I'm fully aware that the pi zero is not recommend or supported.*_ However I was determined to try to use my pi zero after seeing a [tutorial](https://blog.shahada.abubakar.net/post/adding-a-raspberry-pi-zero-w-and-cnc-js-to-the-3018-woodpecker-cnc-router) documenting the installation. I managed to get the CNC.js web app working in the end, so thought I could spare some people a bit of frustration.

### Installing Node.js on Pi Zero

1. Node Package Manager - I took this route and it didn't work, as there are components of the packages that do not support armv6 (as is used on the pi zero, as opposed to armv7 and v8 on the raspi B, B+ I presume). After much faff I went for the Node Version Manager route.

2. Node Version Manager - I found that `nvm install 4` caused some issues with my `npm` commands further in to the installation. After installing the latest version of nvm `nvm install 10`, uninstalling the older node versions (deleting from directory) and performing the 'nvm alias default node' command to point to the newest installed Node version, I was able to perform a successful installation of npm and upgrade it. In hindsight perhaps `nvm use 10` would have worked too.

I'm new to the pi, so had to do a lot of googling to get through this - I thought it was worth posting on here to warn / inform people for future installation. In summary/**TL;DR**, if you're having troubles installing Node.js on the pi zero, **try using** `nvm install 10` **and** `nvm use 10` **rather than version 4.** Of course always follow the official set-up guide first.

Hope that helps,

-C

#### Versions

- CNCjs: 1.9.20
- Node.js: 10.16.0
- NPM: 6.9.0

#### How Did You Install CNCjs?

- [ ] NPM
-[x] NVM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi Zero
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2019-06-17:
I have been unable to install cncjs on a Pi Zero using the "npm install" path.  After hours of cranking away, it always ends up running out of memory before giving up.  The installation path that works for me is to compile everything else on on a PC, copy the files from the PC over to the Pi except for the serialport module, then compile just the serialport module on the Pi.  There are instructions for how to do this in another issue.  I will find them and report back as time permits.
This technique is much much faster than compiling everything on Pi, even if the Pi compilation would eventually succeed.  The difference is a few minutes compared to several hours.
The Pi Zero works just fine for the CNCjs server component.  I use an Android tablet for the UI component.

MitchBradley, 2019-06-17:
Here are the steps for split compilation: https://github.com/cncjs/cncjs/issues/357#issuecomment-490004227

calummmm, 2019-06-17:
Great read Mitch. I've seen another website with CNC software (CNC.js, Chillipeppr, Universal G-code Sender) pre-flashed on a [raspberry pi SD card image](https://wiki.protoneer.co.nz/Raspberry_Pi_CNC_User_Interface_SD_Card_Image). I thought about taking this route should my pi zero installation have failed. 

That said, it wasn't CPU or memory that limited my install, but compatibility issues with running the npm command. May be worth updating the CNC.js raspi install guide with instructions for your split compilation method.

Also - have you installed the UI component separately? Or are simply accessing the UI from a tablet (as I am currently accessing the UI from my macbook).

MitchBradley, 2019-06-17:
The problem with pre-flashed SD cards is that it is highly unlikely that the card will have the latest version of what you want.  That card you mentioned has cncjs 1.9.15; the latest tag is 1.9.20 and I am working off 1.10 at the moment.  The same problem happens with Docker images.  They are great for the first try, but as soon as you need to upgrade, you are back to compiling yourself, unless the maintainer of the Docker image is always keeping up (which is sort of rare in the open source volunteer space).
It would definitely be worth updating the install guide, but there is only so much time in a day and I am deep in a refactor of the CNCjs controller code.
Regarding the UI - I'm not sure what you mean by "installing separately". When you do "npm install", it compiles both the "server" and "app" components.  Both of them are resident on the Pi filesystem.  The server component executes on the Pi under Node.js, while the app component (UI) is presented to the browser, wherever it may be running, via HTTP.  The app then executes in the browser's JavaScript environment on the browser computer. So, regardless of which computer runs the browser, both the server and app (UI) components are "installed" on the Pi.
That said, it is possible to compile them separately, perhaps on different computers, and then copy them around.  Before I discovered the final optimization of just compiling the serialport module on the Pi, I compiled the entire server component on the Pi and compiled the app on a PC.  Compiling the server on Pi took a long time, but ultimately succeeded, while compiling the app on Pi always crashed after several hours of thrashing.
For production use, I do not use the normal CNCjs UI, instead preferring cncjs-shopfloor-tablet, which I wrote according to my requirements.  It has big buttons that are easy to hit on a tablet or touchscreen, does not scroll so the buttons are always on-screen, and avoids distracting information that is not relevant to running a job.  When you are using a powerful machine and something goes wrong, you need to access the controls immediately, rather than having to find the right place on the screen to let you scroll, then scroll so the button you need is visible, then hit a small button.

calummmm, 2019-06-19:
Thanks for the reply - yes, I was clarifying this:
> So, regardless of which computer runs the browser, both the server and app (UI) components are "installed" on the Pi.

The reason being was that I was a little confused when you said:
>  I use an Android tablet for the UI component.

I'll take a look into the shop floor tablet version. Looking forward to v1.10! Albeit dreading the install...

closed 2019-06-19 by calummmm

EvilHacker, 2022-08-22:
Hi [calumkenny](https://github.com/calumkenny),

I am wondering if you were able to run CNC.js successfully on Pi Zero.

I, like you, am aware that CNC.js is recommended to only be run on a Pi 3 or better.

I just find it hard to believe that that much horse power is needed. The current Pi Zero 2 W seems like a very capable machine. There is also a big price difference between Pi Zero and Pi 3/4.

The Pi just needs to serve the web page (to be run in a browser on another machine). I would think that the only realtime capability required is to keep up with USB communications (which end up going to a low-speed UART on the other side). 

What else would CNC.js be doing that Pi Zero could not handle?

MitchBradley, 2022-08-22:
It does work in server-only mode on Pi Zero W.  I used it for some years on a 0W with the UI running on an Android tablet.  There are some installation challenges though.  If you try to compile if on a Pi 0, there is a tendency for it to take forever and eventually crash due to memory or swap space exhaustion.  You cannot easily compile it on another machine because the Pi 0 used a different ARM CPU generation from later Pis, so the serialport module which needs to be compiled to native code will not work.  There is a way around that by compiling only the serialport module on the 0W, then copying everything else over from a bigger machine.  One of the issues has a recipe for doing that.  Basically, it can be made to work, but you have to know how to work around a lot of niggling details about the long and involved NPM compilation process.

MitchBradley, 2022-08-22:
In fact, the recipe in question is linked in one of my comments on this issue.

EvilHacker, 2022-08-23:
Hi [MitchBradley](https://github.com/MitchBradley),

Oh, so you **CAN** run CNC.js in server mode on Pi Zero.

This is exactly what I want to do. I have been using CNC.js on my laptop and would like to continue doing so. I just want to be able to do so without having my laptop tethered to my CNC machine and always on.

Thank you for figuring this out. I will try to replicate what you have done on a Pi Zero 2 W.