cncjs/cncjs #495: stepper motors do not lock automatically- Morrill

sammorrill, 2019-06-19:
my stepper motors will not lock when i start  the program.  Is the a way to add the "M17" command so it automatically happens?

karoria, 2019-06-19:
Make sure you use correct capacity motor and drive. SMPS used to power up drives should give sufficient amperes for holding. I feel there is nothing related to cncjs here. It is merely a hardware thing. Further, some drives have advance features of current management like it reduces current when there is no motion in particular axis. Check those settings also.

MitchBradley, 2019-06-19:
The Marlin configuration file configuration.h lets you choose whether motors are disabled when not in use.  You can set "DISABLE_X" (and Y and Z) to false, or you can set the enable pins like "X_ENABLE_PIN" to nonexistent pin numbers like -1.

For additional help with Marlin try asking in a Marlin forum.

MitchBradley, 2019-06-27:
@sammorrill - have you resolved your stepper locking issue yet

closed 2026-03-13 by cheton