cncjs/cncjs #501: Pokeys CNC Controller Support

GorillaMachines, 2019-07-18:
#### Description

I am looking to get cncjs to control a pokeys CNC control board. They have alot of documentation on how to interface their board with many softwares and wanted to see if cncjs could adapt to use this board. It is an incredible board that has a massivly versitle set of use cases. 

Here is a link to the control board, they have alot of information on their site. Thanks everyone!

https://www.poscope.com/product/pokeys57cnc/

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-07-18:
It would take a lot of work to adapt CNCjs to this board because the board uses Ethernet whereas CNCjs, in its current form, is pretty specific to serial protocol controllers.  It could be done, in the sense that, given enough time and effort, any program can be changed to do pretty much anything.  But it would be a lot of work and I doubt that any of the main CNCjs developers have much incentive to do it.

MitchBradley, 2019-07-19:
I read the PoKeys protocol docs.  I am very impressed.  They have done a very professional job of implementing a highly-featured Ethernet and/or USB connected control engine. It is a big improvement over the Arduino-derived serial controllers (GRBL, etc) that CNCjs supports.  But, having seen it, my estimate of the difficulty of adding CNCjs support for it has increased.  CNCjs is essentially a "GCode sender" program, delegating the job of GCode interpretation and things like corner planning to the controller.  The PoKeys device splits the software responsibility differently, essentially at the Mach 3 "pulsing engine" interface point.  So a fair amount of code would have to be added to CNCjs to implement a controller with a different kind of interface.  It would be do-able, but a lot of work and testing.

GorillaMachines, 2019-07-28:
Thanks so much for the comments. Ever since coming across the Pokeys boards I have been blown away by their functionality, and am trying to find a way to design a control software for the boards that bypasses the restrictions of mach 3 and fully integrates the tools that pokeys can offer. I believe that I will begin more research on what exactly would need to be done to make such a software as that would really further the cnc space. I have designed a PCB that directly integrates with the pokeys board and allows control of a CNC Plasma machine. Currently it is controlled by a skinned version of mach 3, but I very much so want to make a modular software that integrates this boards abilities. Thanks for the comments and if you have any ideas let me know!

[image]

MitchBradley, 2019-07-28:
What is the Arduino for?

GorillaMachines, 2019-07-28:
It is used to interface the pokeys board to the Trinamic Stepper drivers via an spi connection between arduino and stepper drivers, and an i2c connection between the arduino and pokeys. This allows me alot of granular control of the functionality of the drivers, such as load sensing, automatic current changes based on load or speed, and much more.

GorillaMachines, 2019-07-28:
Do you think it would be possible to use an arduino, or pokeys onboard ability to interpret and send serial commands, to locally process serial commands received from cncjs and interpret that into movement commands sent to the pokeys board, or is that better suited for a pc based software? The pokeys boards have the ability to control motion while at the same time run any set of written software along side it, this could be used to possibly make a local gcode interpreter. Just some thoughts.

MitchBradley, 2019-07-28:
How would the arduino communicate with the pokeys board?  It seems that the pokeys boards speak via ethernet or USB, and arduinos typically have neither an Ethernet nor a host mode USB.  

Maybe LinuxCNC could be modified to talk to PoKeys.

Is there any documentation for how to run other software on the PoKeys boards, or to write new plugins?  I couldn't find any info on their website.  It might be possible to inject the GCode interpreter from Marlin 2.0 into the PoKeys software.  If I had enough time I'm sure I could crack that nut, but the difficulty could range from easy to really really hard depending on whether they support user-written plugins.

Do you have a budget for this?

GorillaMachines, 2019-07-28:
Pokeys has an interface that they call poblocks. It is basically a visual based coding platform where you can drag and drop functions, as well as make any function you like. Through this interface you can write programs that can fully control all of the movement functions of the board. With the arduino communicating via i2c with poblocks you can use an arduino to directly control any movement of the board. This can be implemented easily to move to any position at a set speed and acceleration, all through poblocks. Any integer, or rather any register on the poblocks board is available to read and write from with an arduino, functions, or anything you really could think of through the poblocks interface. I think that this method would be a high speed viable solution. If the arduino could interpret the gcode coming from cncjs, then it could send the direct movement commands to pokeys very easily through this mechanism.

GorillaMachines, 2019-07-28:
I have tested this with code that is currently running my machines Z axis depending on some sensor values that are read and interpreted by an Arduino. The arduino can send the movement info over i2c, and the pokeys board listens and maps the incoming value to its motion control system. It has proven to be a very low latency solution. From my understanding mach 3 is doing something very similar, it is sending a constantly changing target position to pokeys and the pokeys motion controller interprets that target and moves the motors accordingly.

GorillaMachines, 2019-07-28:
The question for me is can GRBL, instead of sending steps and directions, send a live target position, feed, and acceleration? Are these values easily available within grbl? If so then that could be a viable solution.

MitchBradley, 2019-07-28:
I would be tempted to use Marlin 2.0 running on a Teensy or ESP32 or some other 32-bit chip, since AVRs are pretty much out of memory if you look at them sideways.  As to whether the values are "easily available", it depends on how good a programmer you are.  All of these controller firmwares are intended to control steppers so they are optimized around the difficult problem of mapping GCode into stepping via kinematics.  The relevant positions and speeds are present at some level, but teasing it out of the overall framework can be gnarly.