cncjs/cncjs #503: Custom Keyboard Shortcuts

ione2k, 2019-07-21:
#### Description
Hi All,

Apologies in advance if I am posing in the wrong place. I could not find a CNCjs forum. I just came across CNCjs as I was looking for a Grbl GUI that can interpret (do something) with M6 Tx (Tool changes). I have been using for a few months UGS Platform which works well but does not recognise M6 commands. To my surprise I found that CNCjs appear to have an active community and already implemented M6 Manual tool changes and some people are working on automatic tool changes which I also want to achieve in the future. The User interface also looks great. 

The only BUT that I have at the moment is that I use a Razer Tartarus v2 as a pendant which control almost everything (except macros) in UGS Platform as it is quite uncomfortable to go back and forth from the PC to the machine to check positions, place a touch plate, etc. The keypad has about 20 keys, plus a joystick, wheel, 3 profiles, etc. which I would like to take advantage linking as many keyboard shortcuts as possible. 

[image]

In my initial CNCjs tests I found that the shortcuts are very limited to basic functions. Is there a way to customise or add shortcuts? I am missing functions like Z probe, macros, connecting to the serial port, feed rate, spindle rates mods, Zero X Y Z, etc.

Thank you in advance for your help and congratulations to all involved for the great work so far.

#### Versions

- CNCjs: 1.9.20

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-07-21:
A better way is to create a pendant specifically made for your keyboard and add shortcuts for those operations. There is a pendant project [cncjs-pendant-keyboard](https://github.com/cncjs/cncjs-pendant-keyboard) which is similiar to your scenario for using a keypad to control the movement. This pendant uses [node-hid](https://github.com/node-hid/node-hid) to [access USB HID devices](https://github.com/node-hid/node-hid#list-all-hid-devices-connected), you might need a little background for programming with Node.js.

**Sample code**
https://github.com/cncjs/cncjs-pendant-keyboard/blob/master/bin/cncjs-pendant-keyboard#L108-L211

For questions to the code, I think @nsfilho can help you get started.

ione2k, 2019-07-22:
Hi @cheton , thank you very much for the quick reply. I have very little knowledge about coding/programming but I am willing to give it a go. Just a question, from what I can see in the sample code it looks like the pendant sends Grbl codes directly to the controller instead of instructing CNCjs to do something? Does this mean that I have to somehow write macros inside the pendant instead of creating them in CNCjs and then instructing to execute them through the pendant?

On a separate note, why the list of shortcuts in CNCjs cannot be expanded (as per UGS) instead with an option to assign any keyboard key or combination of keys so anybody can create a pendant with any hardware and use the driver of the keyboard/joystick to emulate the combination of keys as I do with the Razer for UGS? As an example, razer driver let you even assign/emulate clicks on a specific part of the screen to a given key in the hardware (which I use for some functions that USG does not have shortcut for), and which is my plan B if I cannot achieve what you are suggesting in CNCjs.

cheton, 2019-07-23:
For your first question, you can either send data to Grbl directly r instruct CNCjs to run commands. These APIs are not explicitly recorded on the wiki page, I will take some time to update this part.

#### Write Mode
socket.emit('write', port, line + '\n')

#### Command Mode ([Grbl Controller](https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Grbl/GrblController.js#L1031-L1330))
socket.emit('command', port, cmd, ...args)

Available commands:
* cmd="gcode:load", args=[name, gcode, context, callback]
* cmd="gcode:unload", args=none
* cmd="gcode:start", args=none
* cmd="gcode:stop", args=[options]
* cmd="gcode:pause", args=none
* cmd="gcode:resume", args=none
* cmd="feeder:feed", args=[commands, context]
* cmd="feeder:start", args=none
* cmd="feeder:stop", args=none
* cmd="feedhold", args=none
* cmd="cyclestart", args=none
* cmd="statusreport", args=none
* cmd="homing", args=none
* cmd="sleep", args=none
* cmd="unload", args=none
* cmd="reset", args=none
* cmd="feedOverride", args=[value]
* cmd="spindleOverride", args=[value]
* cmd="rapidOverride", args=[value]
* cmd="lasertest:on", args=[power, duration, maxS]
* cmd="laertest:off", args=none
* cmd="gcode", args=[command, context]
* cmd="macro:run", args=[id, context, callback]
* cmd="macro:load", args=[id, context, callback]
* cmd="watchdir:load", args=[file, callback]

cheton, 2019-07-23:
> On a separate note, why the list of shortcuts in CNCjs cannot be expanded (as per UGS) instead with an option to assign any keyboard key or combination of keys so anybody can create a pendant with [...]

UGS is a standalone Java application, it can control everything including defining the shortcut key. Support for user-defined shortcuts in CNCjs main UI is doable, but this may override browser's keyboard shortcuts. I will consider adding this feature request to 1.10, but it may not be a higher priority task.

ione2k, 2019-07-24:
Thank you very much for your reply. It would be great if you can have a look at that.

joshgibb, 2020-06-26:
ione2k How did you get on with creating hotkeys Within cncjs for your pendant. I am currently wantIng to also do this but using a Stream Deck as the pendant.
I also dont have any experience with coding or programming so would be good to know how you got on.

thiaggo, 2020-09-22:
yeah, same thing here,  i would be nice to have programable custom hotkeys for begginers like me,   if thats is too much work to do, i think puting hotkeys for probing would be good   thank you

thiaggo, 2020-09-22:
and yep, i tried to understand the code and i failed miserably (not a coding person)  @ione2k what did you do? heeelp

ione2k, 2020-09-23:
Hi @thiaggo,

I developed an interest on CNCJS because of the possibility of having some sort of Auto Tool Change, but I have postponed that project for a while meanwhile I increase the size of my CNC machine. I am using USG Platform instead which is not perfect but it has more options than CNCJS and the community appears to be a bit more active, USG Platform does all what I need except capacity for ATC which I do not need at the moment.