cncjs/cncjs #508: Soft limit problems | inverted working area

jasinghuang, 2019-08-14:
#### Description

Hi,
I am very new to cnc software and grbl, I am very appreciate in advance. 

I have a 6040 CNC machine, and tried to set the soft limit, however it didn't work as I think.
[image]

In theory, the red cuboids area should be the work area, but in fact it is in the blue cuboids area.

I have tried many possible of $20, $23. It is just not working. Could someone tell me where the wrong setting?

Here is my config.
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=1 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=1 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1000.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=5.000 (Homing switch pull-off distance, millimeters)
$30=1000 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$80=0
$81=0
$82=2
$100=640.000 (X-axis travel resolution, step/mm)
$101=640.000 (Y-axis travel resolution, step/mm)
$102=800.000 (Z-axis travel resolution, step/mm)
$110=1200.000 (X-axis maximum rate, mm/min)
$111=1200.000 (Y-axis maximum rate, mm/min)
$112=600.000 (Z-axis maximum rate, mm/min)
$120=20.000 (X-axis acceleration, mm/sec^2)
$121=20.000 (Y-axis acceleration, mm/sec^2)
$122=20.000 (Z-axis acceleration, mm/sec^2)
$130=400.000 (X-axis maximum travel, millimeters)
$131=600.000 (Y-axis maximum travel, millimeters)
$132=100.000 (Z-axis maximum travel, millimeters)

ok
feeder> G91
feeder> G0 X100
feeder> G90
ok
ALARM:2 (Soft limit)
[MSG:Reset to continue]

[image]

Thank you.

#### Versions

- CNCjs: 1.9.20
- Node.js: v10.16.2
- NPM: 6.9.0

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [x] Mac
- [ ] Linux

TobyGWilliams, 2019-08-14:
It looks like you are working in negative area for X, Y and Z. Therefore I would suggest setting the x travel to be -100 to 0 instead of 0 to 100. 

This should make the visualisation of the cuboid match the blue lines.

jasinghuang, 2019-08-14:
@TobyGWilliams 
Thanks for the quick reply.
However when I change $3 to 6 ($3=6), it does the same.

AustinSaintAubin, 2019-08-14:
I think was you are wanting to do is change your Machine Reference Position when Homed.
To do that you need to uncomment [HOMING_FORCE_SET_ORIGIN in grbl/config.h](https://github.com/gnea/grbl/blob/b75e5571eeaeb22a88304716fb1e7411f9c28be0/grbl/config.h#L129), then [compile & flash GRBL](https://github.com/gnea/grbl/wiki/Compiling-Grbl) with the [Arduino IDE](https://www.arduino.cc/en/Main/Software) (or other IDE) to your controller.

Have a looks at these article. 
 - [GRBL: Frequently-Asked-Questions](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to)
 - [Inventables: grbl-and-changing-machine-position-reference](https://discuss.inventables.com/t/grbl-and-changing-machine-position-reference/58132)

jasinghuang, 2019-08-15:
@AustinSaintAubin 
Thank you for the solution, I think I understand how it works finally.
Knowing it is not a mistake setting by me, I'll just leave it.

closed 2019-08-23 by AustinSaintAubin