cncjs/cncjs #509: Discord / Slack / IRC community?

arsdragonfly, 2019-08-16:
Should we set up a channel/workspace in order for people to share their experiences and/or seek help?

MitchBradley, 2019-08-16:
There is already a Facebook group in addition to this "forum via GitHub issues".  I personally do not want to have to monitor a third channel in addition to those other two.  Even having two is one too many, because I often forget where some past item arose and have to search two different places;

saxa, 2019-08-16:
I also agree with Mitch, the positive part of communicating on some forum is the simplicity of checking and posting stuff quickly. But there is far way negative things on the other side, forums usually are full of spammers, out of subject things etc... , IRC is good for live chat within the developers and users, and dev to dev too. I think this is not the case here, as the community is quite small. So for me also the good thing is to keep it as it is.

closed 2026-03-13 by cheton