cncjs/cncjs #510: CNCjs with

leoheck, 2019-08-19:
Hello guys.

I'm having issues to connect the CNCjs with my GRLB printer.
It was installed on the Raspberry PI.
Any help would be great since I really want to use CNCjs.

The CNCjs can connect but I can't control the CNC, the arrows are greyed out.
Commands on the CNCjs' console return nothing.

However, the CNC is working since control through the serial console from the command line.

Someone told that installing node serialport with sudo may cause issues. However, I was not able to install the CNCjs without sudo. 

I got this [3018 CNC](https://www.amazon.com/Upgrade-MYSWEETY-Engraving-30x18x4-5cm-110V-240V/dp/B07JZ7PHF2/ref=sr_1_16?keywords=cnc+3018&qid=1566222103&s=gateway&sr=8-16
) recently.

I saw that my CNC was using GRBL 0.9j so I updated to use 1.1j but it didn't help much.
Also, connecting the printer on my computer I can control the printer with the bCNC software.

It is my first time working with CNCs. Also, I never installed node applications before.

#### Versions
> I don't have these values here with me but I will update it by the end of the day when I get home.

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2019-08-19:
One possibility is that you are choosing the wrong serial port instead of the one that is connected to the GRBL controller.  Without seeing a screenshot of what you are doing, I cannot guess further.

MitchBradley, 2019-08-19:
Another possibility is that the ownership of the serial port might not be correct, so cncjs does not have permission to access the port.  Try this:

```
whoami
ls -l /dev/ttyS0
groups
```

If the second command says ".. No such file or directory", replace /dev/ttyS0 with the name of your serial port.

leoheck, 2019-08-19:
Hello @MitchBradley 

I am selecting the right serial port (now it is `/dev/ttyUSB0`)

This is last lines from `dmesg` after connecting the CNC's usb cable.
```
[72304.084479] usb 1-1.3: new full-speed USB device number 6 using dwc_otg
[72304.217567] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.63
[72304.217581] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[72304.217591] usb 1-1.3: Product: USB2.0-Serial
[72304.221158] ch341 1-1.3:1.0: ch341-uart converter detected
[72304.223951] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

```
And on CNCjs 
[image]

The permissions of the serial port are the following, and my (`pi`) user seem to be in the `dialout` group too.
[image]

I am launching `cnc` command with my `pi` user manually.

I can also `chmod 777` as a test but nothing happens.

If you believe CNCjs software is good, I am almost sure it is an issue with the node installation since it is my first time trying to install it. I have no idea how to debug it.

leoheck, 2019-08-19:
While I was typing I got an answer from the CNC.

[image]

Maybe something is delaying the output

MitchBradley, 2019-08-19:
Try adding verbose flags to the command and showing me the console log, after browsing to the cncjs server and opening the serial port.  Here's what it looks like for me:

```
pi@raspberrypi-tinyg:~/cncjs-test $ bin/cncjs -vv
2019-08-19T22:57:56.797Z - info init Loading configuration from "/home/pi/.cncrc"
2019-08-19T22:57:56.875Z - info init Watching "/home/pi/GCode" for file changes
2019-08-19T22:57:59.482Z - debug app app.settings: {"x-powered-by":false,"etag":"weak","env":"production","query parser":"extended","subdomain offset":2,"trust proxy":true,"views":["/home/pi/cncjs-test/dist/cncjs/app","/home/pi/cncjs-test/dist/cncjs/server/views"],"jsonp callback name":"callback","view cache":true,"verbose errors":false,"case sensitive routing":true,"strict routing":false,"view engine":"html"}
2019-08-19T22:58:00.610Z - debug app assets: name=app, asset={"routes":["/e9e533b0/","/"],"path":"/home/pi/cncjs-test/dist/cncjs/app","maxAge":31536000000}
2019-08-19T22:58:00.619Z - debug app > route=app /e9e533b0/
2019-08-19T22:58:00.636Z - debug app > route=app /
2019-08-19T22:58:00.989Z - debug service:cncengine Loaded controllers: Grbl,Marlin,Smoothie,TinyG
2019-08-19T22:58:01.420Z - info init Starting the server at http://127.0.1.1:8000
2019-08-19T22:58:02.035Z - debug service:cncengine New connection from 192.168.2.3: id=kr_dl57dfaNRHoboAAAA, user.id=, user.name=
2019-08-19T22:58:26.751Z - debug app [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open '/home/pi/.cncjs-sessions/YxKI5InRopJmuWLjsM08JeRRTVcJ0yvM.json'
2019-08-19T22:58:26.803Z - debug app [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open '/home/pi/.cncjs-sessions/YxKI5InRopJmuWLjsM08JeRRTVcJ0yvM.json'
2019-08-19T22:58:26.855Z - debug app [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open '/home/pi/.cncjs-sessions/YxKI5InRopJmuWLjsM08JeRRTVcJ0yvM.json'
2019-08-19T22:58:26.907Z - debug app [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open '/home/pi/.cncjs-sessions/YxKI5InRopJmuWLjsM08JeRRTVcJ0yvM.json'
2019-08-19T22:58:26.958Z - debug app [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open '/home/pi/.cncjs-sessions/YxKI5InRopJmuWLjsM08JeRRTVcJ0yvM.json'
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET / 304 388.119 ms
2019-08-19T22:58:27.655Z - debug service:cncengine Disconnected from 192.168.2.3: id=kr_dl57dfaNRHoboAAAA, user.id=, user.name=
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG POST /api/signin 200 246.304 ms
2019-08-19T22:58:29.072Z - debug service:cncengine New connection from 192.168.2.3: id=wKAZGgxWkq5hheyFAAAB, user.id=, user.name=
2019-08-19T22:58:29.315Z - debug service:cncengine socket.list(): id=wKAZGgxWkq5hheyFAAAB
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /e9e533b0/favicon.ico 200 233.082 ms
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /api/commands?_=1566255509950&paging=false 200 2191.757 ms
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /api/state?_=1566255509949 200 2180.124 ms
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /api/machines?_=1566255510267 200 2161.921 ms
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /api/macros?_=1566255510269 200 1986.526 ms
V0y2UOi_GvOpVbANpHi0ctW2iKADV3MG GET /api/mdi?_=1566255510268 200 1969.761 ms
2019-08-19T22:58:45.704Z - debug service:cncengine socket.open("/dev/ttyACM0", {"controllerType":"TinyG","baudrate":115200,"rtscts":false}): id=wKAZGgxWkq5hheyFAAAB
2019-08-19T22:58:45.829Z - debug controller:TinyG Add socket connection: id=wKAZGgxWkq5hheyFAAAB
2019-08-19T22:58:45.943Z - debug controller:TinyG Connected to serial port "/dev/ttyACM0"

```

leoheck, 2019-08-19:
I am also doing it, I will paste it here.

The version of my softwares.
CNCjs: 1.9.20
Node.js: 10.16.3
NPM: 6.9.0

MitchBradley, 2019-08-19:
I am starting to suspect that the serialport module is confused about line endings - or maybe it is an xon/xoff problem.  I vaguely remember that someone was having a problem like that.  I will try to find the issue that described it.

leoheck, 2019-08-19:
The current settings
```
➜  ~ stty -F /dev/ttyUSB0
speed 115200 baud; line = 0;
min = 1; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

And here is the verbose output

[image]

```
➜  ~ cnc -vv          
2019-08-19T23:04:16.671Z - info init Loading configuration from "/home/pi/.cncrc"
2019-08-19T23:04:16.687Z - info init Watching "/home/pi/cnc" for file changes
2019-08-19T23:04:17.013Z - debug app app.settings: {"x-powered-by":false,"etag":"weak","env":"production","query parser":"extended","subdomain offset":2,"trust proxy":true,"views":["/usr/local/lib/node_modules/cncjs/dist/cnc/web","/usr/local/lib/node_modules/cncjs/dist/cnc/app/views"],"jsonp callback name":"callback","view cache":true,"verbose errors":false,"case sensitive routing":true,"strict routing":false,"view engine":"html"}
2019-08-19T23:04:17.225Z - debug app assets: name=web, asset={"routes":["/bfc45480/","/"],"path":"/usr/local/lib/node_modules/cncjs/dist/cnc/web","maxAge":31536000000}
2019-08-19T23:04:17.227Z - debug app > route=web /bfc45480/
2019-08-19T23:04:17.230Z - debug app > route=web /
2019-08-19T23:04:17.278Z - debug service:cncengine Loaded controllers: Grbl,Marlin,Smoothie,TinyG
2019-08-19T23:04:17.359Z - info init Starting the server at http://127.0.1.1:8000
2019-08-19T23:04:20.414Z - debug service:cncengine New connection from 192.168.86.107: id=tDXby63Pg1oAfQr2AAAA, user.id=7e9d0aff-ef66-4907-8e42-c1c6cb034674, user.name=lheck
2019-08-19T23:04:30.612Z - debug service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Grbl","baudrate":115200,"rtscts":false}): id=tDXby63Pg1oAfQr2AAAA
2019-08-19T23:04:30.636Z - debug controller:Grbl Add socket connection: id=tDXby63Pg1oAfQr2AAAA
2019-08-19T23:04:30.676Z - debug controller:Grbl Connected to serial port "/dev/ttyUSB0"
2019-08-19T23:04:35.875Z - debug service:cncengine socket.command("/dev/ttyUSB0", "homing"): id=tDXby63Pg1oAfQr2AAAA
2019-08-19T23:04:45.579Z - debug service:cncengine socket.command("/dev/ttyUSB0", "reset"): id=tDXby63Pg1oAfQr2AAAA
2019-08-19T23:04:51.925Z - debug service:cncengine socket.command("/dev/ttyUSB0", "homing"): id=tDXby63Pg1oAfQr2AAAA
2019-08-19T23:05:00.035Z - debug service:cncengine socket.write("/dev/ttyUSB0", "?", null): id=tDXby63Pg1oAfQr2AAAA
```

MitchBradley, 2019-08-19:
Here is the old problem report:  https://github.com/cncjs/cncjs/issues/257#issuecomment-358544246

leoheck, 2019-08-19:
Running it from the command line I am seeing some errors.

[image]

leoheck, 2019-08-19:
I got this `[MSG:Check Door]` message.

leoheck, 2019-08-19:
I started CNCjs, connected to the `/dev/tyUSB0` then on terminal I ran 
```
stty -F /dev/ttyUSB0 115200 cs8 -cstopb -parenb
```
Now when I do this
```echo '$$' > /dev/ttyUSB0```

I see this responce on CNCjs
[image]

I believe CNCjs is doing something wrong with the serial port now.

MitchBradley, 2019-08-19:
It looks to me as if your serial connection is dropping characters left and right.  Do you have a favorite terminal emulator program e.g. minicom, screen, or picocom?  I have never had much success using echo and cat to diagnose serial problems.

leoheck, 2019-08-19:
Sure, I am doing the same with screen with this command
```screen /dev/ttyUSB0 115200,cs8,-parenb,-cstopb,echo```

My issue is that I don't know how to use it properly. It does not echo the GRBL commands on the screen so I don't know if it is working.

MitchBradley, 2019-08-19:
It is much easier to get local echo working with picocom or cu.  Both are tiny so installing them is quick.

```
sudo apt-get install picocom
picocom -ec --imap lfcrlf /dev/ttyUSB0
```

The exit command is ^A^Q

leoheck, 2019-08-19:
Yes, I was doing right that. Thanks Mitch.

I figured out the issue with the command after installing the picocom
```
➜  ~ sudo lsof /dev/ttyUSB0
COMMAND    PID           USER   FD   TYPE DEVICE SIZE/OFF    NODE NAME
Repetier-  582 repetierserver   13u   CHR  188,0      0t0 1116647 /dev/ttyUSB0
picocom   5013             pi    3u   CHR  188,0      0t0 1116647 /dev/ttyUSB0
```

This old `repetierserver` process was connected to the `/dev/ttyUSB0` too. =\

After stopping this service it started to work just fine.
[image]

Thank you very much for your help and time to debug this. I really appreciate that.

MitchBradley, 2019-08-19:
Hmmm - the other thing that comes to mind is maybe there is some other process that is listening on the serial port and eating some of the characters.  Aha - yep, its the repetier service.

leoheck, 2019-08-19:
In short, it is always good to check other process consuming your resources.

closed 2019-08-19 by leoheck