cncjs/cncjs #511: Authentication error

SVANNER, 2019-08-22:
#### Description

Allowed to log into CNCJS from a client browser using a userid/password when the server is started manually in a terminal window using cnc command. When starting the same cnc command using systemd (service started at boot), the login with the same user is rejected with the message "error Authentication failed".

#### Versions

- CNCjs: 1.9.20
- Node.js: 10.15.2
- NPM: 6.11.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2019-08-22:
When you start cncjs via systemd, it is probably running as root, so it will try to get credentials from ~root/.cncjs instead of from your user home directory.

I start cncjs from my user crontab instead of from systemd.

As non-root user (pi in this case):

```
$ crontab -e
(Add this line to the end:)
@reboot /home/pi/startCNC
(Save file and exit)
```
startCNC is an executable shell script containing 

`/home/pi/GitHub/cncjs/bin/cnc -H 192.168.2.100 -p 8000 -m /tablet:/home/pi/GitHub/cncjs-shopfloor-tablet/src >> $HOME/cncjs.log 2>&1
`

Of course you would use whatever arguments apply to your setup.

You could skip the startCNC intermediate and just put the expanded command line directly in crontab, but I find it useful to have the shorthand name for interactive testing.

itsbrianburton, 2024-10-17:
Bumping as I'm hitting this exact same issue from 5 years ago and want to provide some additional information to help resolve it.  It doesn't ask me for a user running `cncjs` manually, but it does if it executes with systemd.  This is on a brand new installation of cncjs with a default configuration file.

**Environment**
Raspberry Pi
Linux 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
Node: v14.21.3
CNCjs: v1.10.4

`cat /home/pi/.cncrc`
```
{
    "state": {
        "checkForUpdates": true,
        "controller": {
            "exception": {
                "ignoreErrors": false
            }
        }
    },
    "secret": "<secret code>",
    "mdi": [],
    "machines": [
        {
            "name": "Genmitsu 4040-pro",
            "limits": {
                "xmin": 0,
                "xmax": 390,
                "ymin": 0,
                "ymax": 390,
                "zmin": 0,
                "zmax": 100
            }
        }
    ]
}
```

The systemd unit file `cat /etc/systemd/system/cncjs.service`
```
[Unit]
Description=CNCJS
After=network.online.target
Wants=network.online.target

[Service]
Environment="HOST=127.0.0.1"
Environment="PORT=8000"
Type=simple
User=pi
ExecStart=/usr/bin/node /home/pi/.nvm/versions/node/v14.21.3/bin/cncjs -c /home/pi/.cncrc

[Install]
WantedBy=multi-user.target
```

`ps aux` showing the process running as user `pi`
```
pi@sidewinderx2:/var/log $ ps aux | grep cncjs
pi       13553  2.0  7.9 183352 70844 ?        Ssl  16:57   0:13 /usr/bin/node /home/pi/.nvm/versions/node/v14.21.3/bin/cncjs -c /home/pi/.cncrc
```

Results of: `sudo journalctl -u cncjs.service`
```
Oct 17 16:55:29 sidewinderx2 systemd[1]: Started CNCJS.
Oct 17 16:55:38 sidewinderx2 node[13497]: 2024-10-17T15:55:38.568Z - info init Loading configuration from "/home/pi/.cncrc"
Oct 17 16:55:39 sidewinderx2 node[13497]: 2024-10-17T15:55:39.676Z - info init Starting the server at http://127.0.1.1:8000
Oct 17 16:55:40 sidewinderx2 node[13497]: TypeError: Right-hand side of 'instanceof' is not an object
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Object.module.exports [as sign] (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/jsonwebtoken/sign.js:114:58)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at generateAccessToken (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/dist/cncjs/server/api/api.users.js:40:40)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at signin (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/dist/cncjs/server/api/api.users.js:91:18)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Layer.handle [as handle_request] (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/layer.js:95:5)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at next (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/route.js:137:13)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Route.dispatch (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/route.js:112:3)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Layer.handle [as handle_request] (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/layer.js:95:5)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at /home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:281:22
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Function.process_params (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:335:12)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at next (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:275:10)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Layer.handle [as handle_request] (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/layer.js:91:12)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at trim_prefix (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:317:13)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at /home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:284:7
Oct 17 16:55:40 sidewinderx2 node[13497]:     at Function.process_params (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:335:12)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at next (/home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express/lib/router/index.js:275:10)
Oct 17 16:55:40 sidewinderx2 node[13497]:     at /home/pi/.nvm/versions/node/v14.21.3/lib/node_modules/cncjs/node_modules/express-jwt/lib/index.js:128:7
Oct 17 16:55:40 sidewinderx2 node[13497]: Thu, 17 Oct 2024 15:55:40 GMT express deprecated res.send(status, body): Use res.status(status).send(body) instead at lib/middleware/errclient.js:24:11
Oct 17 16:57:14 sidewinderx2 systemd[1]: Stopping CNCJS...
Oct 17 16:57:15 sidewinderx2 systemd[1]: cncjs.service: Main process exited, code=killed, status=15/TERM
Oct 17 16:57:15 sidewinderx2 systemd[1]: cncjs.service: Succeeded.
Oct 17 16:57:15 sidewinderx2 systemd[1]: Stopped CNCJS.
```
Loading the cncjs in the browser forwards me to the login page even though no users are configured.

[image]

cheton, 2024-10-19:
Could you try upgrading your Node.js version to 18 or 20？

cheton, 2024-10-19:
This issue is related to https://github.com/auth0/node-jsonwebtoken/issues/885

itsbrianburton, 2024-10-21:
No dice.  Steps performed to test on Node 18:

```
$ sudo systemctl stop cncjs
$ nvm install 18
$ nvm use 18
$ nvm uninstall 14
$ npm install -g cncjs
$ sudo sed -i 's/v14.21.3/v18.20.4/' /etc/systemd/system/cncjs.service
$ sudo systemctl daemon-reload
$ sudo systemctl start cncjs
```

Same behavior and error as before, I'm pretty doubtful that it's a Node compatibility issue.  I wasn't able to install Node 20.18.0 due to GLIBC errors that I'll need to deal with another time.

I'll take a look at the authentication logic and if I can get it resolved I'll submit a PR.

closed 2024-10-22 by cheton