cncjs/cncjs #513: Possible Correction to Wiki page for using cncjs under WSL

nnnnicholas, 2019-08-23:
#### Description

Hello! I am working on a completely unrelated project that makes use of `npm` package `@serialport`. In [the documentation for using cncjs under Windows Subysystem for Linux](https://github.com/cncjs/cncjs/wiki/%5BDevelopment%5D-WSL-Tips), you mention:

> In node_modules/@serialport/bindings/linux.js, I changed the line

I believe the file is actually located in `node_modules/@serialport/bindings/lib/linux.js`. Does that seem correct? 

Thanks for your wiki page. I am not yet sure if it fixes my problem, but it is the most helpful thing I have found so far.

nnnnicholas, 2019-08-23:
I believe I found a second typo in that wiki page. The page says to replace the linux-list function with a custom wsl-list function. The code for wsl-list.js is 

`function listLinux() {
    const ports = [
        { comName: "/dev/ttyS3"},
        { comName: "/dev/ttyS4"}
    return Promise.resolve(ports);
}
module.exports = listLinux`

I believe there is a missing `]` before the `return Promise.resolve(ports);`.

If I am correct, the code should read 

`function listLinux() {
    const ports = [
        { comName: "/dev/ttyS3"},
        { comName: "/dev/ttyS4"}]
    return Promise.resolve(ports);
}
module.exports = listLinux`

Perhaps @MitchBradley can weigh in. Thanks again, Mitch!

MitchBradley, 2019-08-24:
Thanks for finding and fixing the typos.

nnnnicholas, 2019-08-24:
Great thanks!

closed 2019-08-24 by nnnnicholas