cncjs/cncjs #516: Fatal Error - Allocation Failed

JensAChristensen, 2019-09-13:
#### Description

After running the server for a couple of minutes, it crashed with this message:
pi@raspberrypi:~ $ cnc
2019-09-13T10:07:09.717Z - info init Loading configuration from "/home/pi/.cncrc"
2019-09-13T10:07:11.300Z - info init Starting the server at http://127.0.1.1:8000

<--- Last few GCs --->

[3354:0x2c76e78]   192585 ms: Scavenge 695.5 (722.6) -> 694.5 (723.1) MB, 15.9 / 0.0 ms  (average mu = 0.134, current mu = 0.064) allocation failure 
[3354:0x2c76e78]   192685 ms: Scavenge 695.8 (723.1) -> 695.0 (724.6) MB, 19.4 / 0.0 ms  (average mu = 0.134, current mu = 0.064) allocation failure 
[3354:0x2c76e78]   192788 ms: Scavenge 696.4 (724.6) -> 695.5 (730.1) MB, 26.8 / 0.0 ms  (average mu = 0.134, current mu = 0.064) allocation failure 

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

#### Versions

- CNCjs: 1.9.20
- Node.js: 10.16.3
- NPM: 6.11.3

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
Marlin

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-09-13:
What flavor of Pi?  Are there other processes running that are using a lot of memory?  The "top" command will tell you about process memory usage.

JensAChristensen, 2019-09-13:
It's the Pi 4b 4gb. Nothing else running except the GUI and chromium

JensAChristensen, 2019-09-13:
Tried running the server without the RAMPS board connected, and everything was fine. CNC.js took up about 1.3% of memory without any increase in usage. 
I then attached the RAMPS and the memory usage increased with 0.2% per second until it reached 19.3% and crashed ....

MitchBradley, 2019-09-13:
Sounds like a memory leak in the MarlinController subsystem

MitchBradley, 2019-09-13:
I don't have time to track this down at the moment but if you want to do some experiments to help isolate the problem, here is something to try.
For testing, you can make small changes without recompiling by directly editing the file  dist/cncjs/server/index.js .  In that file there are three occurrences of a line that says "_this.query.issue()".  The purpose of those lines is to periodically ask Marlin for position and temperature updates.  The reason I suspect that is because it is the only think I can think of that happens autonomously, independent of GCode activity.

First, save a copy of index.js so you can easily go back to the unmodified version.  Then comment out those lines (add "//" at the beginning) and restart the server.  If that eliminates the ever-increasing memory usage, we know that there is a leak there.  There may be other leaks but that is a good one to look for first.

JensAChristensen, 2019-09-13:
I do not have a file called dist/cncjs/server/index.js ....

MitchBradley, 2019-09-13:
Sorry, that is the new name of the file in the development version.  In version 1.9.20, when installed globally, the file is /usr/lib/node_modules/cncjs/dist/cnc/app/index.js

JensAChristensen, 2019-09-14:
Commenting out the 3 _this.query.issue()-lines makes no change.

I tried flashing another version of Marlin onto the ramps board as well, that made no change either

MitchBradley, 2019-09-14:
Maybe I can find some time to look into this, but it might not be soon. Memory leaks can be tricky to find and I don't live and breathe Javascript so I will have to learn some debugging techniques.  Cheton might be able to find it faster but he is really busy at the moment.

JensAChristensen, 2019-09-16:
It is only if I connect using /dev/tty/AMA0 I have the problem. If I connect using /dev/tty/USB0 everything is fine ...

cheton, 2019-09-17:
There are several [memory leak issues](https://github.com/serialport/node-serialport/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aclosed++memory+leak) discussed in the node-serialport repository, some users reported that memory usage is steadily increasing.

I'm not sure whether the memory leak issue was resolved in `node-serialport@6.2.2` (used in CNCjs 1.9.20), but you can try below solution if ttyAMA0 is used by Bluetooth.

https://github.com/serialport/node-serialport/issues/1583#issuecomment-399946145
> #### Cause
> On RPi Zero W (and probably RPi 3) port serial1 is linked to ttyAMA0 and used by Bluetooth
> #### Solution 2 (maybe better)
> If Bluetooth is not needed, add dtoverlay=pi3-disable-bt to /boot/config.txt and reboot. Now full UART /dev/ttyAMA0 is usable on GPIO14 and 15.

Otherwise, connecting through the serial device /dev/ttyUSB0 should be the right choice if the board is connected using an USB cable.

nullbreached, 2019-11-15:
No help I realize but I'm experiencing this issue as well with a PI3. I'm just barely experienced enough to get this up an and running on a PI … so I apologize. I see you're discussing serial port stuff, I didn't know how to create a non-root user so I had to install CNCjs with the --usafe-perm parameter. Is this connected to the issue?

MitchBradley, 2019-11-15:
a) The --unsafe-perm thing is unlikely to be related to the problem
b) If you use one of the normal Pi distros like Raspbian, you will automatically have a non-root user named "pi"

heduobacht, 2020-04-15:
So it seems that the error is only happening when you try to connect to /dev/ttyAMA0  where nothing is connected (at least no CNC or anything else). If you connect the CNC via USB-Port it is called /dev/ttyAMC0 and not ttyAMA0. When you connect via this real connection the Problem is not occuring anymore. So simply ignoring or hiding the ttyAMA0 interface should do the trick. Then nobody can try to connect via this nonexistent? interface. I am not a programmer guy so i only wanted to give some hints to get to a solution.

MitchBradley, 2020-04-15:
Unfortunately, there are cases where you do need to use ttyAMA0.  The popular Protoneer Pi CNC shield connects via ttyAMA0, so excluding that would break compatibility with that shield.

This is one of those annoying "damned if you do and damned if you don't" situations.

One might also consider trying to detect quickly whether or not a controller is connected to the port. CNCjs in fact does try to do that, but it's not perfect, due to some inherent challenges.  The four different protocols that CNCjs supports (Grbl, TinyG/g2core, Smoothie, Marlin) all have different behavior at startup and reconnect, and even within a single controller type, the behavior can change depending on which microprocessor you are using and how its serial port is implemented.  It's a really difficult problem.  To do it perfectly, if that is even possible (conflicting requirements), would require constant testing against an ever-growing list of controllers.  Marlin 2.0, for example, currently supports 11 different base microcontrollers.  Commercial products like Microsoft Windows have large testing teams in well-equipped labs, supported by the price of Windows.

CNCjs development is mainly just one guy (Cheton) with a full-time job doing other stuff.

xemsilet, 2021-12-28:
I got this code for my PI I had my GPU_MEM set to 512 changed it to 256 seems to be working fine now on the PI 4 4G model

emcniece, 2022-03-27:
☝️ Can anyone else confirm that increasing GPU_MEM resolves this issue?

xemsilet, 2022-03-27:
> ☝️ Can anyone else confirm that increasing GPU_MEM resolves this issue?

It is the issue of overclocking the gpu if you got 1-2 gig model you may have a problem I gave my neighbor my 4 gig model and I overclocked the gpu and got the error I could be wrong on the model but it is worth the info but overclocking the gpu shouldn’t be needed

MBerka, 2023-09-11:
I had the same error and pattern on a Pi 3B+ with containers, happening specifically for `/dev/tty/AMA0`, `node-serialport@9.2.0` and newer versions of Node. @cheton 's link led in the right direction. Disabling via overlay was not working, but `systemctl disable bluetooth` followed by a reboot both stopped the recurring program crashes and caused the program to begin receiving data.