cncjs/cncjs #517: Can't find the .cncrc file on Mac?

victorober, 2019-09-17:
#### Description

Can't find the .cncrc file on Mac?
I can't login into Cncjs so I have to open the .cncrc file but its nowhere to be found ?
(Beginner to this type of program)

Cheers Victor !

#### Versions

- CNCjs: 1.9.x 

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x ] Mac
- [ ] Linux

MitchBradley, 2019-09-17:
I no longer have a Mac so I cannot say for certain, but I can give you a couple of clues that might help you find it:
1) On most systems, .cncrc is in the user's home directory https://www.cnet.com/how-to/how-to-find-your-macs-home-folder-and-add-it-to-finder/ .
2) Since the name .cncrc begins with a period, some commands like "ls" will not show it in a listing unless you ask specifically.  You can see them with "ls -a".  Search the web for "macos show hidden files" to learn other ways to see such files, e.g. in the Finder.

cheton, 2019-09-18:
This `.cncrc` file is probably under your user home. For example, the path of my `.cncrc` file is located at `/Users/cheton/.cncrc`.

```sh
tw-cheton-mbp:~ cheton$ pwd
/Users/cheton

tw-cheton-mbp:~ cheton$ cat .cncrc
{
    "state": {
        "checkForUpdates": true,
        "controller": {
            "exception": {
                "ignoreErrors": false
            }
        }
    },
    "secret": "...."
}
```

If you forgot your password, follow the steps in the wiki page and specify an empty array for the "users" property.
https://github.com/cncjs/cncjs/wiki/FAQ#forgot-your-password

victorober, 2019-09-18:
Yes ! I found it, thank you guys.

closed 2019-09-18 by cheton