cncjs/cncjs #518: Why is there a google analytics script embedded in the software?

IHadToPickaUsername, 2019-09-17:
Not an issue per say, but it seems kind of weird that the software sends informations about what we the users do in cncjs to a google analytics account.

This "feature" seems to have been implemented [april 22 2019 by Cheton ](https://github.com/cncjs/cncjs/blob/master/src/app/lib/analytics.js) without anyone knowing about it.

Checking with Wireshark tells me that it shares pretty much every changes I make to the interface, buttons I click and so on, as expected with a google analytics script.

While it is easy to disable/block the said script, an option to turn it off (or at least ask at first startup) would be preferable.

cheton, 2019-09-18:
Google Analytics tracking was initially added on Sep 26, 2017 (https://github.com/cncjs/cncjs/commit/07c8db3c711058f3561cc2d769a4f3afe2c8673a), the feature is available since the 1.9.11 release (https://github.com/cncjs/cncjs/releases/tag/v1.9.11).

The purpose of adding user behavior tracking is to know how many users are using this software around the world. So far the app only [tracks page views](https://github.com/cncjs/cncjs/blob/master/src/app/containers/App.jsx#L43) (i.e. navigating between workspace or the settings tab), no sensitive user actions or privacy information are recorded.

I do agree there should be a page that allows users to configure the analytics setting (will be _deactivated_ by default) and know what kinds of information will be collected, so they can decide whether to report anonymous usage data to the GA server.

I'm currently refactoring the code base to prepare 1.10 release, however, it may take a period of time. If you all think it is worth doing, I can make a small patch release (e.g. 1.9.21) to support configurable analytics setting.

closed 2019-09-25 by IHadToPickaUsername

dlitz, 2019-10-19:
As of CNCjs 1.9.20 this is still quietly enabled by default.  Is this not a GDPR violation?