cncjs/cncjs #522: development on the basis of CNCjs

NigelNN, 2019-10-08:
#### Description

I've been following this project for some time, and I'd like to do some development on the basis of CNCjs. Such as _Input coordinates and radius automatically generate Gcode then punch holes_. But now I met some problems.

I have now completed the algorithm for generating Gcode, but I can't combine my algorithm with CNCjs.

Can you tell me are there any method in CNCjs that I can send Gcode to machine? Or any information that will help me. I am very willing to contribute to CNCjs within my capabilities.

Thank you for any help.

#### Versions

- CNCjs: latest
- Node.js: latest
- NPM: latest

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-10-10:
https://github.com/kreso-t/cncjs-kt-ext shows one way to connect external programs to CNCjs.  You could set up a CNCjs macro to accept the parameters, then it would send them to your program, which would then send back the GCode.  That's basically how cncjs-kt-ext works.

rlarge, 2019-10-15:
I was wondering if you could use a custom widget for that also from what i looked at you need to get the secret key stored in the ~/.cncrc file.  My plan time permitting is to incorporate Node RED into a widget to process/watch preform flows for a airbrush project.  Its pretty much an advanced macro setup with visual programming.  If anyone has any ideas/direction that would be the best i would love to hear it.  Thanks