cncjs/cncjs #523: Raspberry Pi 4 Standard Install Issue

Protoneer, 2019-10-09:
#### Description

Try to install on a RPi 4.

Error messages:
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:196:23)
gyp ERR! stack     at ChildProcess.emit (events.js:209:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.75-v7l+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport
gyp ERR! node -v v12.10.0
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok
npm WARN react-datepicker@1.5.0 requires a peer of react@^16.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN react-datepicker@1.5.0 requires a peer of react-dom@^16.0.0 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-09T20_24_09_590Z-debug.log

#### Versions

- CNCjs: latest
- Node.js: 12.10.0
- NPM: 6.10.3

#### How Do You Install CNCjs?

-  NPM

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x ] Linux

Protoneer, 2019-10-09:
Install script I ran was as follow:
`
echo "prefix=~/.npm" >> ~/.npmrc
cat ~/.npmrc
npm config set user 0
npm config set unsafe-perm true

npm install -g socket.io
npm install -g socket.io-parser
#npm install -g serialport
npm install -g cncjs
#sudo npm install --unsafe-perm -g cncjs
export PATH=$PATH:~/.npm/bin/
`

MitchBradley, 2019-10-09:
https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)* installs Node v10.  It appears that you have Node v12 installed.

CNCjs currently uses serialport ~6.2.2.  As I understand it, Node v12 support was first added in serialport v7.1.5.

It might work if you install serialport before cncjs, thus possibly overriding the serialport version specification in cncjs/package.json .  Alternatively, you could try changing cncjs/package.json to refer to serialport 7.1.5.

Let us know what you discover.

Protoneer, 2019-10-10:
I gave it a try but in the end reverted to Node 8 to get it going. 😉

tillig, 2020-01-08:
Same issue on Raspberry Pi 3B+. I tried with Node v13 and got the same compilation failure. Reverting to Node 8 works.

As a related/aside, the [wiki page on installing via Package Manager](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*) which says it's the recommended way currently shows Node 8 as the thing to install. This differs from the [getting started docs on the cnc.js.org home page](https://cnc.js.org/) which shows Node Version Manager installing Node 4. I'm curious if it'd be better to link the home page to the wiki rather than having several disparate docs to maintain.

AustinSaintAubin, 2020-02-01:
I am working on an install script for the Raspberry Pi. In the meantime here are my notes as of a few days ago. https://github.com/cncjs/cncjs-pi-raspbian/blob/master/README_BUILD-NOTES.md