cncjs/cncjs #526: UGS and CNCjs not communicating with the Arduino UNO

artvdw, 2019-10-15:
It doesn't matter which of the two pieces of software I use, when I click 'run' (after Grbl has been uploaded and com ports checked etc) the computer screen comes alive and I can watch the tool move around, on the screen,  as the code scrolls by, but there is NO movement in the CNC. There isn't even a hint that it is trying to move.  My Arduino Uno has a CNC shield and I have downloaded the latest versions of Arduino IDE and UGS, CNCjs.  Has anyone experienced something like this?
Anyone have any suggestions before I try Mach 3?

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

neilferreri, 2019-10-15:
Sounds like you have communication working ok.  Probably a wiring issue.
You'll need to provide more details.

MitchBradley, 2019-10-15:
do you have a separate power supply for the motors connected to the driver shield? The power from the Uno board is not enough to run motors.  A photo of the whole system and closeups of the driver shirld wiring would help

MitchBradley, 2019-10-15:
When you turn on the motor power supply, the motors should "lock", which means that it will become much harder to rotate the motor shaft by hand, and they might make a faint humming sound. With the power off, stepper motors can be rotated easily with only the slightest resistance. If the motors do not lock with motor power applied to the driver shield, it could be caused by several problems
- incorrect stepper wiring
- incorrect polarity of motor power supply
- incorrectly installed stepper driver modules (if the shield has modular drivers; some do and some don't)
- stepper drive current set way too low
Most stepper modules (or non-modular driver shields) have instructions for setting the drive current. Consult the instructions for your drivers and set the current appropriately for your motors.

If your motors do lock but do not move in response to jog commands from a user interface, the problem is likely due to GRBL misconfiguration. GRBL must be told which Uno pins drive which motors. That depends on both the shield design and the way that motors (X Y and Z) are connected to the shield.

artvdw, 2019-10-15:
I do have a separate power supply for the motors.  I also hooked up one of the motors to the Arduino Uno and a stepper driver according to a video I saw on YouTube and the motor ran.  I used code in the Arduino that I borrowed from the video.

Here are two photos of the set up of what I am working with.  Sorry for the ‘can of worms’ on the desk, but it is easier here at the desk where I can access a Mac and a PC.  It will look neater, when I do the final wiring!

I just sent this message with the two photos and it was rejected as the photos were too large, so I am trying again.

                            

I also applied power to the Arduino (which is meant to power the motors) and the motors turned easily, so it looks like I will have to look into that!

MitchBradley, 2019-10-15:
If you continue to have problems with photos, you could try the cncjs users group on Facebook. I follow that group

MitchBradley, 2019-10-15:
I am really surprised that the Arduino is supposed to power the motors. The numbers just don't add up. Motors need more snow than an Arduino board is rated to supply. The usual setup is to have a supply that can source 4 or more amps at 12V connected to the driver shield, with the Arduino supplied from 5V / 0.5A.  The 5V for the Arduino can come either from a regulator on the shield or from the USB port.

artvdw, 2019-10-16:
Mitch,

I am not powering the motors with a 5V supply, but rather with a 12V - 10A supply. I am going to check the amperage flowing through the motor drivers and make sure they aren’t set too low.

Thanks for the tip about the CNCjs group.

MitchBradley, 2019-10-16:
Post photos on Facebook. I approved your membership request

artvdw, 2019-10-17:
Well, I have read more than once that a person, having problems connecting, used a different Arduino, and bingo, it worked.  So at this point, I have ordered another Arduino, and also another CNC shield and will try once more, when they arrive.  

So, it will be pretty quiet from this end until the new parts arrive.

MitchBradley, 2019-10-24:
The problem turned out to be a combination of bad wiring and incorrectly set driver current, leading to a blown driver module that dragged down the Arduino power - or something like that.  Anyway, the problem is now resolved - and it had nothing to do with CNCjs.

closed 2019-10-24 by MitchBradley