cncjs/cncjs #527: Keyboard input movement ignores limit switches

ubidefeo, 2019-10-25:
#### Description
I use CNCjs to run my Shapeoko 3 XL.
When I use keyboard to move around and the machine hits the homing switches it will push through and cause belt skipping.
I've been searching in the issues but haven't found anyone posting about it.
I think a test for switches should be made at every step, and it doesn't seem to be the case.
When I use the terminal to send a `G0` command it won't test against limits either.
Is this something which can be set somewhere?

Any pointers appreciated 🙏 
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-10-25:
CNCjs does not test the limit switches directly; that is the controller's (GRBL's) job. CNCjs could not possibly do it because CNCjs is not a "real-time" program, so it cannot respond fast enough to stop movement instantly when a limit switch is reached.

The likely problem is that, when jogging, the machine is going so fast that, when it hits the switch, it is not possible to stop quickly enough to avoid ramming into something. It is as if you slam on the brakes of a car 1 meter from an obstacle while moving fast; you will hit the obstacle.

One solution is to configure "soft limits" in GRBL. Search the web for "setting soft limits in grbl".  You will also need to home the machine every time you power it on, so GRBL knows the actual position in machine coordinates. With that turned on, if you try to issue a command that would go beyond the software limits, GRBL will refuse to do it.  GRBL will also enter alarm state in that case, so you will have to manually soft reset to cancel the alarm.  The reported position will be correct because a "soft stop" uses a controlled deceleration, like braking gently to a controlled stop in a car.

Another solution is to reduce the jogging speed so the machine is going slow enough to actually stop in time.  In that case you will also enter alarm state, but the situation is worse because hitting a limit switch causes a "hard stop" - the machine just turns off the motors instantly.  In that case, the momentum of the machine can cause the machine to skip and "lose steps", thus rendering the reported position unreliable.

You really should avoid hitting limit switches except in the special case of homing cycles, which move the machine slowly and carefully to avoid problems.  In normal operation, limit switches are emergency precautions to prevent disaster, not intended for regular use.

This means that, for keyboard jogging, you really should tap the keys carefully, instead of just holding one down and letting it go all the way.  If you hold down a key, it is likely to auto-send more keystrokes than you want.  At some jog increments, the keyboard can send much faster than the machine can move.  Suppose that you have the jog increment set to 10 mm and the jog speed is 1000 mm/min. Each jog move takes 1/100 of a min, or 0.6 sec.  Keyboards often repeat at 10 characters per second, so if you hold the key down for 2 seconds after it starts repeating, you have "queued up" 20 jog commands for a total of 200 mm, but the machine has only moved about 30 mm by the time you release the key, and will continue to move 170 mm thereafter.

I suspect that what you really want is for the machine to jog while you are holding the key down and then stop with a controlled deceleration when you release it.  The way to approximate that is to reduce the jog increment until the machine stops soon after you release the key.  It might be difficult to find an exact increment that results in smooth movement (too-small increments will make the movement jerky) and stopping immediately (too-large increments will make the machine continue to move after you release), but perhaps you can find an acceptable compromise.

closed 2019-10-25 by MitchBradley

ubidefeo, 2019-10-25:
hi @MitchBradley 
thank you so much for the exhaustive answer.
I am new to the CNC world and as I learn I come up with new questions.
You understood what I wanted to achieve and gave me a reason why it wouldn't be possible.
I have read something about soft-limits but haven't gotten to that point yet, and now I think this is what I'll look at next :)

cheers
ubi