cncjs/cncjs #528: Signin Not Working for Desktop app

markus4e, 2019-10-26:
#### Description

Signin is not working.  I am using my username and password to no avail.  And I have no way to go back and reset my signin information.  I have checked the Windows location for the password and it states none is thee.  I have uninstalled the Desktop app and reloaded with no solution.  I need help since there is no way it is letting me on.  I used it this morning with no problem and came back to it stating it closed.  Went to open the app and it would not allow me to signin.

#### Versions

- CNCjs: 1.9.2
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-10-26:
The password for the Windows app is usually stored in C:/Users/YOURNAME/.cncrc .  In your report, you stated that you checked "the Windows location" but you did not specify the specific place that you checked, so we cannot tell if your idea of the location was mistaken.
In problem reports, it is important to be very specific.

markus4e, 2019-10-26:
I thought I had looked at .cncrc, but apparently not.  I see my username and I see a password that is all gibberish about 57 characters long with symbols I have not seen used in a password.  First of all the username was different than I expected which is OK since it could be one I chose, but the password is clearly not.  I tried inserting these symbols as the password and it did not work.  Can I change the password between the quotes or how do I reset it so it works?

markus4e, 2019-10-26:
I just uninstalled using CNCjs uninstall program.  I then trashed the .cncrc file.  I checked to make my windows 10 is 64 bit which it is.  I reinstalled the CNCjs desktop app and it comes up without asking for a username and password?  Is it automatically using my GITHUB username and password?  I checked the .cncrc file and it shows no username or password.  I have not hooked up to my CNC yet, but I have been able to bring app up several times.  I will try with my XCarve and close down after CNC use to see if I have the signin problem again.  I will report later.

MitchBradley, 2019-10-26:
- It is not necessary to enroll a user and password.  I normally do not.  The case where you would want to protect it with a user/password login is if CNCjs is reachable from an outside network.  For the typical home-use case, it's not necessary.  If there is no "users" section in the .cncrc file, or if there is a "users" section with no user entries inside it, CNCjs does require a login.
- The gibberish password is encrypted.  Storing an actual password in a file is a very bad idea because anyone that manages to read the file can see the password.  The way this works is that, when you first create the password, the computer passes it through a one-way encryption function and stores the result, which looks like gibberish. Later, when you go to log in, the password you supply is re-encrypted with the same function and the result is compared with the stored one. The encryption function has the following properties:
    - If you have the gibberish, it is not feasible to convert it back to the original "cleartext" password
    - It is not feasible to find another "cleartext" string that encrypts to the same gibberish
"Not feasible" means that it requires an insane amount of computation that would take years on the fastest computers.  Such "one way" functions are an underpinning of computer security.

markus4e, 2019-10-28:
Mitch,  Thanks for the response.  I must have set up an account inside CNCjs not realizing it which caused this problem.

Separate question:  I cannot figure out how to use the visualizer app to run a simulation of my gcode before connecting directly to my XCarve.  When connecting directly to XCarve, the pressing the play button runs the CNC which I do not want to do until I have made sure it simulates correctly.  How do I do this?

MitchBradley, 2019-10-28:
At the moment there is no way to do this without extra hardware in the form of another Arduino running GRBL.  If you have a spare Arduino you can use that, or you can buy an Arduino Nano clone for $1 from china, or for a few dollars from a US supplier.  Load GRBL onto it, and connect to that instead of the XCarve to run the simulation.
Many people do their simulating in the CAM program; Fusion 360 in my case.