cncjs/cncjs #533: Feature Request - Auto Leveling

leoheck, 2019-11-08:
Hi, I was using this [OpenCNCPilot](https://github.com/martin2250/OpenCNCPilot/issues/28), it is awesome, I was thinking here, is it possible to add this functionality to the CNCjs?

aldenhart, 2019-11-08:
Note that g2core has a leveling function built in that you could use.

leoheck, 2019-11-08:
Hi @aldenhart what is g2core? do you have any extra reference?

leoheck, 2019-11-08:
I found this [What-is-g2core](https://github.com/synthetos/g2/wiki/What-is-g2core). How this is related to the CNCjs?

leoheck, 2019-11-08:
Maybe related with https://github.com/cncjs/cncjs/issues/32

aldenhart, 2019-11-08:
Yes, What is g2core is the connection. g2core is one of the CNC firmwares supported by CNCjs.

leoheck, 2019-11-08:
I am using the GRBL 1.1 firmware. Do you have any reference about how to use this kind of feature?

ril3y, 2019-11-08:
grbl is not g2core.  G2core is completely different :(

leoheck, 2019-11-08:
Oh, I see, so g2core is also a firmware like GRBL.
So, any chance to have these auto-leveling feature included on CNCjs anytime soon?
I was not able to understand if something was implemented on issue #32

MitchBradley, 2019-11-08:
This request is a duplicate of issue #32 .   Please see that for a solution to your problem.

closed 2019-11-08 by MitchBradley