cncjs/cncjs #534: Closing one UI closes all

MitchBradley, 2019-11-23:
CNCjs 1.9.20 installed on Raspberry Pi, Node v8.5.0, with "npm install"  (the problem exists in older versions too).

Start two instances of the CNCjs UI in browsers.  The browsers can be on one machine or different machines; it doesn't matter.

Open a connection to a controller on browser 1 and then open a connection to the same controller on brower 2.  Both browsers can control the machine and see its status as expected.

Now close the connection on one browser.  The connection closes on that browser and also on every other browser that is connected to that controller.

I think this is the wrong behavior.  If you are running CNCjs on a laptop and then open a connection on your phone to monitor it from somewhere else, you don't want the laptop connection to close when you close the phone connection.

I would be willing to attempt a fix for this, but I am a little confused about where the fix should go, since there are so many places that handle serialport open and close events.  If you can point me in the right direction I can do the work.

cheton, 2019-11-23:
The design concept of the close connection button _at the very beginning_, was to force close **serial connection** no matter what device you operate. If you want to close **socket connection**, just simply close the browser window.

I did notice that this is probably a UX defect that may cause unexpected result to users because it does not display a warning message to inform that the serial connection will be terminated. I'm currently working on [new UX enhancements](https://github.com/cncjs/cncjs/blob/feature/redux/src/app/widgets/Connection/Connection.jsx#L768) on the feature/redux branch. Now clicking on the [Close] button will popup a modal window. To resolve the UX issue, I would give two buttons on the footer of the popup modal, one is to "terminate the serial connection", the other one is to "close the socket connection", so user will get a clear message on what action to be performed.

MitchBradley, 2019-11-23:
The existing behavior is confusing because Open and Close work differently so the UI behavior is inconsistent. For Open, the UI views of the two browsers are independent, while for Close, they are synchronized.  I can't think of a good use case for simultaneous close.
I think there are two possible consistent behaviors:
1) Both Open and Close are independent. In this model, the UI works like a multi-line office telephone.  Either telephone can initiate a call, and the other can join the call.  When either one hangs up or switches to a different line, the other is still "on the call".  The call does not disconnect until both extensions hang up - or the other end hangs up.  This behavior is the way I expected it to work, since it is familiar and useful - office phones work that way because it is the obviously correct behavior for calls. With this model, you can start a job on UI A talking to controller A, then UI B can connect and monitor the job, then UI B can disconnect from controller A and reconnect to controller B without affecting the work in progress on A.
2) Alternatively, all UIs that are attached to the same server could view the same connection at all times.  In that model, opening a connection on one opens on all, and the same for closing.  That model makes it difficult or impossible to work with two different controllers attached to the same server, but it is consistent and easy to explain.

Perhaps the intention of the hybrid independent-open , coupled-close model was to reset the controller as a side effect of closing, but that would be better done with an explicit controller reset. Closing the serial connection happens to work as a reset on some controllers, but does not work in all cases.
I think the proposed modal dialog is a change for the worse.  I have always found it very annoying when a user interface requires an extra confirmation step for a straightforward operation.  To me, that is bad UI design, except in the special case when the operation has severe consequences.
Also, I think that the proposed "close serial connection" or "close socket connection" choice will confuse users more than help them, since very few users have the faintest idea what a "socket" is.

saxa, 2019-11-23:
Wouldnt it be simpler to just make cncjs the first started app as main and other started apps as slave with just monitoring functions ?

MitchBradley, 2019-11-23:
> Wouldnt it be simpler to just make cncjs the first started app as main and other started apps as slave with just monitoring functions ?

That would be very restrictive.  I often use both the main UI and the cncjs-shopfloor-tablet UI for controlling the same machine.  The former has some features that the latter lacks, while the latter is much safer to use for simple job control due to its large buttons.  Making one of the monitor-only would break that use case.

MitchBradley, 2019-11-23:
Re @cheton 's proposals, I prefer the second version.

I really need to get back to work on the code to synchronize with machine state so that using the workflow state would be possible.

The state of my work when I got interrupted a few month ago is as follows. 

1. I combined the sender, feeder and workflow state machines into one "sendflow" machine that handles, in the correct order, situations like stop-at-error, pause due to M1 and pause due to user input, reporting the correct message for the current situation.  It also handles feeding while sending is paused, by using a stack of data sources.
2. The sendflow machine emits a new set of state reporting events that the UI can be handled very easily without caring which controller type is in use.  For backwards compatibility, the old set of events is still supported.
3. Much of the logic that was duplicated between the different controller files, and duplicated between sender and feeder cases, is subsumed into the sendflow machine, with the differences pushed down into the ...Runner modules.  The files GrblController.js, TinygController.js, etc just go away.

The synchronized operation was working well with TinyG and almost working with GRBL.  The problem with GRBL centers around GCode errors, which may not stop the machine.  Marlin is a larger problem due to its poor serial protocol.  It might be necessary to adopt a "best effort" strategy.