cncjs/cncjs #535: Custom Commands bugs

Muralha, 2019-12-01:
#### Description

1 - On the Axes Settings, Custom Commands tab, any down Order button is not working. 

2 - Custom command button glitched positioning:

Added 1 button has a separator ("---------") with 100%, then added 3 more with 1/3% ("Center X", "Center Y" and "Probe Z"). 

[image]

Decided to delete the 1st button and it messes up the buttons position.

[image]

Changing to other percentage different from 1/3 fixes the problem.

btw, couldn't we get some spacing like the space between Axis and the jog control buttons? 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-12-12:
The percentage went wrong (especially the fraction 1/3) when screen or browser's zoom level is not 100%, I guess your zoom level might probably be 110%, 125%, or more. Just to let you know I was performing a big UI/UX improvement in a separate branch, the layout issue will be resolved in CNCjs 1.10 release.

Muralha, 2019-12-12:
The problem is on the electron app.
On the browser it's "fine", just a little too close to the buttons above, like on the images above.

With View > Actual Size, the 1st button must be > 1/3, or else it messes up.
Not even the widget at full screen fixes the problem.

I've tested zooming in a bit and after some zoom in, it fixes it.