cncjs/cncjs #536: Preserving blank lines (discussion)

MitchBradley, 2019-12-02:
I recently added a feature in cncjs-shopfloor-table to highlight the GCode line that is currently executing on the machine.  It works for g2core.  I'm not sure if it is possible to implement accurately on other controllers but I intend to try.

To make this feature work well, I had to disable the code that omits empty lines, so the line count tracks the original source.  Empty lines tend to occur infrequently and consume little buffer space, so they typically do not cause line bandwidth problems .

I intend to make a PR for this change, but there is a design decision to discuss.  It seems prudent to make this configurable, perhaps with an option named sendEmptyLines.  The question is where to put the option.  It would fit in the controller section alongside exception.ignoreErrors, but that raises the question of whether both sendEmptyLines and exception.ignoreErrors should really apply equally to all controllers.  Ideally, there would be such a variable for each machine that is connected to the server, since the best settings could be different for different machines.

In most cases it makes no difference because most setups have only one machine per server.  I'm just trying to think of how it should really work, since I am contemplating attaching three machines to the same server - a GRBL laser engraver, a GRBL mill, and a Marlin 3D printer.

Ideas?

MitchBradley, 2019-12-02:
Maybe such configuration options should be in a machine profile.  A machine profile could also refer to a serial port and controller.  Instead of directly selecting a serial port in the connection widget, you could connect to a machine, which would then open the appropriate serial port.  This seems better from a user interface standpoint, since the machine is the real focus of the user's attention, while the serial port and controller type are "under the hood" details.

The workspace configuration could also be subordinate to the machine, since different machines have different capabilities and thus need different widget sets.

cheton, 2019-12-05:
If we're going to release a hotfix version for 1.9, I'd suggest put it to the same place of the controller section shared by all controllers to make minimal changes to the configuration. For a future release like 1.10, I prefer to your second approach by implementing the idea of creating a serial or socket connection in accordance with the machine profile.

Machine-based configuration is a good approach for configuring machine profile and workspace settings, I will start to figure out how to get there. Also, it will be great if you can help find some products or services that provide similiar UI flow for reference.

MitchBradley, 2019-12-05:
@cheton That sounds like a good approach - controller-generic for now and move toward a more machine-centric configuration approach later.

My test version for preserving line numbering is working well.  I ran a large batch of parts yesterday and the synchronization was flawless.  I'll tidy up the code and issue a PR in the next few days.

MitchBradley, 2019-12-05:
I suspect that it will be hard to find programs that exemplify the multi-machine workflow, since very few UIs other than CNCjs are capable of supporting multiple machines at once.  The common situation is a single computer handling a single machine, so everything about the configuration is focused on one situation.

The UI flow that I envision is similar to the pendant example, where the initial entry is a connection screen, but instead of choosing a port, you choose a machine, and the connection to a port with a specific controller type happens based on the configuration of that machine.  If that machine has been selected for auto-connect, that screen is bypassed and the connection is automatic.

Of course there is the initial problem of setting up the machine configuration.  If there are no machines, or if you want to add a machine, we need a way of creating one.  Perhaps it could be similar to the existing flow, with the addition of a "save this setup as a new machine" function.

Once a machine profile has been created, the day-to-day need for Connection, Console, and controller-specific widgets would be greatly reduced.  They would still be useful for debugging purposes, but for normal usage, they could be omitted from the workspace to declutter it.

In the ideal world, once things are properly set up, the user should not have to care which port is in use and which controller is driving it, etc.  The UI should focus as singularly as possible on the machine.

mmdoogie, 2020-06-26:
Hmm I didn't notice this discussion previously

I was trying to do something similar and solved it in the opposite manner -- making sure the g-code listing returned by the gcode:load event and eventually displayed/rendered has the blank lines removed as well.  This was a small change and the goal of PR #617. I think even with the line removal removed or moved to an option there's still stuff like the %wait added at the end of the program that would still make #617 useful -- ensuring what's displayed in the UI is what is being sent in the background.

Still testing but I've got a branch where I highlight the currently-executing line number in the g-code listing view and it seems to work well on a Grbl controller as well, as another early data point.